ImpulseAdventure/GUIslice #6: Button elements resident in Flash memory

ImpulseAdventure, 2017-04-29:
To support memory-constrained targets (such as Arduino), it would be useful to have some limited support for buttons but without the full overhead of the `gslc_tsElem` structure.

nanokatz, 2018-01-06:
Hi There;

I've implemented this in my local copy of the 0.9.1 version of GUI slice. Further I have implemented the following:

Text labels where only the text string is in local memory
AdafruitGFX custom fonts (though it looks like in 0.9.2 you have done this)
Support for Adafruit ILI9341 in 8Bit mode
Support for Adafruit ILI9341 with simple resistive touchscreen

I'd like to submit some pull request for these features but I think it will take me sometime to redo the code for the current version not to mention just clean it up.
But I hope to get to it in the next month or so!

ImpulseAdventure, 2018-01-07:
Hey @nanokatz - a PR for the flash-resident buttons / text labels would be great! I'm curious as to what approach you ended up using (eg. defining a separate list of reduced-element structures, modifying the existing one, etc.?).

FYI -- towards a similar goal (ie. reducing memory consumption for Arduino), I have recently been considering whether it may be feasible to implement a "clone" element type. The thinking is that one often instantiates a large number of (stylistically) similar elements within a display. One could associate multiple clones to a single master element. The clones (which could include buttons/controls) would inherit all of the common attributes of the master (style, callback functions, etc.) but retain their own small subset of parameters: ID, position/size, text ptr, current state. The clone elements could probably be RAM-resident as they should be small. I still need to determine a clean way of ensuring that existing routines using `tsElem` could adapt to clones as well. The selection between full elements (including masters) and clones could be done via the `gslc_tsElemRef` arrays. The end-goal would be to enable a large number of GUI elements on the baseline Arduino devices.

I saw your separate PR #23 for the **88bit ILI9341** and **TouchScreen** adapters -- those would certainly be useful too.

Thanks for your efforts; much appreciated!

nanokatz, 2018-01-07:
I took an approach of modifiying the existing setup. In a nutshell I added a temporary element for storage to the maingui object. commands such as findelembycoord load the element from PROGMEM into RAM then return a pointer to this temporary storage.  This is working very well but sofar I haven't tried it with the compound elements such as selnum.

Regarding memory usage though this change is what made the library actually usable for me. With the buttons and dynamic text labels in PROGMEM, with guislice, sdfat, rtclibs and my own library that uses about 1/2kb the total memory usage is 4553 bytes. (To give you a sense one page alone has 5 sliders, 10 buttons, 8 dynamic labels and 4 static ones)

ImpulseAdventure, 2018-01-07:
That would be excellent! With your implementation do you support capturing the changed state of the temporary element due to the calls to ElemSetRedraw / ElemSetTxtStr / ElemSetGlowEn? Thanks

nanokatz, 2018-01-08:
Restoring the states of glow/redraw would be possible in theory (though right now I force the redraw for text elements). However I was giving this some thought and looking at the code a bit. Right now each element structure has separate variables for various booleans. In total it uses 6 bytes per element just for boolean values. My thought is that perhaps it would be better to convert these to a single byte where each bit is used for such a flag (much as various flags are done). This byte could then be stored with in the elementref array and restore after copying from progmem.  The best would be also to convert all elements to use this single byte style as that would be a savings of 5 bytes per element; and on something like Arduino where every bit counts I think that is good! This would of course require rewriting a fair bit of code.

For the text of the buttons I think the method for the dynamic progmem text labels could also be used without much change.

ImpulseAdventure, 2018-01-09:
Regarding the state booleans, I agree... I intend to make the optimization (was raised as useful for #24). After compressing the mutable state, your suggestion of storing them in the `tsElemRef` array seems reasonable.

Reference: the associated pull request for this issue is: PR #25

ImpulseAdventure, 2018-01-11:
Good news.... recent commit dec189a (in branch `nanokatz_02c0ab2`) has added support for glowing state in Flash-based buttons! They now operate much like RAM-based buttons. Example `arduino_min/glsc_ex07_ard_min.ino` has been updated to demonstrate these.

Note: This was done without changing the location of the element's `bGlowing` state. I did review what would be involved in moving an element's ephemeral state into the `tsElemRef` array but I haven't yet determined the best way of accessing the `tsElemRef` pointer from within the functions that are currently passed `tsElem*`.

@nanokatz - Please give this a try and let me know if this release works for you.

nanokatz, 2018-01-11:
Sounds good! I'll give it a try tomorrow and let you know how it goes.

Regarding the mutable state location; I've also been trying to come up with a good way of doing that that actually results in still saving memory. The simplest way would be to store a pointer to the ref element within the element itself. But there another 2bytes eaten. Still a savings of 3 bytes but if it can be avoided all the better. 

Another approach is just to do the compression and leave the states in the element array. You could then add a second state variable to the ref array that is only used for flash elements (and restore that when the element is called from progmem, much like what you have done so far). 

I was going to play around with the second approach myself once we got our codes more into sync. I still plan also to look at expanding the types of elements that can be kept in progmem to include some of the fancier ones such as selnum and sliders as now they are the biggest memory hogs for me.

nanokatz, 2018-01-12:
I gave it ago today with f491572 and everything seems to be working just fine. The buttons now glow when selected, text is still updating.

ImpulseAdventure, 2018-01-13:
FYI - I managed to get sliders working from Flash as well (with a new `ElemXSliderCreate_P()` call).

However, one challenge is that I think it would be better if the user could call the existing mutator (eg. `ElemXSliderSetStyle()`) and accessor functions (like `ElemXSliderGetPos()`) just like they would after calling `ElemXSliderCreate()`. I'm less keen on creating separate Flash variants for many of the API calls.

The root of the problem is that a `tsElem*` pointer is passed as a parameter throughout most of the APIs, but such an address would be insufficiently qualified to determine if it was RAM or Flash.

Thinking about it more, I believe the cleanest solution may be to revise the majority of the library's APIs to pass `tsElemRef*` pointers around instead. This could solve several problems:

- Enables a common API for both RAM and Flash based elements. The API calls will have a short check at the start for Flash and optionally copy into RAM if needed. This would also help trap accidental writes to read-only (Flash) element structs via API mutator functions.
- We have an easy way of storing/retrieving mutable state via the `tsElemRefFlags` that are also in the `tsElemRef` struct. This should enable the proper reuse of the existing redraw / event tree (including glowing state) without doing things like forcing a redraw upon element update calls, etc.

Thoughts?

ImpulseAdventure, 2018-01-22:
@nanokatz - Some good news on the Arduino RAM optimization front...

I have now completed a major update to the library (future version `0.10.0`) that utilizes element reference (`tsElemRef*`) pointers throughout instead of element (`tsElem*`) pointers. This has resulted in a far cleaner implementation when mixing between SRAM and Flash. Impact to user code is largely limited to renaming `tsElem*` to `tsElemRef*` and the addition of `tsGui*` to many of the APIs that were missing it.

The nice thing is that we can now use all of the original API calls to update elements, etc. (such as `ElemSetTxtStr()` or `ElemXSliderSetStyle()`) without resorting to special handling for Flash. In fact, I was able to remove all special casing of Flash and centralize it in `GetElemFromRef()`.

I should be able to easily address the last few remaining extended element types (eg. Gauges, Textbox, Graphs, etc.) and move them all into Flash as well. Note that I have temporarily deactivated the compound element (eg. XSelNum) while I review the best way to re-implement it. I also haven't re-tested the LINUX examples yet with this release.

All in all, with these mods done the SRAM consumption of the examples has been optimized considerably.

Please have a look at 08ab14a in the new branch `API_ElemRefPtr` and let me know if it works for you.

ALittleSlow, 2018-01-22:
Sweet!

#VIstrong

nanokatz, 2018-01-25:
@ImpulseAdventure Sorry I haven't replied sooner! It looks like you've made a lot of progress from my one rather hasty commit! I won't have time in the next couple of weeks to test things myself but it all looks good. But later in February I'll report back and try to give an idea of the memory savings in my case.

As for the whole ref pointer rather than element pointer I had the same idea. But I was rather hesitant to suggest it because it was such a large change!

ImpulseAdventure, 2018-01-26:
Thanks for the followup, @nanokatz -- sounds good. Without a doubt the `ElemRef*` change was pretty involved but I hope it strikes a good balance between increased usability for Arduino users against the cost of an extra indirection level. Given the magnitude of the change, it would be nice to get a quick test from someone else at some point before I merge this branch back into the master.

I will continue to look for optimizations so that more examples are able to run on the baseline **Arduino UNO**. The `pXData` elements can definitely be optimized. I would also like to address the FLASH consumption of error messages in `DEBUG_ERR` mode (#31). I may also convert many of the `enums` to defines w/ `uint8_t` as I believe they default to `uint32_t`.

As for the **STM32** -- I have previously done some testing with the **Feather M0** / **Cortex-M0** (should be similar to **STM32 F0**), but it would be great if you end up being able to explore additional device configurations later.

ImpulseAdventure, 2018-01-30:
I think I am close to merging this branch (for version **0.10.0**) into the master. Almost all of the `arduino_min` examples run on the baseline Arduino, though there is still some further optimization left to do. @ALittleSlow - in case you're interested in testing it out, clone the latest in the `API_ElemRefPtr` branch (eg. f5722e6). If you need any assistance, just send me an email.

ALittleSlow, 2018-01-31:
I've been tracking the thread with interest. Can't say yet when I'll be
able to test...

-----
#sutherlandspring <https://twitter.com/hashtag/sutherlandspring>

ImpulseAdventure, 2018-02-04:
Closing feature request as it has now been merged into the master branch. Please feel free to reply to this discussion if there are general concerns with the strategy otherwise file a new ticket for specific issues.

closed 2018-02-04 by ImpulseAdventure

nanokatz, 2018-02-27:
@ImpulseAdventure I'm finally back from vacation and with hardware in my hands. It looks like all the changes are working for me and pretty well.  The only issue I am having are to due with the sliders (have to access the extra data and modify it somehow) and changes I made that were never meant to be shared (I have sliders with +/- buttons as well as labels for example). The memory savings are pretty much as expected as slider elements are no longer in memory.

Also I don't think I should mention any plans I have for more changes to make....you always beat me too it!