ImpulseAdventure/GUIslice #10: ESP8266

universam1, 2017-07-17:
Amazing work! 

As I'm completelly on the ESP8266 I could not determine how far this would be possible to run on this Arduino board?
Afaik the best library which is also Adafruit GFX compatible is https://github.com/Bodmer/TFT_eSPI
Would it  be possible to extend in this regard?

Thank you!

ImpulseAdventure, 2017-07-17:
Hi and thanks!
Yes, I expect I should be able to get the library working with the ESP8266.
I am waiting for an ESP8266 board to arrive and plan to adapt for it right away.

universam1, 2017-07-17:
Thank you so much!
While digging through documentation I realize it might be ESP typical to use the SPIFFS filesystem instead of an SDCard.
Could this be added as a driver as well?

ImpulseAdventure, 2017-07-18:
Thanks for the pointer. Once I have the core graphics operational I will look into the file system.

ImpulseAdventure, 2017-07-30:
Good news -- I have confirmed that GUIslice runs on the **ESP8266**!

I tested an **ESP-12E (NodeMCU 1.0)** connected to an **ILI9341** with a few GUIslice examples and they ran without any changes to the GUIslice code. I only modified the pinout selection in `GUIslice_config.h` to match my board. I have not yet tested touchscreen drivers or SD operation. Once those are tested I could take a look at SPIFFS.

So, you could try compiling GUIslice in the Arduino IDE with the ESP8266 board selected and let me know how it goes. Perhaps start with an example such as `gslc_ex04_ard.ino`.

***NOTE1***: Compiling on the ESP8266 via Arduino IDE just followed the usual process of adding the board package (`http://arduino.esp8266.com/stable/package_esp8266com_index.json`) and installing "esp8266".

***NOTE2***: In my case, I used the following main changes to `GUIslice_config.h`, but you may want to adjust for your own wiring and display:
~~~~
#define DRV_DISP_ADAGFX
#define DRV_TOUCH_NONE
#define DRV_DISP_ADAGFX_ILI9341
#define ADAGFX_PIN_CS    D2
#define ADAGFX_PIN_DC    D1
#define ADAGFX_PIN_RST   D4
#define ADAGFX_PIN_SDCS  D3
~~~~

Thanks and let me know how it goes!

universam1, 2017-07-31:
Thank you @ImpulseAdventure for the efforts, that sounds great!

I have to use the great library https://github.com/Bodmer/TFT_eSPI because I am using the 4" ili9486 display which is reliant on this library. 
For to make that work it seams I have to do a lot of tweaks to your library, are you in favor of adding this support?

ImpulseAdventure, 2017-08-14:
Definitely! A quick update for you:

With only minimal changes I have now successfully run GUIslice on the ESP8266 (ESP-12E) using the `TFT_eSPI` library (instead of `Adafruit-GFX`) on a generic 1.44" TFT display (which uses a `ST7735` display chip). Next, I will try the same thing with Bodmer's `ILI9341` display driver on a larger TFT.

This involved minor modifications in creating a new `GUIslice_drv_tft_espi` driver file which was very similar to the Adafruit-GFX driver.

Thanks for pointing me towards Bodmer's library!

ImpulseAdventure, 2017-09-11:
Integration of the `TFT_eSPI` driver layer into GUIslice has now been completed and rolled into the GUIslice repository. Let me know if it works with your displays. Note that I haven't tested touch support yet, so some minor changes may be upcoming once I get it wired up for testing.

In `GUIslice_config.h`, we can select the library by uncommenting this line:
`#define DRV_DISP_TFT_ESPI`
As per TFT_eSPI documentation, one needs to configure the PIN definitions in the library's `User_Setup*.h` file.

Following the touch testing, I'll look into the SD/SPIFF.

ImpulseAdventure, 2017-10-08:
@universam1 - Did you have an opportunity to see if the updated libraries work for your setup?

Thanks!

closed 2017-10-13 by ImpulseAdventure

rtwfroody, 2017-10-29:
I'm experimenting with using the Bodmer/TFT_eSPI library in my project, which consists of an Adafruit Feather HUZZAH 8266 with the 2.4" TFT touch screen. Graphics display works great (fast!) with this setup (using `#define DRV_DISP_TFT_ESPI`), but I can't make the touch screen part work. `#define DRV_TOUCH_NONE` compiles and that's how I've been testing. I'd like to use `#define DRV_TOUCH_ADA_STMPE610` but that code is deactivated in GUIslice_drv_adagfx.cpp by a check for `DRV_DISP_ADAGFX` early in that file. I may try to separate that out later, but I wonder if anybody has already done that.

ImpulseAdventure, 2017-10-30:
Tim - I have migrated your comment to a new issue #18.