ImpulseAdventure/GUIslice #14: Crash on Feather Huzzah ESP8266

xaviex, 2017-09-22:
After configuration, when running example 7, the device crashes with a fatal exception(28) on gslc_Update being called

ImpulseAdventure, 2017-09-25:
Hi - thanks for pointing this out.

I have confirmed that the issue exists on the ESP8266 with the optional use of `PROGMEM` (FLASH optimization) in the Button instances in some of the examples. Disabling the `PROGMEM` usage should avoid the problem until I have identified how the ESP core is handling `PROGMEM` differently from Arduino and uploaded a new release.

In the meantime before I update the examples, here are some replacements that you could test out: (note that this applies to all instances of `gslc_ElemCreateBtnTxt()`.
```
  static const char mstr4[] PROGMEM = "Kitchen...";  
  pElem = gslc_ElemCreateBtnTxt(&m_gui,GSLC_ID_AUTO,E_PG_MAIN,
    (gslc_tsRect){140,65,80,20},(char*)mstr4,strlen_P(mstr4),E_FONT_TXT,NULL);
  gslc_ElemSetTxtMem(pElem,GSLC_TXT_MEM_PROG);        
  gslc_ElemSetCol(pElem,GSLC_COL_GRAY_DK2,GSLC_COL_GRAY_DK3,GSLC_COL_BLUE_DK1);    
  gslc_ElemSetTxtCol(pElem,GSLC_COL_WHITE);
```
becomes
```
  static const char mstr4[] = "Kitchen...";  
  pElem = gslc_ElemCreateBtnTxt(&m_gui,GSLC_ID_AUTO,E_PG_MAIN,
    (gslc_tsRect){140,65,80,20},(char*)mstr4,strlen(mstr4),E_FONT_TXT,NULL);
  //gslc_ElemSetTxtMem(pElem,GSLC_TXT_MEM_PROG);        
  gslc_ElemSetCol(pElem,GSLC_COL_GRAY_DK2,GSLC_COL_GRAY_DK3,GSLC_COL_BLUE_DK1);    
  gslc_ElemSetTxtCol(pElem,GSLC_COL_WHITE);
```
In general, you'll see I removed the `PROGMEM` keyword on the static string assignment, changed `strlen_P()` to `strlen()` and also commented out the `gslc_ElemSetTxtMem()` call.

xaviex, 2017-09-25:
Hm, its still crashing when i remove and switch around those calls, anything else that needs to be changed in 7 specifically to get it running w/o crash?

ImpulseAdventure, 2017-09-26:
Ok I have created a simplified version that removes all FLASH optimization.

[gslc_ex07_esp.ino (beta test)](https://gist.github.com/ImpulseAdventure/c607955a040c7743bd7c220ebed0a2cd)

Please give this a try and let me know if your results are any different. On my local testing with an ESP-12E, I no longer see any exceptions.

Thanks!

ImpulseAdventure, 2017-09-30:
@xaviex - if you find that the revised code (in the above linked gist) works better I will create a full set of examples for the ESP8266. Thanks!

xaviex, 2017-09-30:
@ImpulseAdventure, This works far better! (And by that i mean it doenst crash which is fantastic) Thank you for your help!

ImpulseAdventure, 2017-10-05:
Arduino examples under `/arduino` have now been rewritten (see https://github.com/ImpulseAdventure/GUIslice/commit/d88d0d5ac1c3fce0d3ec469dd86f272f32651cac) to use a simplified memory mode that should work better on ESP8266 devices. Let me know how it goes!

I will be moving the memory-optimized versions of the examples into a separate set under `/arduino_min`.

closed 2017-10-11 by ImpulseAdventure