ImpulseAdventure/GUIslice #15: Library needs a library.properties file in root to be seen by Arduino IDE

Bodmer, 2017-10-20:
Adding the attached to the root folder means the Arduino ID recognises the
[library.properties.txt](https://github.com/ImpulseAdventure/GUIslice/files/1403291/library.properties.txt)

 library.

Attached as a text file so delete the .txt in the file name.

Bodmer, 2017-10-20:
I have it working very well on an ESP8266 using the TFT_eSPI library and a serial SPI Waveshare TFT. But I cannot figure out how to change the code to support an XPT2046 touch controller.

ImpulseAdventure, 2017-10-21:
Thanks very much @Bodmer for pointing this out. Hopefully this update addresses the IDE visibility.
Glad to hear the library seems to be working well.
Regarding the XPT2046, I will open a separate feature request issue for it and close this one.

closed 2017-10-21 by ImpulseAdventure