ImpulseAdventure/GUIslice #16: Add support for XPT2046 touch driver

ImpulseAdventure, 2017-10-21:
As pointed out by @bodmer in #15, GUIslice doesn't currently support the `XPT2046` resistive touch controller.

I understand that the [TFT_eSPI](https://github.com/Bodmer/TFT_eSPI) library already provides support directly, so I intend to start with that as a driver. Later, I may look at supporting PaulStoffregen's [XPT2046_Touchscreen](https://github.com/PaulStoffregen/XPT2046_Touchscreen) library for those who aren't using TFT_eSPI.

ImpulseAdventure, 2017-10-21:
@Bodmer -- I have created a very quick attempt at adding support for your integrated touch driver.

Please try the following replacement files (based on latest GUIslice repo https://github.com/ImpulseAdventure/GUIslice/commit/3e27039827be7a262d54e3545b8179fff266c5a1 )
https://gist.github.com/ImpulseAdventure/f2ba19273fae7dc44044b3c60a97b7be

In the `GUIslice_config.h`, you'll see that I added a `#define DRV_TOUCH_TFT_ESPI`.
Assuming the config file has been modified (as in the past to work with TFT_eSPI), ensure that `DRV_TOUCH_TFT_ESPI` is the only DRV_TOUCH_* define uncommented.

Note that I'm not able to test support for XPT2046 as I don't have a display with this resistive controller. But if you are able to give it a try, that would be great. Also note that the gist version is completely untested (I haven't tried compiling it on the ESP8266 yet :)

Bodmer, 2017-10-21:
Wow, that was quick thanks. I hope to have time to test it later today. Thanks very much.

ImpulseAdventure, 2017-10-21:
FYI - The gist has been updated to correct a few typos. Should be close to what we need now.

Bodmer, 2017-10-23:
I have been working on improving the touch sensing reliability, so now looking at GUIslice is next on my list.

ImpulseAdventure, 2017-10-30:
Proposed changes from the gist (to add TFT_eSPI's integrated touch handling support) have been rolled in to master now.

ImpulseAdventure, 2017-11-04:
Hi @Bodmer - I was wondering if you might have had an opportunity to see if the revised code might work with your XPT2046 integrated touch handler? Thanks!

Bodmer, 2017-11-04:
Hi, Yes I tried it but so many error messages that I was not sure where to start trying to identify what was going wrong. I will have another look this weekend.

Bodmer, 2017-11-04:
I noticed the changes have been merged into the master so I have taken a copy of that and it compiles fine (I had lots of errors with the GIST but that may be my fault).

The XPT2046 touch screen handler needs some calibration values in the form:

    uint16_t calData[5] = { 321, 3498, 280, 3593, 3 };
    m_disp.setTouch(calData);

So I added that to gslc_DrvInit and the examples are responding to screen touches. Some examples do not respond as I expect, the radio and check box outlines are highlighted on a touch but the graphic does not change. Sliders work fine.

ImpulseAdventure, 2017-11-05:
Great --  thank you for checking this and confirming that the display is generally working.
I will add your calibration step in a further update.

As for the touch anomalies you are noticing, I don't seem to be observing those on the Adafruit `STMPE610` touch display I'm testing on. The fact that you see the elements highlighting (glow state) but not moving to a selected state suggests to me that either:
1) the library may be seeing noisy touch coordinates (ie. initial touch-down inside element but touch-up is detected outside the element meaning it won't be selected) or
2) the range in pressure reading returned from the XPT2046 doesn't return to 0

To confirm this, you can uncomment `#define DBG_TOUCH` in the `GUIslice_config.h` file which will enable some visual tracing of the touch coordinates received from the `XPT2046` driver. Perhaps that might help identify if it there is consistency in the `getTouch()` call responses. In addition, the `DBG_TOUCH` mode will report touch state transitions to the Serial monitor.

The `GUIslice.c` `gslc_TrackTouch()` function currently expects the touch pressure to go to `0` for an "up" event. If your display touch driver leaves some residual non-zero pressure reading after a press, then it could explain why you aren't observing the transition to a selected state. The `DBG_TOUCH` serial output will report the pressure readings (do they go to zero after a touch? Do you see a `TouchUp` event reported?) If not, then we could try changing the `==0` & `>0` comparisons in `gslc_TrackTouch()` to allow some residual readings to be detected as "up". (eg. change to `<10` & `>=10` respectively). Note that DBG_TOUCH may interfere slightly with certain refresh operations.

`gslc_ex04_ard.ino` is probably the best example to use for this testing.

PS> If you find it easier to communicate over email, please try me at guislice @ impulseadventure.com.

Thanks again for your time

ImpulseAdventure, 2017-11-12:
@Bodmer -- in your XPT2046 development, do you happen to recall approximately what range in value you observed from `getTouch()` (ie. during and after releasing a touch press). thx

Bodmer, 2017-11-13:
The getTouch() function reads the XPT2046 raw data values for each axis, these tend to be in the range 300 to 3600. See [Default calibration ](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.h#L467)values.

These values are scaled into screen pixel coordinates x,y. If the calibration values are correct then the top left of the screen = 0,0 and the bottom right = width,height.

ImpulseAdventure, 2017-11-14:
Thanks @Bodmer for the calibration data point -- that would appear to explain the observations per my previous entry.

On that basis, I'll add a configurable touch threshold to the TFT_eSPI integration similar to what is done internally within the FT6206 (I understand that the controller clips pressure to zero below the calibration threshold).

Bodmer, 2017-11-15:
The library also uses touch pressure ( Z ) with no touch you get a noisy value of about 15, the library only responds to a touch when the value is above a certain [threshold](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.cpp#L3780). The cheap resistance based screens are less sensitive to touch at the corners so the threshold value is set lower for this reason.

ImpulseAdventure, 2017-11-25:
Just a brief followup -- after revisiting the TFT_eSPI library I realized that `getTouch()` is actually defined to return `false` upon touch release, so no additional thresholding would be necessary in GUIslice. Therefore, the library should be seeing the correct post-filtered presses OK thanks to your `validTouch()` check.

As referenced earlier, running with `#define DBG_TOUCH` may help reveal whether the issues you were noting (with some controls not updating) was related to the touch capture or some other library issue. Unfortunately, I have been unable to recreate your observations with the ESP8266 & touch displays that I have access to. I will leave this open for a while in case anyone else runs into this issue or I manage to find it in my continued testing. Thanks!

ImpulseAdventure, 2018-01-02:
I haven't been able to recreate the original observations involving specific controls not updating. A fair amount has changed since the initial TFT_eSPI driver integration, so it would be good to re-check the behavior with the latest release at some point (though unfortunately I don't have a display with `XPT2046` to test with at the moment).

In commit b5eae4a I added your default calibration data into the `GUIslice_config.h` file under `TFT_ESPI_TOUCH_CALIB` and called TFT_eSPI `setTouch()` in `gslc_DrvInitTouch()`. This has not been tested but is based on your recommendation.

I will close the ticket for now, but will be happy to recreate a new issue regarding the control behavior should you continue to observe it again. Thanks!

closed 2018-01-02 by ImpulseAdventure