ImpulseAdventure/GUIslice #18: ESP8266 touch support for STMPE610

ImpulseAdventure, 2017-10-30:
Migrating @rtwfroody comment from #10:

> I'm experimenting with using the Bodmer/TFT_eSPI library in my project, which consists of an Adafruit Feather HUZZAH 8266 with the 2.4" TFT touch screen. Graphics display works great (fast!) with this setup (using `#define DRV_DISP_TFT_ESPI`), but I can't make the touch screen part work. `#define DRV_TOUCH_NONE` compiles and that's how I've been testing. I'd like to use `#define DRV_TOUCH_ADA_STMPE610` but that code is deactivated in GUIslice_drv_adagfx.cpp by a check for `DRV_DISP_ADAGFX` early in that file. I may try to separate that out later, but I wonder if anybody has already done that.

Hi Tim! Yes, I can definitely add support for the `Bodmer/TFT_eSPI` mode to use `Adafruit STMPE610` & `FT6206` drivers for touch handling. I had started with the TFT_eSPI's integrated touch handling mode, but these other external drivers will be useful too. Your `GUIslice_config.h` file will just need to define `DRV_DISP_TFT_ESPI` and `DRV_TOUCH_ADA_STMPE610`. The only thing missing in the library should be the additional includes at the top of `GUIslice_drv_tft_espi.cpp`. I will be posting an update shortly for you to test.

ImpulseAdventure, 2017-10-30:
Tim - changes have been made but not tested on the ESP8266 yet. GUIslice should now support both STMPE610 and FT6206 touch drivers in the TFT_eSPI display mode. Let me know if you find an opportunity to give it a try. Thanks!

rtwfroody, 2017-10-30:
I'm excited to try this. Hopefully I'll have time tonight.

rtwfroody, 2017-10-30:
It's not working for me. I can use:
```
#define DRV_DISP_TFT_ESPI          // Arduino: Bodmer/TFT_eSPI library
#define DRV_TOUCH_NONE          // No touchscreen support
```
and observe that the screen is drawn much faster than with the Adafruit library.

If I switch to:
```
#define DRV_DISP_TFT_ESPI          // Arduino: Bodmer/TFT_eSPI library
#define DRV_TOUCH_ADA_STMPE610  // Arduino: Use Adafruit STMPE610 touch driver
```
then it resets several times. (I can tell by some WiFi debugging I have going on.) Eventually it seems to work, but now clearing a widget (`gslc_DrawFillRect(pGui, pElem->rElem, GSLC_COL_BLACK);`) is really slow. Switching pages is also really slow.

I assume the initial resets are because of accessing uninitialized memory. (Could be my bug or yours.)

I assume the very slow drawing might be due to the touch code changing the SPI speed from whatever the display code set it to.

This works fine:
```
#define DRV_DISP_ADAGFX            // Arduino: Adafruit-GFX library
#define DRV_TOUCH_ADA_STMPE610  // Arduino: Use Adafruit STMPE610 touch driver
```

ImpulseAdventure, 2017-10-31:
Thanks Tim for giving this a try! I now understand why you were mentioning `DRV_DISP_ADAGFX`. I don't have a good setup to test GUIslice using TFT_eSPI on the ESP8266 with a STMPE610 -- I might see if I can cobble together a 2.8" TFT shield to test but may not get to it for a little bit.

In the meantime, have you managed to use the TFT_eSPI library's example code successfully with the Adafruit STMPE610 driver? It may be simplest for us to rule out any conflicts there before we add in the GUIslice layer.

@bodmer: are you aware of any challenges we may face when running TFT_eSPI with the Adafruit STMPE610 touch driver?

Thanks!

Bodmer, 2017-10-31:
Hi, try switching on [transaction support](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L184) in the TFT_eSPI library setup file by removing //.

ImpulseAdventure, 2017-11-01:
Thanks @Bodmer, that appears to be it!

I recreated the issue in a Adafruit 2.8" TFT touchscreen with ESP-12E using TFT_eSPI + Adafruit STMPE610. After making the change to uncomment `#define SUPPORT_TRANSACTIONS` in TFT_eSPI's `User_setup.h`, I no longer observed the performance impact.

Also, @rtwfroody, there were some memory access issues in the simplified Arduino examples that I have now fixed in https://github.com/ImpulseAdventure/GUIslice/commit/30fb3f8f747a813c8a0661c6b0b7b471947ac8f0 so you may want to update your code if you were based upon those.

Hope that helps!

ImpulseAdventure, 2018-01-02:
With the changes including the recommendation for `SUPPORT_TRANSACTIONS` in TFT_eSPI and an adjustment to the main loop touch polling rate (commit 084f5b5), I believe a number of the original issues may have been addressed.

I have been successfully testing ESP-12E / ESP8266 with an ILI9341 display and STMPE610 touch controller. @rtwfroody -- please feel free to open a new issue if you still observe outstanding problems with the latest release.

Thanks!

closed 2018-01-02 by ImpulseAdventure