ImpulseAdventure/GUIslice #20: Add support for new fonts from Adafruit-GFX

ImpulseAdventure, 2017-11-25:
Since the recent Adafruit-GFX library updates have added support for alternate fonts via `setFont()`, it would be good to incorporate this into GUIslice.

ALittleSlow, 2017-11-25:
Awesome! Looking forward to a trial run.

#VIstrong

ImpulseAdventure, 2017-12-11:
Initial changes to support Adafruit-GFX alternate fonts is now ready for testing on Arduino targets. Still have to complete changes for TFT_eSPI. If anyone else is interested in beta testing this feature, please let me know.

Current implementation involves a breaking change (for Arduino users) in the `gslc_FontAdd()` call. Instead of passing in an empty string (used in SDL mode for the font filename), users can either supply `NULL` for the default font or else a pointer to the imported font structure. For example, revised code:
```
#include "Fonts/FreeSansBold12pt7b.h"
gslc_FontAdd(&m_gui,E_FONT_TXT,NULL,1); // Default font
gslc_FontAdd(&m_gui,E_FONT_BTN,&FreeSansBold12pt7b,1); // Custom font for buttons
```

ImpulseAdventure, 2017-12-15:
Alternate font support has been added in latest commit.

ImpulseAdventure, 2017-12-15:
Example usage for alternate fonts can be found in `arduino/gslc_ex04_ard/gslc_ex04_ard.ino`:
- Uncomment `#define USE_EXTRA_FONTS`
- This example simply changes the font used for the Quit button

Reopening issue to address descenders:
- As the alternate fonts often provide support for descenders below the baseline (such as letter 'y'), the current text centering (on vertical axis) may appear off in some cases. This will be addressed in a further update.

ImpulseAdventure, 2018-01-02:
Latest commit d14dd25 enables TFT_eSPI to have full control over text positioning (via TFT_eSPI sizing and datum logic). This mode is selected automatically by the `DRV_OVERRIDE_TXT_ALIGN` definition in `GUIslice_drv_tft_espi.h`.

closed 2018-01-02 by ImpulseAdventure