ImpulseAdventure/GUIslice #24: Support for element visibility

ImpulseAdventure, 2018-01-07:
It would be useful to support dynamic displays that show / hide GUI elements on the basis of the current state (e.g. A button that is only visible when a radio button / checkbox is set). This may just involve adding a ElemSetVisibleEn() and then integrating this with the existing redraw functionality. The visibility status could potentially be stored by expanding tsElem:bValid into a multi-bit state field.

rrroonn, 2019-01-09:
Definitely something that should be added.  The ability to Hide() or Show() is needed, along with cascading redraws of affected regions.
Thank you.

rrroonn, 2019-01-12:
> Agreed. I will move this feature into the shortlist for implementation as
> it should be relatively straightforward and potentially quite useful. Note
> that I may initially implement this with a full-display invalidation (I
> don’t expect show/hide to be done frequently), but clearly it would be
> desirable to support an optimized invalidated region. There are other
> features that could benefit from this optimized invalidation, so it would
> be worthwhile.

maybe just fill with screen color if hiding a visible window - a good first
step ...

personally, I dont expect a full windows system on small micros.  I mainly
would want to show/hide windows in a simple manner based on state:

EG  if a system is disarmed, make the ARM-HOME and ARM-AWAY button visible
otherwise make the DISARM button visible.  things like that.

I can create the page with three fields invisible.  At run time, I can
decide which to show/hide.  Other solutions (other than hide/show) are
possible, but a lot less intuitive.
thanks,

Ron

ImpulseAdventure, 2019-01-16:
@rrroonn -- I have now integrated a preliminary implementation of the element visibility feature into the [WIP24-Visibility](https://github.com/ImpulseAdventure/GUIslice/tree/WIP24-Visibility) branch 
- Users can call `ElemSetShow()` to hide or show an element dynamically
- Example **ex04** has been modified to demonstrate the callbacks from the checkbox (via #74 ) used in order to hide/show the progress bar
- Note that this initial implementation triggers a page redraw upon hiding an element (typically slow on Arduino, not very noticeable on RPi). Upcoming enhancements to this feature will incorporate a selective / optimized redraw during element hiding (ie. determining which elements need to be redrawn with selective clipping once they are revealed).

ImpulseAdventure, 2019-02-22:
Visibility feature has now been integrated into master via #117
- Partial redraw was implemented using clipping regions. A more advanced redraw by determining element coverage within the clipping region could be be implemented that would reduce some overdraw.
- Note that final API naming is `ElemSetVisible()`
- Thanks again, @rrroonn for the input.

closed 2019-02-22 by ImpulseAdventure