ImpulseAdventure/GUIslice #42: Polish words pl_PL

eloaders, 2018-04-02:
Does GUIslice support Polish characters?

ą, ć, ę, ł, ń, ó, ś, ź, ż

ImpulseAdventure, 2018-04-05:
Hi - provided you have a TrueType font that offers the characters you need, I understand you should be able to convert it using the Adafruit-GFX [fontconvert utility](https://learn.adafruit.com/adafruit-gfx-graphics-library/using-fonts#adding-new-fonts) and then load it just like any other extended font in GUIslice: (see example `gslc_ex04_ard.ino`)

```
#include <Fonts/myfont12pt7b.h>
...
gslc_FontAdd(&m_gui,E_FONT_TXT,GSLC_FONTREF_PTR,&MyFont12pt7b,1);
...
gslc_ElemCreateTxt(&m_gui,GSLC_ID_AUTO,E_PG_MAIN,(gslc_tsRect){20,60,50,10},
(char*)"cześć",0,E_FONT_TXT);
```
(Note that I haven't tried using fontconvert or foreign characters yet but I believe you may need to ensure that the special characters appear in the first 128 characters of your TTF font)

eloaders, 2018-04-07:
SDL mode only supports file-based fonts.

eloaders, 2018-04-11:
Raspberry Pi Zero. SDL1.2. Raspbian Stretch.

ImpulseAdventure, 2018-05-02:
Good news... GUIslice now supports foreign characters via UTF-8. This is enabled on a Raspberry Pi (in SDL mode) with the new command `ElemSetTxtEnc()`. Here is some sample code showing Polish characters:

```c++
gslc_FontAdd(&m_gui,E_FONT_EXTRA,GSLC_FONTREF_FNAME,"Amiko-Regular.ttf",36);

pElemRef = gslc_ElemCreateTxt(&m_gui,GSLC_ID_AUTO,E_PG_MAIN,(gslc_tsRect){40,110,240,60},
    "cześć",0,E_FONT_EXTRA);
gslc_ElemSetTxtEnc(&m_gui,pElemRef,GSLC_TXT_ENC_UTF8);
```

Sample screenshot:
[image]

ImpulseAdventure, 2018-05-04:
@eloaders - the extended language support should be available now. I will close the issue as implemented but please feel free to reply to indicate if it works for you or if you encountered any difficulties. Thanks again for the suggestion.

eloaders, 2018-05-05:
It works just like I expected. Thank you very much.

eloaders, 2018-05-05:
The textbox does not work.

`  // Create textbox
  pElemRef = gslc_ElemXTextboxCreate(&m_gui,E_ELEM_TEXTBOX,E_PG_MAIN,&m_sTextbox,(gslc_tsRect){644,20,375,220},E_FONT_TEMP,(char*)&m_acTextboxBuf,TBOX_ROWS,TBOX_COLS);
  gslc_ElemSetTxtEnc(&m_gui,pElemRef,GSLC_TXT_ENC_UTF8);
  gslc_ElemXTextboxWrapSet(&m_gui,pElemRef,false);
  gslc_ElemSetCol(&m_gui,pElemRef,(gslc_tsColor){181,200,245},(gslc_tsColor){59,58,120},GSLC_COL_BLACK);
  gslc_ElemSetTxtCol(&m_gui,pElemRef,GSLC_COL_ORANGE);`

ImpulseAdventure, 2018-05-05:
You're right. The existing `XTextbox` component processes individual characters in order to support embedded color transitions within the textarea, but this character-based handling does not currently support UTF-8.

However, I have now added a new mode of operation (`#define GSLC_FEATURE_XTEXTBOX_EMBED 0`) that uses string rendering which should be much faster. This mode will also permit the use of UTF-8 foreign character sets but not embedded color transitions.

To use the original implementation (with inline / embedded color support), one can configure `#define GSLC_FEATURE_XTEXTBOX_EMBED 1`.

The code has now been implemented and I have successfully tested the Polish font above. Once I have completed some further testing, I will release the update.

ImpulseAdventure, 2018-05-06:
Hi @eloaders -- updates to support foreign characters in the XTextbox have now been added and included in the latest repository.

Please try it again, and ensure you configure `#define GSLC_FEATURE_XTEXTBOX_EMBED 0`.