ImpulseAdventure/GUIslice #45: arduino_stm32: 'class Adafruit_ILI9341_STM' has no member named 'setFont'

NickNifontov, 2018-04-05:
try to compile sample at Stm32.

'class Adafruit_ILI9341_STM' has no member named 'setFont'

ImpulseAdventure, 2018-04-06:
Hi @NickNifontov !

As you may be aware, the upstream **Arduino_STM32** library required some minor changes to accommodate libraries such as **GUIslice**. The good news is that the upstream repo is just completing testing and (https://github.com/rogerclarkmelbourne/Arduino_STM32/pull/461) should be merged any day now. Once that happens, I will immediately update the GUIslice ***master*** and then you could test it out easily. I have already made the changes -- I'm just waiting for the upstream merge.

I have been using GUIslice successfully with an ***STM32F103*** -- which device are you interested in using?

For reference: I believe the version you were testing was (GUIslice branch "arduino_stm32" PR #39), which is based on an alternate patch to the upstream **Arduino_STM32** that is less likely to get merged. I think the error could have been due to using the unpatched version of the upstream **Adafruit_GFX_AS** library (within **Arduino_STM32**).

My recommendation is to wait a couple more days for the upstream update to complete and then everything should work without any patches :)

NickNifontov, 2018-04-06:
It's really good news. I hope all will works 👍  
About 2-3 month i'll tried to use a Master copy of GUIslice but no lucky

My Board is STM32F103C8T6 (Blue Pill, 128kbytes) - http://wiki.stm32duino.com/index.php?title=Blue_Pill
Display ILI9341 - https://www.ebay.com/itm/2-4-SPI-TFT-LCD-Display-2-4-Inch-Touch-Panel-LCD-ILI9341-240x320-5V-3-3V-New/162550483037?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2060353.m2749.l2649

Currently i have directly use Adafruits_GFX_AS + Adafruit_ILI9341_STM + my privimitive code for GUI
For touch i use modified library (it's work :o) ) - 

i'll plan to use GUIslice for touch also ( DRV_TOUCH_TFT_ESPI  ). ok?
[Touch.zip](https://github.com/ImpulseAdventure/GUIslice/files/1883190/Touch.zip)

NickNifontov, 2018-04-06:
TOUCH
DO_MOSI =PB15
DI_MISO=PB14
CS =PB12
CLK_SLK=PB13 
IRQ=PB4

Display
CS=PA9 
DC=PA10
RST=VCC

ImpulseAdventure, 2018-04-15:
@NickNifontov -- I have now merged the changes that use the latest Arduino_STM32 release. Please re-download [Arduino_STM32](https://github.com/rogerclarkmelbourne/Arduino_STM32) and give the latest GUIslice release a try.

ImpulseAdventure, 2018-05-01:
Thanks for the followup @NickNifontov . Closing.

closed 2018-05-01 by ImpulseAdventure