ImpulseAdventure/GUIslice #46: Element ComboBox

eloaders, 2018-04-12:
Is it possible to add combobox to the library?

ImpulseAdventure, 2018-05-05:
@eloaders & @NickNifontov -- yes, this is a good suggestion.
I have some ideas in how a ComboBox can be implemented but please note that it may take some time before I can start on it. thanks!

ImpulseAdventure, 2019-03-06:
Long overdue update! Now that the extended elements have been split out #121, it is much easier to add new element types. That said, I have now begun working on developing the **ListBox** and **ComboBox** elements.
- An `XListBox` element has now been created (currently in development branch testing)
- An `XComboBox` element will be created next. The XComboBox would be modeled after the standard dropdown listbox. This component would probably utilize the compound element mechanism.

ImpulseAdventure, 2019-03-15:
The new XListbox element has been merged into the master, available by including `elem/XListbox.h`

Example screenshot of single-column **XListbox** control with scrollbar integration:
[image]

Sample code to create a 3 column XListbox:
```c++
#include "elem/XListbox.h"

// Allocate storage for listbox element and buffer
gslc_tsXListbox  m_sListbox;
static uint8_t   m_pXListboxBuf[BUF_LEN];

// Create XListbox control
m_pElemListbox = m_pXListboxCreate(pGui, E_LISTBOX, E_PG_MAIN, &m_sListbox, rRect, E_FONT, m_pXListboxBuf, BUF_LEN, 0);
gslc_ElemXListboxSetSize(pGui, m_pElemLIstbox, 5, 3); // 5 Rows, 3 columns

// Add items
gslc_ElemXListboxAddItem(pGui, m_pElemListbox, "Item1");
gslc_ElemXListboxAddItem(pGui, m_pElemListbox, "Item2");
...
```