ImpulseAdventure/GUIslice #48: Problem compiling ESP32 + TFT_eSPI + FT6206

vitormhenrique, 2018-05-22:
Hello,

I tried to find other people that had the same issue but no luck.
I'm trying to use the library with a huzzah32 (ESP32) using the TFT_eSPI library and the FT6206 capacitive library. I have the [2.8" TFT LCD with Cap Touch](https://www.adafruit.com/product/2090).

But I can not get it to compile, it gives me the error:
`lib/GUIslice/src/GUIslice_drv_tft_espi.cpp:822:21: error: 'class TFT_eSPI' has no member named 'getTouch'`

I got the TFT_eSPI working before, so I know that my configuration is OK under the "User_Setup.h"

Even if I try to disable touch on "GUIslice_config_ard.h" the error still persists. It's always trying to include and compile this section of code inside the **GUIslice_drv_tft_espi.cpp**

<pre><code>
/ Confirm that TOUCH_CS has been defined otherwise the
// m_disp.getTouch() call won't be defined in TFT_eSPI
#if defined(DRV_TOUCH_TFT_ESPI)
  #if !defined(TOUCH_CS)
    #error "To use DRV_TOUCH_TFT_ESPI, TOUCH_CS needs to be defined in TFT_eSPI User_Setup"
  #endif
#endif

bool gslc_DrvGetTouch(gslc_tsGui* pGui,int16_t* pnX, int16_t* pnY, uint16_t* pnPress)
{

  if ((pGui == NULL) || (pGui->pvDriver == NULL)) {
    GSLC_DEBUG_PRINT("ERROR: DrvGetTouch(%s) called with NULL ptr\n","");
    return false;
  }

  // Use TFT_eSPI for touch events
  uint8_t     bPressed = 0;
  uint16_t    nX=0;
  uint16_t    nY=0;
  int16_t     nOutputX, nOutputY;

  bPressed = m_disp.getTouch(&nX,&nY);

  // Perform any requested swapping of input axes
  if( pGui->nSwapXY ) {
    nOutputX = (int16_t)nY;
    nOutputY = (int16_t)nX;
  } else {
    nOutputX = (int16_t)nX;
    nOutputY = (int16_t)nY;
  }

  // Perform any requested output axis flipping
  if( pGui->nFlipX ) {
    nOutputX = pGui->nDispW - 1 - nOutputX;
  }
  if( pGui->nFlipY ) {
    nOutputY = pGui->nDispH - 1 - nOutputY;
  }

  // Assign coordinates
  *pnX = (int16_t)nOutputX;
  *pnY = (int16_t)nOutputY;

  if (bPressed > 0) {
    *pnPress = 1;
  } else {
    *pnPress = 0;
  }

  return true;
}</code></pre>

It's very likely that I'm missing something, could you give a hint?

ImpulseAdventure, 2018-05-22:
Hi @vitormhenrique --

In the GUIslice config, I presume you are selecting `DRV_TOUCH_ADA_FT6206` (which matches your capacitive touch driver). Since you are loading an external touch driver (ie. outside **TFT_eSPI**'s built-in **XPT2046** driver), we should probably comment out `#define DRV_TOUCH_IN_DISP` if you haven't already. (Looks like I need to make this clearer in the config file)

If possible, could you attach your `GUIslice_config_ard.h` and `User_Setup.h` files?
- Lastly, could you confirm that you are compiling with the Arduino IDE, and that you are using the latest TFT_eSPI (eg. version 0.18.21)?

Thanks!

robertknorton, 2018-08-07:
@vitormhenrique I am currently attempting to get an ESP32 working with an Adafruit 3.5 inch TFT display. Would you be willing to share your configuration and .h files? My code seems to compile but not display properly.

robertknorton, 2018-08-07:
Thank you so much. I appreciate the files for reference. Once I test things out and get more to report I'll let you know how they were able to help me.

ImpulseAdventure, 2018-08-08:
Hi @robertknorton -- can you let me know a little more about your setup?

I understand you are using an **ESP32** with the **Adafruit 3.5" TFT**.
- Is your display the "touchscreen breakout board" with resistive touch? (Adafruit *PID: 2050*)
- Are you using the `DRV_DISP_TFT_ESPI` mode in `GUIslice_config_ard.h`?
- Can you confirm that the TFT_eSPI library examples are working for you (both display and touch)? You should find these in the TFT_eSPI library under:
- - `examples/480x320/Keypad_480x320/Keypad_480x320.ino`
- - `examples/480x320/TFT_graphicstest_one_lib/TFT_graphicstest_one_lib.ino`
- If the TFT_eSPI examples are running fine (display and touch), what do you see when you run the GUIslice examples? (eg. using @vitormhenrique config, perhaps modified as shown below)

Note: Your 3.5" display may require some changes from Vitor's config since I think you are using a different display driver and touch driver than the 2.8" setup he has. In particular, I think you'd want:

`User_Setup.h`:
- Uncomment `#define HX8357D_DRIVER`
- Comment `#define ILI9341_DRIVER`

`GUIslice_config_ard.h`:
- Uncomment `#define DRV_TOUCH_ADA_SIMPLE`
- Comment `#define DRV_TOUCH_ADA_FT6206`

Thanks

robertknorton, 2018-08-13:
Hello @ImpulseAdventure 

I tried messing around with the TFT_eSPI library examples and they are failing to compile.

I have made the recommended changes except for the touchscreen. I am currently not using touch, just button controls. 

Here is some code from my current GUIslice_config_ard.h

// =============================================================================
// DISPLAY CONFIGURATION - SELECT
// =============================================================================

  // Specify the graphics driver library
  // - Uncomment one of the following graphics drivers DRV_DISP_*
  //   applicable to the device type in use

  // --------------------------------------------------------------
  // Arduino / ATmega / AVR / Cortex-M0 / nRF52:
  //  #define DRV_DISP_ADAGFX           // Adafruit-GFX library
  // --------------------------------------------------------------
  // ESP8266 / ESP32 / NodeMCU:
  //  #define DRV_DISP_ADAGFX           // Adafruit-GFX library
    #define DRV_DISP_TFT_ESPI         // Bodmer/TFT_eSPI library
  //  #define DRV_DISP_M5STACK          // m5stack/M5Stack library
  // --------------------------------------------------------------
  // STM32:
  //  #define DRV_DISP_ADAGFX           // Adafruit-GFX library
  //  #define DRV_DISP_ADAGFX_AS        // Adafruit-GFX-AS library
  // --------------------------------------------------------------

// =============================================================================
// TOUCH CONFIGURATION - SELECT
// =============================================================================

  // Specify the touchscreen driver
  // - Uncomment one of the following touchscreen drivers DRV_TOUCH_*
  //   applicable to the controller chip in use

    #define DRV_TOUCH_NONE            // No touchscreen support
  //  #define DRV_TOUCH_ADA_STMPE610    // Adafruit STMPE610 touch driver
  //  #define DRV_TOUCH_ADA_FT6206      // Adafruit FT6206 touch driver
  //  #define DRV_TOUCH_ADA_SIMPLE      // Adafruit Touchscreen
  //  #define DRV_TOUCH_TFT_ESPI        // TFT_eSPI integrated XPT2046 touch driver

// =============================================================================
// DISPLAY CONFIGURATION - DETAILS
// - Graphics display driver-specific additional configuration
// =============================================================================

// -----------------------------------------------------------------------------
#if defined(DRV_DISP_ADAGFX)

  // The Adafruit-GFX library supports a number of displays
  // - Select a display sub-type by uncommenting one of the
  //   following DRV_DISP_ADAGFX_* lines
  //#define DRV_DISP_ADAGFX_ILI9341         // Adafruit ILI9341
  //#define DRV_DISP_ADAGFX_ILI9341_8BIT  // Adafruit ILI9341 (8-bit interface)
  //#define DRV_DISP_ADAGFX_ST7735        // Adafruit ST7735
  //#define DRV_DISP_ADAGFX_SSD1306       // Adafruit SSD1306
  #define DRV_DISP_ADAGFX_HX8357        // Adafruit HX8357
  //#define DRV_DISP_ADAGFX_PCD8544       // Adafruit PCD8544

#elif defined(DRV_DISP_ADAGFX_AS)

  // The Adafruit-GFX-AS library supports a number of displays
  // - Select a display sub-type by uncommenting one of the
  //   following DRV_DISP_ADAGFX_* lines
  //#define DRV_DISP_ADAGFX_ILI9341_STM     // Adafruit ILI9341 (STM32 version)

#endif

#if defined(DRV_DISP_ADAGFX) || defined(DRV_DISP_ADAGFX_AS)
  // For Adafruit-GFX drivers, define pin connections
  // - Define general pins (modify these example pin assignments to match your board)
  // - Please refer to "docs/GUIslice_config_guide.xlsx" for detailed examples
  #define ADAGFX_PIN_CS    14   // Display chip select
  #define ADAGFX_PIN_DC    15   // Display SPI data/command
  #define ADAGFX_PIN_RST   16   // Display Reset (some displays could use pin 11)
  #define ADAGFX_PIN_SDCS   4   // SD card chip select
  #define ADAGFX_PIN_WR    13   // Display write pin (for parallel displays)
  #define ADAGFX_PIN_RD    12   // Display read pin (for parallel displays)

  // Use hardware SPI interface?
  // - Set to 1 to enable hardware SPI interface, 0 to use software SPI
  // - Software SPI may support the use of custom pin selection (via ADAGFX_PIN_MOSI,
  //   ADAGFX_PIN_MISO, ADAGFX_PIN_CLK). These pin definitions can be left blank in
  //   hardware SPI mode.
  #define ADAGFX_SPI_HW     1

  // Define custom SPI pin connections used in software SPI mode (ADAGFX_SPI_HW=0)
  // - These definitions can be left blank in hardware mode (ADAGFX_SPI_HW=1)
  #define ADAGFX_PIN_MOSI
  #define ADAGFX_PIN_MISO
  #define ADAGFX_PIN_CLK

  // Set Default rotation
  // - Note that if you change this you will likely have to change
  //   ADATOUCH_FLIP_X & ADATOUCH_FLIP_Y as well to ensure that the touch screen
  //   orientation matches the display rotation
  #define GSLC_ROTATE     1

ImpulseAdventure, 2018-08-15:
Hi @robertknorton --

When you indicate that you tried the "*TFT_eSPI library examples and they are failing to compile*", can you confirm:

1) Whether you were using **TFT_eSPI** alone or combining **TFT_eSPI** with **GUIslice**? If combined, could you let me know if the TFT_eSPI standalone examples (eg. `TFT_graphicstest_one_lib.ino`) work?

2) Can you list some of the errors that were reported during compile?

Thanks!

imalipusram, 2018-10-06:
Dear Calvin, 
I'm running into problems using the Arduino IDE compiling the first example (gslc_ex_01_ard) for ESP32 and ILI9431. The code compiles but it seems to me the code from the TFT_eSPI library does not get implemented (The display doesn't do anything at all except the backlight is on):

I added some code to the example to check for the connections which should be established in the eTFT library:

```
#if defined(TFT_MISO)
  Serial.print ("TFT_MISO: " + String(TFT_MISO)  + "\r\n");
#else
  Serial.print("TFT_MISO not defined !!!\r\n");
#endif
```

And the output at runtime shows me that TFT_MISO has not been defined.

Then, I inserted tracers into GUIslice_drv_tft_espi.cpp

```
// Compiler guard for requested driver
#define CHECKPOINT_2  // by ws
#include "GUIslice_config.h" // Sets DRV_DISP_*

#if defined(DRV_DISP_TFT_ESPI)
#define CHECKPOINT_3 // by ws
// check if we get here at all
```
Which have not been defined if I check them in my code

```
#if defined(CHECKPOINT_2)  // coded into GUIslice_drv_tft_eSPI.cpp
  Serial.print("CHECKPOINT_2 GUIslice_drv_tft_eSPI.cpp reached\r\n");
#else
  Serial.print("CHECKPOINT_2 in GUIslice_drv_tft_eSPI.cpp NOT REACHED !!!\r\n");
  #endif

#if defined(CHECKPOINT_3)  // coded into GUIslice_drv_tft_eSPI.cpp
  Serial.print("CHECKPOINT_3 in GUIslice_drv_tft_eSPI.cpp reached\r\n");
#else
  Serial.print("CHECKPOINT_3 in GUIslice_drv_tft_eSPI.cpp NOT REACHED !!!\r\n");
#endif
```
However, DRV_DISP_TFT_ESPI gets defined in GUIslice_config_ard.h, as intended.
GUIslice_drv_tft_eSPI.h is found, as the query for the define of _GUISLICE_DRV_TFT_ESPI_H_ is positive, CHECKPOINT_4 in GUIslice_drv.h is reached:
```
#elif defined(DRV_DISP_TFT_ESPI)
  #include "GUIslice_drv_tft_espi.h"
  #define CHECKPOINT_4  // by wo 20181006
```

To make it more confusing, the compiler reports that TFT_eSPI has been used and the libarary also is compiled
```
Compiling library "TFT_eSPI"
Using previously compiled file: /tmp/arduino_build_522274/libraries/TFT_eSPI/TFT_eSPI.cpp.o
```
Restarting the IDE which is deleting the tmp files doesn't change the outcome.

Do you have any clues what's going on and how to fix this?

If it should matter, I'm running 
Ubuntu 18.04.1 LTS 
gcc (Ubuntu 7.3.0-27ubuntu1~18.04) 7.3.0
arduino-1.8.3

Thanks for your help!

Wolfgang

ImpulseAdventure, 2018-10-08:
Hi Wolfgang -- thanks for the report. Could you confirm a couple items for me?
1) Are you using a M5Stack? If so, you might want to update to the latest version of GUIslice (0.10.3) as I addressed a M5 compilation issue in the past day.
2) When you run the TFT_eSPI's graphics test example (from the Arduino IDE's Examples -> TFT_eSPI -> TFT_grapicstest_one_lib), I assume the display is working properly?
3) When running the gslc_ex01_ard example, if you enable the Serial Monitor and reset, is anything reported? (may need to adjust the Arduino IDE monitor's baud rate to match your board's serial output).

thanks!

imalipusram, 2018-10-08:
Hi Calvin, 

thanks for getting back to me. I'm working with an ESP32 dev board [1].  Meanwhile, I have found it does not generate the clock signal and some other I/Os also seem dead, so I'll have to wait with further checks until a new board arrives.
However, with an NodeMCU (ESP8266), your example code is working well. And I have learned that #define does not work across libraries, so most of the testing I have reported in my previous post, was garbage.

Sorry for the confusion and thanks again!

Wolfgang

[1] https://www.aliexpress.com/item/ESP32-Development-Board-WiFi-Bluetooth-Ultra-Low-Power-Consumption-Dual-Cores-ESP-32-ESP-32S/32799057508.html

ImpulseAdventure, 2018-10-08:
Thanks Wolfgang for the update indicating that it may have been a board issue. Hopefully the replacement works better (and doesn't take too long to arrive!). Good to hear that the NodeMCU/ESP8266 examples ran fine. Let me know if any further help is needed.

imalipusram, 2018-10-13:
Hi Calvin! 

Got a new ESP32 (NodeMCU-32S) and tested the examples (ILI9341 w/o touchscreen, waiting for an ILI9488 with touchscreen to arrive) and did some tests:

Examples 1..4 compile and seem to run ok as far as I can see.

Example 2 has warnings:
```
/home/wo/Arduino/libraries/GUIslice/arduino/gslc_ex02_ard/gslc_ex02_ard.ino: In function 'void setup()':
/home/wo/Arduino/libraries/GUIslice/arduino/gslc_ex02_ard/gslc_ex02_ard.ino:57:19: warning: unused variable 'bOk' [-Wunused-variable]
   bool            bOk = true;
                   ^
```
Examples 1..4 have these warnings:
```
/home/wo/Arduino/libraries/GUIslice/src/GUIslice_ex.c: In function 'gslc_ElemXCheckboxFindChecked':
/home/wo/Arduino/libraries/GUIslice/src/GUIslice_ex.c:792:18: warning: comparison is always true due to limited range of data type [-Wtype-limits]
     if (nCurType != GSLC_TYPEX_CHECKBOX) {
                  ^
/home/wo/Arduino/libraries/GUIslice/src/GUIslice_ex.c: In function 'gslc_ElemXTextboxDraw':
/home/wo/Arduino/libraries/GUIslice/src/GUIslice_ex.c:2187:21: warning: variable 'bEncUtf8' set but not used [-Wunused-but-set-variable]
   bool              bEncUtf8;
                     ^
```
Example 5 yields an error: 
```
gslc_ex05_ard:63: error: 'gslc_tsXSelNum' does not name a type
 gslc_tsXSelNum              m_sXSelNum[3];
 ^
/home/wo/Arduino/libraries/GUIslice/arduino/gslc_ex05_ard/gslc_ex05_ard.ino: In function 'bool InitOverlays()':
gslc_ex05_ard:152: error: 'm_sXSelNum' was not declared in this scope
   pElemRef = gslc_ElemXSelNumCreate(&m_gui,E_ELEM_COMP1,E_PG_MAIN,&m_sXSelNum[0],
                                                                    ^
gslc_ex05_ard:153: error: 'gslc_ElemXSelNumCreate' was not declared in this scope
     (gslc_tsRect){160,60,120,50},E_FONT_BTN);
```

Example 6 yields errors, too:
```
/home/wo/Arduino/libraries/GUIslice/arduino/gslc_ex06_ard/gslc_ex06_ard.ino: In function 'bool CbDrawScanner(void*, void*, gslc_teRedrawType)':
gslc_ex06_ard:118: error: narrowing conversion of '(((int)nX) + -30)' from 'int' to 'int16_t {aka short int}' inside { } [-Werror=narrowing]
   gslc_DrawFrameRect(pGui,(gslc_tsRect){nX-30,nY-20,60,40},GSLC_COL_BLUE_DK2);
                                           ^
gslc_ex06_ard:118: error: narrowing conversion of '(((int)nY) + -20)' from 'int' to 'int16_t {aka short int}' inside { } [-Werror=narrowing]
   gslc_DrawFrameRect(pGui,(gslc_tsRect){nX-30,nY-20,60,40},GSLC_COL_BLUE_DK2);
                                                 ^
gslc_ex06_ard:123: error: narrowing conversion of '(((int)nX) + 1)' from 'int' to 'int16_t {aka short int}' inside { } [-Werror=narrowing]
   gslc_DrawFillRect(pGui,(gslc_tsRect){nX+1,nY+1,10,10},GSLC_COL_RED_DK2);
                                          ^
gslc_ex06_ard:123: error: narrowing conversion of '(((int)nY) + 1)' from 'int' to 'int16_t {aka short int}' inside { } [-Werror=narrowing]
   gslc_DrawFillRect(pGui,(gslc_tsRect){nX+1,nY+1,10,10},GSLC_COL_RED_DK2);
                                               ^
gslc_ex06_ard:124: error: narrowing conversion of '(((int)nX) + 1)' from 'int' to 'int16_t {aka short int}' inside { } [-Werror=narrowing]
   gslc_DrawFillRect(pGui,(gslc_tsRect){nX+1,nY-10,10,10},GSLC_COL_GREEN_DK2);
                                          ^
gslc_ex06_ard:124: error: narrowing conversion of '(((int)nY) + -10)' from 'int' to 'int16_t {aka short int}' inside { } [-Werror=narrowing]
   gslc_DrawFillRect(pGui,(gslc_tsRect){nX+1,nY-10,10,10},GSLC_COL_GREEN_DK2);
                                               ^
gslc_ex06_ard:125: error: narrowing conversion of '(((int)nX) + -10)' from 'int' to 'int16_t {aka short int}' inside { } [-Werror=narrowing]
   gslc_DrawFillRect(pGui,(gslc_tsRect){nX-10,nY+1,10,10},GSLC_COL_BLUE_DK2);
                                          ^
gslc_ex06_ard:125: error: narrowing conversion of '(((int)nY) + 1)' from 'int' to 'int16_t {aka short int}' inside { } [-Werror=narrowing]
   gslc_DrawFillRect(pGui,(gslc_tsRect){nX-10,nY+1,10,10},GSLC_COL_BLUE_DK2);
                                                ^
gslc_ex06_ard:126: error: narrowing conversion of '(((int)nX) + -10)' from 'int' to 'int16_t {aka short int}' inside { } [-Werror=narrowing]
   gslc_DrawFillRect(pGui,(gslc_tsRect){nX-10,nY-10,10,10},GSLC_COL_YELLOW);
                                          ^
gslc_ex06_ard:126: error: narrowing conversion of '(((int)nY) + -10)' from 'int' to 'int16_t {aka short int}' inside { } [-Werror=narrowing]
   gslc_DrawFillRect(pGui,(gslc_tsRect){nX-10,nY-10,10,10},GSLC_COL_YELLOW);
                                                ^
cc1plus: some warnings being treated as errors
```
Example 7 compiles and runs,
Example 10 compiles and runs.

Best regards,
Wolfgang