ImpulseAdventure/GUIslice #49: Waveshare e-ink 2.7

NickNifontov, 2018-05-29:
What about to add support of Waveshare e-ink display 
for example [https://www.aliexpress.com/store/product/264x176-2-7inch-E-Ink-display-HAT-for-Raspberry-Pi-Three-color-SPI-Interface-No-Backlight/407494_32829483976.html?spm=2114.12010612.0.0.77a1556bNme7S9]

for example GxEPD (base class is a subclass of Adafruit_GFX)
support AVR, STM32Duino
[https://github.com/ZinggJM/GxEPD](
https://github.com/ZinggJM/GxEPD)

you can use as driver

?

ImpulseAdventure, 2018-06-06:
Hi Nick -- thanks for the suggestion.

It should be possible to add support for the **Waveshare e-ink display**, though GUIslice's touch functionality wouldn't be needed. Could you confirm which libraries you are using with GxEPD? (I noticed in `GxFont_GFX.h` that it refers to options for `GxFont_GFX_TFT_eSPI.h`, `Adafruit_ftGFX.h` and `U8g2_for_Adafruit_GFX.h`).

For the display you referenced (*2.7" E-Ink RPi 3 color hat*), I presume you are using the `GxGDEW027C44` driver? Note that at this time I don't have an E-Ink display to test with, but could order one -- I would probably select a standalone model so that I could test with STM32 / ESP32, as well as the RPI.

ImpulseAdventure, 2018-07-04:
@NickNifontov - FYI, I had ordered a *2.7" RPi HAT*, and am waiting for it to arrive (can be pretty slow!). BTW, I selected a two-color version as it appears that the full refresh times on the 3-color versions can be close to 15s versus 2s for the equivalent B&W display.

If you could let me know what graphics library you selected for **GxEPD**, that would be great (ie. `GxFont_GFX_TFT_eSPI.h`, `Adafruit_ftGFX.h` and `U8g2_for_Adafruit_GFX.h`).

Thanks!

ImpulseAdventure, 2018-10-20:
Closing as stale request. If others have interest in support for eInk, please reply and I’d be happy to re-open. Note that I’m considering adding touch emulation by physical button/GPIO presses (or keyboard), which may make these non-touch displays more functional in GUIslice.

closed 2018-10-20 by ImpulseAdventure

thzzlan, 2021-10-30:
Hi,
would it still be possible to add support for Waveshare epaper displays, for example with the GxEPD2 library?
I hope it's okay to still reply to this issue.

ToXIc-Dev, 2022-05-21:
> Closing as stale request. If others have interest in support for eInk, please reply and I’d be happy to re-open. Note that I’m considering adding touch emulation (#66) by physical button/GPIO presses (or keyboard), which may make these non-touch displays more functional in GUIslice.

Would be cool to have support for this.

This would be very useful for [this](https://banggood.com/LILYGO-T5-4_7-inch-E-paper-Screen-ESP32-V3-Version-16MB-FLASH-8MB-PSRAM-WIFI-Bluetooth-Display-Module-p-1791808.html) lillygo display with esp32 on the back.

It uses its [own driver/library](https://github.com/Xinyuan-LilyGO/LilyGo-EPD47).