ImpulseAdventure/GUIslice #50: Waveshare 7inch HDMI LCD

balwantbishnoi27, 2018-06-23:
Hii, 
I am using GUIslice to built UI for my Project and i am using Waveshare 7inch HDMI LCD.  
When i tried to built Using MAKEFILE present in linux 
For First case : No Touchscreen enabled it works fine
But when I run Second program i am getting this error
ERROR : DrvInit() error in SDL_Init(): Unable to open /dev/fb1
ERROR : Init() failed

ImpulseAdventure, 2018-06-24:
Hi @balwantbishnoi27 --

Thanks for letting me know about your error message. To help narrow down the cause for you, could you let me know a couple details:
- Could you clarify which two programs you were running? eg. `test-sdl1`, `test-sdl2`, `gslc-ex01`, etc.
- In `GUIslice_config_linux.h`, what option was enabled for `DRV_DISP_*`?
- What option was enabled for `DRV_TOUCH_*`?

thx

balwantbishnoi27, 2018-06-25:
Hi @ImpulseAdventure 

I have tslib and SDL library installed and i done test for both libraries.
TSLIB  :  
  sudo TSLIB_FBDEVICE =/dev/fb1 ts_print  **WORKING**
  sudo  TSLIB_TSDEVICE=/dev/input/touchscreen ts_print  **WORKING** 
  sudo TSLIB_FBDEVICE=/dev/fb1 ts_test  **NOT WORKING**
  sudo TSLIB_TSDEVICE=/dev/input/touchscreen **WORKING**

Now i use **Makefile** to compile example programs 

**Case 1:** 
   #define DRV_DISP_SDL1
  #define  DRV_TOUCH_NONE
Result for gslc-ex01, ..gslc-ex02......gslc-ex11
ERROR : DrvInit() error in SDL_Init(): Unable to open /dev/fb1
ERROR : Init() failed

But for ./test-sdl1 it is working and getting pattern. 

**Case 2:**
 #define DRV_DISP_SDL1
  #define  DRV_TOUCH_TSLIB
Result for gslc-ex01, ..gslc-ex02......gslc-ex11
ERROR : DrvInit() error in SDL_Init(): Unable to open /dev/fb1
ERROR : Init() failed

But for ./test-sdl1 it is working and getting pattern. 

**Case 3:**
 #define DRV_DISP_SDL1
  #define  DRV_TOUCH_SDL
Result for gslc-ex01, ..gslc-ex02......gslc-ex11
ERROR : DrvInit() error in SDL_Init(): Unable to open /dev/fb1
ERROR : Init() failed

But for ./test-sdl1 it is working and getting pattern. 

**Case 4:**
 #define DRV_DISP_SDL1
  #define  DRV_TOUCH_NONE
and I **Commented  UserInitEnv();**
Result for gslc-ex01 : 
A black rectangle box in left hand side . Working fine.

Result for gslc-ex02:
 ERROR: DrvFontAdd(/usr/share/fonts/truetype/droid/DroidSans.tft) failed in TFT_OpenFont failed
Segmentation fault

balwantbishnoi27, 2018-06-27:
Hi @ImpulseAdventure 

First i would like to Thank you for help you provided. 

I am using Raspberry pi with following 
Device : WAVESHARE 7inch HDMI LCD (C). USB Capacitive Touch Screen 1024 x 600 Pixel. Rev2.2
Distributor ID : Raspbian
Description : RASPBIAN GNU/Linux 9.4
Release : 9.4
Codename  : Stretch.

**Result for issue 1**

- **Issue Resolved** when I commented `setenv((char*)"SDL_VIDEODRIVER","fbcon",1);` in `UserInitEnv()` and all code are running fine with following configuration
`#define DRV_DISP_SDL1
#define DRV_TOUCH_SDL`

- I get **NO Output** from following at command prompt
`printenv | egrep SDL_VIDEODRIVER`
`printenv | egrep FRAMEBUFFER`

 

**Result for issue 2**

- **Issue Resolved**. I replace path of FONT_DROID_SANS with another existing font. 

**Issue Noticed**
- When i am using following cofiguration in GUIslice_config_linux.h 
`#define DRV_DISP_SDL1
#define DRV_TOUCH_TSLIB`

I am getting following error : 
ERROR: TsOpen() failed.
ERROR: InitTouch() failed to touch driver init
ERROR: Init() failed

ImpulseAdventure, 2018-06-27:
Glad to hear that the updates worked!

As for `TsOpen()`, it seems possible that the `TSLIB_TSDEVICE` config parameter might not be matching your particular environment setup.

- When you installed **tslib**, was it from sources? (please refer to the steps on my [Installing tslib](https://www.impulseadventure.com/elec/rpi-install-tslib.html) page).
- Could you report the output of the following from the command-line?
`ls -l /dev/input/`

It may be easiest to email me directly at guislice at impulseadventure (.com) so that I can determine how your setup might be different from the typical RPI defaults. For the benefit of others, I could then summarize the updates here later.

thanks!

balwantbishnoi27, 2018-07-03:
Hi, @ImpulseAdventure  
Thanks for your help to resolve issue faced by me for Waveshare Touchscreen Functionality. I would like to thank you for this amazing library you created with touchscreen Functionality for Raspberry Pi. I am loving it man Awesome..

Thanks,