ImpulseAdventure/GUIslice #52: m5stack ILI9341 not name a type

wuyuanyi135, 2018-10-05:
Hi,

I am trying to use this library in a M5Stack board. I have configured the ard.h file to the proper options but the compilation failed because `ILI9341` does not exist.

It seems like M5Stack headers do not provide this class anymore. I replace the definition with the global display object and it worked.
```diff
-  ILI9341 m_disp = ILI9341();
+ #define m_disp m5.Lcd
```
Is this a bug? Probably the upstream library update broke the symbol.

ImpulseAdventure, 2018-10-07:
Thank you, you're right... Looks like I didn't notice the updates to M5Stack. I have made a few changes and released [v0.10.3.](https://github.com/ImpulseAdventure/GUIslice/releases/tag/v0.10.3) Let me know if this works for you (as I am not able to test on M5Stack hardware at the moment, though I have tested the compilation).

wuyuanyi135, 2018-10-10:
Looks good to me. Thank you!

closed 2018-10-10 by wuyuanyi135