ImpulseAdventure/GUIslice #54: Wrong Display Width/Heigth when using Rotation on DRV_DISP_ADAGFX_ILI9341_STM

espHorst, 2018-10-08:
Hi,

I'm using the GUIslice and it works perfectly for DRV_DISP_ADAGFX_ILI9341_STM with the default setting 
#define GSLC_ROTATE     1

When setting
#define GSLC_ROTATE     2

clipping is wrong. 
Exchanging
      pGui->nDispW = ILI9341_TFTHEIGHT;
      pGui->nDispH = ILI9341_TFTWIDTH;
with
      pGui->nDispW = ILI9341_TFTWIDTH;
      pGui->nDispH = ILI9341_TFTHEIGHT;
seems to work.

Best regards.

ImpulseAdventure, 2018-10-08:
You are right -- the `nDispW` / `nDispH` initializers in `GUIslice_drv_adagfx.cpp` will need to be swapped according to the `GSLC_ROTATE` setting from the config (naturally, the majority of my testing was in the default orientation). We would also want to adjust the `ADATOUCH_`* swap and flip settings to ensure the touch handling maps accordingly. Once I have had an opportunity to test out the alternate orientations I'll post an update. Thanks!

espHorst, 2018-10-08:
Thanks for your fast response! I have

espHorst, 2018-10-08:
If you need some one to test the alternate orientations - I can do it. I have here a setup with a XPT2046 touch running. By the way - I have added here for myself the XPT2046 to the code. If there is interest, I can provide the code, maybe via pull request or any other way.  Best regards.

ImpulseAdventure, 2018-10-08:
Hi @espHorst -- if you want to try adjusting the `nDisp`* initializers upon `GSLC_ROTATE` (perhaps even tested against the **XPT2046** touch) and submitting as a PR that would be a big help!

ImpulseAdventure, 2018-10-13:
Closing as fixed by @espHorst PR #55 . Thanks!

closed 2018-10-13 by ImpulseAdventure