ImpulseAdventure/GUIslice #59: Need for an touch handler driver that can be configured by the application?

espHorst, 2018-10-14:
In order to speed up tests with different touch displays I have created an touch handler cpp touch driver class and included this in the GUIslice lib as an own driver which can be selected as the others via a define. It can then be configured and connected to any touch in the app. If you think this is useful for others I can prepare a PR including the abstract driver and an example.
Here a slight overview:

= the touch handler class =================
class TouchHandler {
 public:
  TouchHandler() {};

  void setCalibration(uint16_t ts_xMin, uint16_t ts_xMax, uint16_t ts_yMin, uint16_t ts_yMax);
  void setSize(uint16_t _disp_xSize, uint16_t _disp_ySize);
  //order: map, swap, constraint, flip 
  void setSwapFlip(bool _swapXY,bool _flipX,bool _flipY);

  THPoint scale(THPoint pIn);
  virtual THPoint getPoint(void);

private:
    //landscape perspective: x: width, y: heigth
    uint16_t disp_xSize = 320;
    uint16_t disp_ySize = 240;

    uint16_t ts_xMin = 398;
    uint16_t ts_xMax = 3877;
    uint16_t ts_yMin = 280;
    uint16_t ts_yMax = 3805;
    
    bool swapXY = true;    
    bool flipX = false;    
    bool flipY = true;    
};

= including the touch handler in the main ========================
In the application progam it is included with the XPT2046 as an example by

#include <XPT2046_touch.h>
#include <GUIslice_touch_handler.h>
SPIClass mySPI(2); //Create an SPI instance on SPI1 port.
XPT2046_touch touchDriver(PB12, mySPI); // Chip Select pin, SPI port
class MyTouchHandler: public TouchHandler {
   public:
      THPoint getPoint(void) {
          TS_Point pt = touchDriver.getPoint();
          THPoint ps = scale( THPoint(pt.x,pt.y,pt.z) );
          return ps;
      };
};
MyTouchHandler myTouchHandler;

void setup()
{
  //Start touch driver
  touchDriver.begin();

  //Set the used touch handler
  setTouchHandler(&myTouchHandler);
...

ImpulseAdventure, 2018-10-14:
Hey @espHorst — this could be a helpful option to provide. If I understand correctly, users would have two options:
- Config file specifies one of the existing built-in touch drivers and all associated settings
- Config file specifies a new user-handler mode. In this mode the application can specify the touch driver, associated includes and settings (eg. calibration, touch sensitivity, etc.). One advantage here is that multiple touch handlers can be selected from a single binary.

Apart from running regressions (across multiple touch handlers), do you foresee other use-cases that might be of interest to users?

Thanks a lot for taking the initiative on this!

espHorst, 2018-10-15:
Hi ImpulseAdventure, 
yes, that's the two options for the users. 
Indeed, as you mentioned, this can not only be useful for testing but also for having multiple touch handlers available in the binary.  
It would also be possible to compile multiple touch handlers into the library. You can specify, which ones to compile. Linker will select then, which ones are included in the application. Thus you can have only one library configured for applications using different touch screens.   
This will be a busy week, so I will continue on this on Thursday or Friday. Best regards.

ImpulseAdventure, 2018-10-22:
Thanks again for the enhancement. Closed by integration into PR #64

closed 2018-10-22 by ImpulseAdventure