ImpulseAdventure/GUIslice #61: Can not use TFT_eSPI together with Adafruit_TouchScreen

lukasfischer83, 2018-10-15:
Defining `DRV_TOUCH_ADA_SIMPLE` together with `DRV_DISP_TFT_ESPI` results in 

> undefined reference to `gslc_TDrvInitTouch'

> undefined reference to `gslc_TDrvGetTouch'

So one can not use an ILI9341_8BIT with 4-wire resistive touch on ESP32 with the recommended TFT_eSPI driver.
As already mentioned as TODO in the comments, the include of TouchScreen.h, settings of ports and init should not be done in GUIslice_drv_adagfx.cpp, but in a more independent place.
Thanks, Lukas.

ImpulseAdventure, 2018-10-15:
Hi Lukas -- thanks for noticing this. Not many test combinations had been done with the "simple" touchscreen mode. I believe we were only missing the preprocessor wrappers. Please give the latest commit a try and let me know if it works (I don't have an display that uses this particular touchscreen interface). As for moving the settings out, I agree that these should be migrated into the config, and will look into that next if the above works for you.

ImpulseAdventure, 2018-10-15:
I have now checked in an update (204681a) that enables you to configure the **Adafruit_TouchScreen** (`DRV_TOUCH_ADA_SIMPLE`) from the main settings file (`GUIslice_config_ard.h`). The following defines have been added:
- `ADATOUCH_PIN_XM`, `ADATOUCH_PIN_XP`, `ADATOUCH_PIN_YM`, `ADATOUCH_PIN_YP`, `ADATOUCH_RX`

Hopefully this addresses your suggestions, @lukasfischer83 ?

ImpulseAdventure, 2018-10-15:
I noticed another section of code that required duplication to match what already existed for other drivers. The changes have been made in ( fc2d895 ), so if you have an opportunity to retry that would be great. I don't have screen to test the operation directly, unfortunately.

lukasfischer83, 2018-10-16:
Hi, I tested the new version. With the DBG_TOUCH define I was able to see touch events coming in, allthough they are not scaled propperly. This can be fixed easily by adding  `|| defined(DRV_TOUCH_ADA_SIMPLE)` to line 1112 in `lib/GUIslice/src/GUIslice_drv_tft_espi.cpp`.
Thanks for the quick fixes, Lukas

ImpulseAdventure, 2018-10-17:
Good catch, thanks for the update Lukas. Scaling has now been added.

closed 2018-10-17 by ImpulseAdventure