ImpulseAdventure/GUIslice #62: Font antialiasing for TFT_eSPI

lukasfischer83, 2018-10-17:
I just did a quick and dirty test, and implementing antialiased fonts seems feasible for at least Text Buttons and other elements where background color is known (?).
In gslc_DrvFontAdd() I added 
```
  if (eFontRefType  == GSLC_FONTREF_FNAME){
    m_disp.loadFont((const char*)pvFontRef);
    return NULL;
  }

```
And was able to load the example antialiased font from the TFT_eSPI example from SPIFFS (which would be neat for images too, especially since TFT_eSPI supports jpeg). The lib also ignores the call to `setTextFont(1)` in the gslc_DrvDrawTxt* functions when we return NULL.
Then I hardcoded the background color in the two m_disp.setTextColor() calls, which also needs the BG color as second argument for blending against the background of the text.
It works and the output looks so beautiful, but it still needs some tinkering so the necessary BG color is chosen by the "parent" element's color, which might not be that obvious. Thrilling?

ImpulseAdventure, 2018-10-18:
That is fantastic, @lukasfischer83 !

I hadn't previously looked at the anti-aliased font capability of **TFT_eSPI**, but it does look promising to add support for this into GUIslice. One implementation that I'm considering would be the following:
- Alter `gslc_DrvFontAdd()` in a similar way as you have shown above
- Add a `colElemFill` parameter to the end of `gslc_DrvDrawTxt()` and `gslc_DrvDrawTxtAlign()`. This way the user can specify the background color based on their knowledge of the parent element overlap. It won't work in all situations, but for the majority it probably will.
- Add a new mask bit for `gslc_teTxtFlags` for `GSLC_TXT_ANTI_NONE` / `GSLC_TXT_ANTI_EN`, ie. `GSLC_TXT_ANTI = (1<<6)`
- Create a new function `gslc_ElemSetTextAntialias()` to set this bit
- Within `gslc_DrvDrawTxt*()`, the calls to `setTextColor()` would select between `(colTxt)` and `(colTxt,colElemFill)` on the basis of this antialias bit

Feel free to give this a try if you like, otherwise I'll test out the concept later in the weekend.

ImpulseAdventure, 2018-10-25:
For reference, this feature request is currently being addressed in Lukas' PR #63

ImpulseAdventure, 2018-11-01:
Anti-aliased fonts (smooth fonts) feature now completed in PR #75 
Note that the implementation by @lukasfischer83 was more straightforward than the proposal above and didn't involve the addition of extra flag bits or element mutator functions.

closed 2018-11-01 by ImpulseAdventure