ImpulseAdventure/GUIslice #71: Support for static gslc_Pages / "banner pages"

Grtschnk, 2018-10-25:
Thank you for your great software! Compiling and running it on ESP32 with ESP-IDF. Enjoying it a lot. :)
Issue below:
___

Problem: 
I am presenting data on different pages (e.g. Log, Settings, etc..). I also want to display some status information all the time (battery status, wifi connectivity, etc.), regardless of which page is selected. At the current moment, I would need to create a separate 'Status page' and switch to it periodically (or on request), or I would need to create an element multiple times for each page, and update all of them,

Idea:
1. Option: Elements can be added to multiple pages at once 
2. Option: Multiple pages can be displayed at once: display the "banner" page on top of another page (either in horizontal/vertical direction or in layer direction).

With both options, the library user needs to be mindful of leaving space for the "Status" related elements (either leaving space inside the page for these elements, or leaving space as the area will be overlapped by a status page). But I think this is not an issue, as it does not differ much from the current situation. After all, we are doing embedded programming and not some HTML ;)

I could imagine possible necessary code changes would relate to issues #9 and #24 

I am currently looking at the code, see if I can add this functionality to my fork. Is this functionality that could be added rather easiliy (by myself) or do you think this would require more rework / is not possible?

(I also added drivers/support for loboris TFT driver for ESP32, let me know if you'd like a pull request - currently a bit concious of my coding and git abilities)

ImpulseAdventure, 2018-10-26:
Great suggestion.... some people have previously asked for the same concept: marking some elements as global, so that they are active/visible even after transitioning between pages.

Giving some more thought to it, I think there are several ways that this could be done, but my preference would be to implement an approach such that:
- Users only need to create an element and mark it as belonging to a "global" page
- Internally, no duplication of storage elements would occur. This way, one could add a large number of pages and not consume excess memory for any global element. Replicating elements would have also made it difficult to update & synchronize the elements later, so I'm less keen on a duplication strategy.
- The way I intend to implement it is to modify a couple of the page handlers (for redraw and touch handling) to process both the current page and (if enabled) the global page. Naturally, users would need to be mindful of any visual overlaps (active page takes precedence) as you point out.

**Good news** -- I have a prototype working of this now! If all goes well with the testing, I might be able to roll this feature out soon for you to try.

As for the **loboris TFT** driver, that is great to hear. Assuming the changes are relatively self-contained (ie. basically a new `GUIslice_drv_loboris` file), I'd be happy to take a look at a PR, though note that I have a couple other pending PRs that I would likely need to address first. Perhaps you can open a separate issue to describe the loboris TFT support.

thanks!

espHorst, 2018-10-26:
Would it be possible to display any two pages at the same time? This would even enable swtiching in between different status pages.

ImpulseAdventure, 2018-10-27:
@espHorst -- interesting idea.

I have revised my prototype to support selecting [and dynamically changing] the "global page".
The new example I created has a status bar on top and a tab dialog with content that changes according to the "current page".

It is very easy for the user to do this. Basically you create pages as before, but one (or more) page can contain the "global elements". Let's call this the global page.
- Call `SetPageCur()` to switch between the active (primary) pages
- Call `SetPageGlobal()` (a new function) to select the global page that will also be visible. Calling with `GSLC_PAGE_NONE` will disable the global page mode.

ImpulseAdventure, 2018-10-27:
Conceptually there would be little difference in the treatment of a “global” page versus the “current” page. I was mainly attempting to simplify the usage model somewhat, and changes to the API. That said, I see value in extending this concept.

### Multi-layer Concept ###
Thinking further about the implementation, I see that the current solution could also go a long way to address the request to support pop-up dialogs in #9 (which could offer confirmations or even a reusable pop-up numerical entry page, for example). I suspect the most common use-cases might lend themselves to using 3 “layers” (where a layer is really a reference to a page): back, current and top, though of course it could be generalized further if it made sense. Do you think three might be enough?

In this way, a “back” layer could provide the status and “chrome” that could be common between pages. In my `gslc_ex19` example, I have put a tab dialog and the associated tab selector buttons on the “back” layer. The “current” layer is the one that users will most commonly interact with (this is named the “current page” today). A “top” layer could be activated occasionally to allow for alert dialogs (yes/no) or data entry. Naturally, these layers could be assigned to `GSLC_PAGE_NONE` to disable.

Thanks to your suggestion of enabling dynamic changes to these “layers”, we could very easily select a new “top” layer to provide any arbitrary input dialog / modal to the user while the “current” visible layer stays as-is. For example, select `PG_ALERT_YES_NO` or `PG_ENTRY_NUM` as needed for the top layer.

The nice thing about this layered approach is that we can show through the other layers, so a data entry or alert box would still reveal the elements dynamically updating in the back and current layers.

Furthermore, it would be a simple matter to add a flag per-layer that enables or disables the touch. Therefore, when we present a modal dialog in the top layer, the back and current layers have their touch disabled. Deactivating this top layer would restore the touch on prior layers.

I was thinking of calling these additional two pages "layers", but really they are just two extra "pages" that can be shown at the same time as the current "page". Introducing the concept of a "layer" may help clarify the mechanics / restrictions / usage model.

Would it make more sense to call these layers?

### API ###

There are a couple things we should keep in mind:

- Draw order: we would want to render back, current and top in that order
- Touch: we would want to accept touch input in reverse order (top, current then back).
- The “current page”: Today, users are likely to leverage the APIs involving the “PageCur” (ie. To determine which page we’re on, or what page to transition to). If one provides an arbitrary list of pages/layers, then one would have to decide how to preserve this functionality.

If we restrict it to 3 layers, then the APIs could be tailored to the common use-cases, such as:
- `SetLayerBack(nPageId)`
- `SetPageCur(nPageId)` - existing
- `SetLayerTop(nPageId)`
- `SetLayerTouchEn(nPageId,bTouchEn)`

If we make the set of layers arbitrary then we could offer:
- `SetPageCur(nPageId)` - existing
- `SetLayerPage(nLayerInd,nPageId)`
- `SetLayerTouchEn(nLayerInd,bTouchEn)`

To present a modal popup dialog, we might:
- Save layer touch enables
- `SetLayerTop(PG_ALERT_YES_NO)`
- `SetLayerTouchEn(PG_BACK,false)`
- `SetLayerTouchEn(PG_MAIN,false)`

Perhaps one could even offer a simple wrapper for users that automatically saves/restores the state, so the user just calls:
- `ShowModal(PG_ALERT_YES_NO)`
Saves state, assigns top layer and disables lower layer touch per above
- ... handle popup dialog
- `HideModal()`
This sets SetLayerTop(GSLC_PAGE_NONE) and restores state

With all of the above, we could substitute "Page" for "Layer" if it makes more sense.

### Implementation ###

Behind the scenes, we could certainly implement this with an array of pages (either pointers or page IDs).

One important decision point is whether to include the storage of these layers in the existing GUI structure (wherein I think we would want to limit it to 3 layers), or if we provide a new set of APIs for the user to register their own array of layers/pages.

When introducing features like this I have to be mindful of the memory footprint (particularly on SRAM) for Arduino devices. If is incorporated internally, then 3 layers would hopefully consume only ~ 8 bytes for 3 Page IDs and the current/saved per-layer touch enables.

Thoughts?

ImpulseAdventure, 2018-10-27:
Thanks for the feedback!

In general I would prefer to keep the implementation simple yet scalable, so I agree with your thoughts on the modal dialog requirements. I don't think modal stacking is likely to be common, particularly with smaller TFT displays, and the benefit of background updates is marginal.

### Modal Dialog ###
Your suggestion of leaving the main page unrefreshed and drawing the overlay (eg. modal dialog) seems reasonable but we'd have to disable the redraw of background upon needs-full-redraw (`bRedrawFullPage`) detection when updating the modal dialog. Usually the `bRedrawFullPage` is triggered when updating an element (with no fill, ie. transparent).

### Page within an Element ###
Embedding a page within an element is definitely worth exploring further.

I'm not sure if you are aware, but I did implement the notion of a "compound element" a long time back (see the `XSelNum` example). However, I think the idea of leveraging the `tsPage` structure would be better than how the "compound element" is currently implemented (my preference would be to deprecate this as it adds complexity) and far more flexible.

So, to avoid any confusion in terminology, let's assume we can create a `tsXLayer` element. This element would contain a `tsPage`. When you call `ElemXCreateLayer()` you pass in the `nPageId` of a page that you have already created.

Redraw order would be managed by the user in the order in which elements were added to the root page. I have to think a bit about how events (eg. touch, redraw, tick, etc.) are distributed but I expect that I should be able to leverage the event tree we already have.

Changing background status elements would be as simple as updating elements on the status (background) page.

Is the above generally in line with what you are thinking?

Note: When developing `XSelNum` I found that I occasionally needed to access the element parents; I'll have to see if we still require such mechanisms in the "page element" (layer) approach. I hope to keep things simple.

Thanks again for the great feedback -- I appreciate the input!

espHorst, 2018-10-28:
I'm really impressed about the structured way you approach the new feature. I see that you definitely  picked up all the ideas and came up with even more and even better ones.
Here some further thoughts/proposals on the various topics:

### Example Application
Perfect!

### Page / Layer / Layout
Layer is definitely a better word than page. On other gui systems I have seen often the name
**Layout**  for this element, as it "layouts" other elements. Like
* a window (GUIslice: tsGui) contains a layout (GUIslice: tsPage) 
* the layout (GUIslice: tsPage) contains elements and there exist elements which are further layouts (your proposal: layer)
Maybe Layout would be a good name describing what the element does.

### Page/Layer/Layout: One name for multiple things
At the moment I'm struggling sometimes with the naming page / the proposed name layer / the proposed name layout as they seem to be one name for multiple things. 
page/layer/layout seem to refer to 
* one page, the "page itself" 
* a set of pages (the collection of pages in the array)
* the management functions for a page set

I propose the following naming:
* Stack or Tab or more to come (management of the layoutset, see below)
* LayoutSet (just the array pointing to layouts)
* Layout (one layout itself)

I propose the following structure:

And there are more element, called e. g. "stack" or "tab" (stack: just switchable pages as today,
tab: a tab style element in the future) which is capable of managing a LayoutSet. Using this it looks like
* tsGui
* tsPageArray (a set of pages) including multiple 
* tsPage (on page, the page itself) including multiple 
* tsElement

up to here the same as today, now there are new elements comming in.
Instead of a compound element there is a stack (and in the future maybe a tab, ...) element

* tsStack, manages a layoutSet, this is one or multiple 
* tsLayout, this includes a "elementSet", one or multiple 
* tsElement

I think, this is no change in functionality to what you described, I was just thinking about another naming make the architecture more obvious.

### Page Storage
The  tsPage[] array(today) and in the future maybe also the tsLayoutSet[]  keeps the pointers to layouts (i. e. pages). Layouts(i.e. pages) themselves are stored as today.
Elements like tsStack or tsTab just mangage the LayoutSet.

What do you think?

espHorst, 2018-10-28:
Here an example how the layoutSets can be used without needing any data memory:

```
//this allows us to use the new naming just by reusing the glsc_tsPage
#define glsc_tsLayout gslc_tsPage

glsc_tsLayout layoutMain;
glsc_tsLayout layoutConfig;
glsc_tsLayout layoutStatus;

//create an LayoutSets, using const moves the pointer array to flash (at least for the STM32F1)
glsc_tsLayout* const layoutSetUser[] = {&layoutMain,&layoutConfig};
glsc_tsLayout* const layoutSetStatus[] = {&layoutStatus};
```

I have successfully compiled this on my STM32F1.
At first I made an mistake placing the const keyword. Thus as information:
```
//NOTE: remember that const applies to what is on its immediate left, if nothing on the left then on the right. 
//http://duramecho.com/ComputerInformation/WhyHowCppConst.html
```

espHorst, 2018-11-04:
I have tried the struct initialization, here just a small example:
```
struct element_t {
    uint16_t width;
    uint16_t heigth;
};

element_t eButtonRotate = { .width = 20, .heigth = 20 };
```
and it works. I'm just not sure if it works in plain C (I have here the Arduino C++), literature says it does. Maybe this would be an interesting way to create the layouts.
Even without the need to call any functions in the setup() function.

Therefore this could also be a basis for #76 . 
GUI builders create a "meta" language, this could be such a struct initialization header file.
But for the beginning this file could be done by hand.

Would you be interested about discussions towards this direction? 
Opening a side branch in order to try and to discuss some ideas?

ImpulseAdventure, 2018-11-05:
Thanks a lot for the great suggestions, @espHorst.

From a terminology perspective, I think another term I've seen used to describe your *Layout* might be a *pane*. Let's stick with layout for now. Note that the your *ElementSets* are represented today by a `tsCollect` (ie. an element collection).

### Example ###

Rereading your comments, I believe I understand better what you are proposing with respect to the *layoutSets*. Taking the scenario I provided earlier, would this concrete example be representative of what you were suggesting?

```
tsGui
- CurPage (root): ptr to tsLayout[0]"User"
- tsLayout[0] "User"
  - Contains single tsCollect:
  - tsElem[0]: GSLC_TYPE_STACK
    - Manages tsLayoutSet
    - tsLayoutSet*
      - [0]: ptr to tsLayout[1]"Main" or tsLayout[2]"Config"
      - [1]: ptr to tsLayout[3]"Status"
  - tsElem[1]: GSLC_TYPE_TXT (title)
- tsLayout[1] "Main"
  - Contains single tsCollect:
  - tsElem[0]: GSLC_TYPE_BTN (text button)
  - tsElem[1]: GSCL_TYPE_TXT
  - ...
- tsLayout[2] "Config"
  - Contains single tsCollect:
  - tsElem[0]: GSLC_TYPE_TXT
  - tsElem[1]: GSLC_TYPEX_CHECKBOX
  - tsElem[2]: GSLC_TYPEX_CHECKBOX
  - tsElem[3]: GSLC_TYPE_BTN (text button)
- tsLayout[3] "Status"
  - Contains single tsCollect:
  - tsElem[0]: GSLC_TYPE_TXT (time display)
  - tsElem[0]: GSLC_TYPEX_GAUGE (progress bar)
```
			
To keep things simple I would probably start with all Layouts using absolute coordinates, rather than inheriting positioning from the parent container (**Stacks** / **Tabs**), but naturally I'll be looking for opportunities to keep this relative if possible.

From a rendering perspective, we would begin at the GUI root page/layout. We issue a `PageEvent(DRAW)` on `Layout[0]` which steps through each element in Layout[0]'s tsCollect. Upon encountering `Layout[0].Elem[0]`, we would issue a further `PageEvent(DRAW)` on Layout[1].

In this particular "stack" example: When the user wishes to switch between the "Main" and "Config" screens, they wouldn't be using `SetPageCur()` (which would effectively change our "root" layout). Instead, they would be calling something like `Layout[0].Elem[0].SetCurLayout(1/2)`.

Your suggestion regarding the tsLayoutSet variable length array storage without extra memory allocation is definitely worth looking at.

Just a heads-up: I intend to release my initial implementation of the **Keyboard / GPIO control** (#66) and then can begin a more thorough review of the approach you have outlined here.

### Meta-language ###

I am always open to suggestions for improving ease of use… using struct initialization could be another avenue worth exploring further (perhaps you could open up a separate issue for this?). Note that I had explored something similar when creating the `ElemCreate*_P()` macros to support FLASH-based UI elements in #6.

espHorst, 2018-11-06:
thanks for the well structured example. Mainly based on this example here some findings:

- maybe a layout is just a tsCollect - each layout only contains only a single tsCollect
- LayoutSet storage
  > Your suggestion regarding the tsLayoutSet variable length array storage without extra memory allocation is definitely worth looking at. Presumably we would still require mutability in the pointer values to enable SetCurLayout(), so not entirely clear how this should best be implemented.

  I think there is only the need to store one single pointer in the tsGui(for the "root" layout") and one in each tsStack. Those are pointing to the currently displayed layout. Do you think it is better to store the pointer itself or if it is even better to store just the index into a layoutSet? I see no need to keep the layoutSets themself mutable, do you? 
- Absoute coordinates: I agree, lets start with absolute coordinates.
- Meta-language: I will definitely have a look on the ElemCreate*_P() macros - this sounds interesting!
  >Just a heads-up: I intend to release my initial implementation of the Keyboard / GPIO control (#66) and then can begin a more thorough review of the approach you have outlined here.

  That's not a problem. I'm not in a hurry :-)

espHorst, 2018-11-09:
I had a look on the ElemCreate*_P() macros. I think this is a very clever way to save ram. Are there any drawbacks in using those macros? Why don't always use macros like this and save ram? On non Arduino ATMEGA chips we could just drop the PROGMEM or just define it as empty.

ImpulseAdventure, 2018-12-12:
Just a quick heads-up that I have now posted the code that I developed for the original "global page" layer strategy referenced in my *Oct 27* comment. If interested, please see branch [WIP71-GlobalPage](https://github.com/ImpulseAdventure/GUIslice/tree/WIP71-GlobalPage) and the accompanying example `ex24_tabs`. Code has been posted for both Arduino and LINUX/RPI. This example demonstrates a simple dialog with tabs navigating multiple "real" pages, while the tab buttons, frame and title bar are from a "global" page.

Note that I still intend to explore the more comprehensive layout strategy that espHorst referenced above, but in the meantime I thought I'd post this since there were requests to see the original proof-of-concept.

Here is a screenshot from one page of the demo:
[image]

Obviously the appearance of the tab dialog can be greatly improved -- this was just a proof-of-concept.

ImpulseAdventure, 2019-02-27:
The basic "Page Stack" implementation has now been integrated into the core library. The code supports an arbitrary number of "pages" in a stack that can be shown concurrently (parameterized to 3 at the moment). The suggested usage model is: "base page", "current page" and "overlay page". This way the API is fully compatible with existing programs, as they will continue to run on the "current page".

For example, you can now show:
- Base layer: background controls that appear on every "screen"
- Current layer: the main page in use, perhaps showing one particular view of a tabbed dialog
- Overlay layer: popup dialogs, alerts, data entry boxes, etc.

Example ex24 showing a popup alert on top of a tabbed dialog:
[image]

After carefully reviewing the great proposals discussed in this issue and testing out the alternate implementations, I decided to proceed with this approach as it appears to deliver the biggest functionality gains for users with almost zero impact to the existing API. While I would prefer to revise the naming applied to "pages", this would have implied a large API naming change. I believe the current implementation will still provide much of the capability that @espHorst has described above (though we don't use relative coordinates).

Along with this update, the initial ability to create popup dialog boxes is now available. To show a popup dialog, one creates a "page" for it as before, then simply:
- `gslc_PopupShow(&m_gui,E_PG_ALERT,true);`

This release also supports modal and modeless dialogs (as controlled by a parameter in `PopupShow()`

Again, a huge thank-you to @espHorst @Grtschnk and @Pconti31 for all of the feedback and suggestions on this feature!

closed 2019-02-27 by ImpulseAdventure