ImpulseAdventure/GUIslice #76: Add a GUI Builder for UI layouts and code generation

Pconti31, 2018-11-03:
It's a nice package but it does have a few drawbacks.  First there is a fair amount of book keeping to keep track of the number of UI pieces.  Various defines, enums, and UI storage have to be declared.  UI pieces must have x, y coordinates, along with height and width. Colors and fonts need to be selected and mapped.

ImpulseAdventure, 2018-11-17:
Thanks for the feedback! As with any implementation, some tradeoffs naturally need to be evaluated. I'll attempt to address a few of the observations made here, but as always, I'm open to suggestions if an alternate approach might simplify the usage model.

**Keeping track of defines & enums**
- Most examples demonstrate ways in which UI elements can be modified at run-time. For these elements, one needs some manner of handle to reference elements following creation. A few different methods could be used here: user-enumerated element IDs, retaining individual element pointers, etc. It seemed to me that defining enums (eg. `E_ELEM_*`) made for the simplest approach, but perhaps there are other strategies to consider here.
- Note that the user does not need to define any enums for the UI elements if dynamic access is not required. In this case, automatic ID assignment can be used, which simply involves passing `GSLC_ID_AUTO` into the `ElemCreate*()` functions instead of the user allocated enums.

**UI Storage**
- An early design decision to address embedded (low-memory) platforms was for the library to **avoid the use of dynamic memory allocation**. This way the user is granted full visibility and deterministic control over the runtime memory requirements. This approach should minimize the risk of heap fragmentation and memory leaks.
- A side-effect of this is that the user code becomes responsible for instantiating storage for the element arrays within each page of the GUI. In most cases this isn't particularly burdensome, but there are a couple cases wherein this is less convenient:
  - Needing to keep track of how many elements you need to provide storage for. Currently, an error is reported (when `DEBUG_ERR` is enabled) if more elements are created than the storage allocated.
  - For FLASH-based operation (ie. `arduino_min` examples), the user needs to keep track of the number of elements that are stored in FLASH versus those stored in RAM. An extra define is used to track this. If no low-memory optimization is required, then this additional count doesn't need to be managed.
- A good part of the API could be simplified from the user perspective if I were to backtrack on this particular design decision, and instead allow the GUIslice library to instantiate the elements and pages as they are needed. I have assumed that most low-memory / Arduino users would prefer that the library avoid dynamic memory allocation, but would be interested in hearing any feedback on this point.

**Element coordinates & sizing**
- The current element creation API expects to be provided with coordinates and dimensions. I think this is a pretty natural expectation for most GUI libraries. Clearly there could be an alternate approach that leverages some sort of **automatic layout feature** wherein the GUI is responsible for positioning and sizing the elements to fit the screen.
- If there was interest, I could explore such a feature further. For example, the existing ElemCreate*() APIs could be called, but each with a dummy (0,0) coordinate. A "layout" function could be called (prior to the first screen update) that repositions the newly created elements according to some basic grid scheme, perhaps with certain rows/columns expanding whereas others could be statically sized (think standard web table definition).

That said, the addition of the [GUIslice Builder](https://github.com/ImpulseAdventure/GUIslice/wiki/GUIslice-Builder) should certainly help alleviate a good number of the implementation details mentioned above!

As I think the Builder does indeed address most of these, I intend to close this issue but will keep it open for a bit longer in case any of the above ideas are worth exploring further (eg. auto layouts, dynamic memory allocation, etc.) or any other strategies that might ease implementation.

Thanks

ImpulseAdventure, 2018-11-24:
Closing as majority of implementation tasks are addressed by [GUIslice Builder](https://github.com/ImpulseAdventure/GUIslice/wiki/GUIslice-Builder)

closed 2018-11-24 by ImpulseAdventure