ImpulseAdventure/GUIslice #80: GUIslice Builder - beta test bug reports

ImpulseAdventure, 2018-11-13:
With the launch of the **GUIslice Builder** (#79) as a *beta release*, it is likely that issues may be encountered in its operation or associated documentation. This issue will be used to capture user feedback concerning these reports.

This particular entry will also be updated periodically to reflect any known open issues, so as to avoid report duplication.

### Bug Report Info ###
For any reports, please indicate the following details:
- **Operating system** (eg. `Windows 10`)
- **GUIslice version** (eg. `0.10.4`)
- **GUIslice Builder version** (from About menu, eg. `0.10.4-beta4`)
- **Target platform**; defined in Edit -> Options -> Target Platform (eg. `arduino`)
- If a crash occurred, it would be helpful to attach the crash log created in the Builder `/logs/` subfolder

### Current Known Issues ###
As of release `0.10.4-beta4`:
- **None** (this will be updated as needed)

ImpulseAdventure, 2019-01-30:
### Summary of Issues in **Builder 0.10.4-beta7**:
NOTE: Many of these issues have already been resolved in the upcoming **Builder 0.11.0-rc1** version, but have been captured here so that users are aware of them in the meantime

- Internal38: Dragging item in treeview past end of branch (#92)
  - **Workaround**: avoid dragging past end
  - **STATUS**: Fixed in 0.11.0-rc1
- Internal39: Renaming first page can cause generate to fail (#95)
  - **Workaround**: avoid renaming first page (E_PG_MAIN)
  - **STATUS**: Fixed in 0.11.0-rc1
- Internal45: Code gen fails, sometimes duplicating variables for Page$3 and beyond (#95)
  - **Workaround**: Rename variables as noted in #95 
  - **STATUS**: Fixed in 0.11.0-rc1
- Internal48: Crash on generate (in macOS) when space in filename
  - **Workaround**: avoid spaces
  - **STATUS**: Still open as have not reproduced this yet in development environment (Windows & LINUX).
- Internal50: Delete not active on new blank page
  - **Workaround**: Add any element to page first, then delete the page.
  - **STATUS**: New

vitormhenrique, 2019-02-14:
Hello Calvin and Paul,

I stated using the builder to setup my UI, one thing that it would be really nice is the ability to save colors or palette of colors to reuse them.
Some elements would also beneficial if they had an option to have a "transparent" background.

vitormhenrique, 2019-02-14:
Another cool thing to add would be having the primitives on the tool box (lines, rectangles, circles) 
They can be useful to create custom UI's.

ImpulseAdventure, 2019-02-14:
Hi Vitor -- good suggestions!
- Regarding the palette of recent colors, I understand this has been rolled into the latest Builder release (**0.11.0-rc1**). Perhaps Paul can send you a binary to test out.
- For transparent background elements, which elements were you thinking of?
  - In your particular case, would the transparency be useful to reveal an image background or just other elements beneath?
  - GUIslice supports element transparency, but the flag is not currently enabled in most of the Builder elements.
- For primitives: GUIslice supports two types of drawing APIs, immediate draw mode primitives (eg. `DrawFillCircle()`) and another set for the creation of GUI elements supporting redraw and touch tracking (eg. `ElemCreateBox()`, `ElemCreateLine()`, etc.).
  - At this time, only the Box element is currently supported in the Builder, so we could take a look at adding (at least) the line element as it seems like it would be useful as a divider.
  - In the meantime, I wonder if a "Box" with height or width=1 might be a temporary workaround?

vitormhenrique, 2019-02-14:
Hello Calvin and Paul,

- I just noticed the recent colors on the last build that Paul gave me, and it is persistent between projects, so that it more than enough I think.

- My specific scenario was to see elements beneath, for example I was building an ui with two sliders on top of each other, to display my gimbals. 
[image]

- I think lines and circles would be really good, especially if the circles could be Elements that can be updated, ui could use circles to show status of something with different colors, for example on off / read and green (also again on my previous screen shot I would like to add a transparent circle around the sliders... 

Like this:

[image]

ImpulseAdventure, 2019-02-19:
@vitormhenrique -- from what you describe, I assume you are displaying these virtual gimbals to represent positional status rather than accepting touch control. If so, the most flexible solution is to use the "custom draw callback" approach. In the Builder, create a "Box" element and then enable the custom "Draw Function". This will create a callback routine that enables you to draw the shapes according to your sensor input, while still integrating with the standard GUIslice redraw event hierarchy. Please see `ex06_ard_callback` for an example of this method and please don't hesitate to let me know if you have any further questions.

bittenbybytes, 2019-03-25:
Hi, I noticed that the builder crashes on code generation once a graph is included in the design, leaving an incomplete source file (missing section markers). It boils down to a simple mistake in typecasting in CodeGenerator.java:1424 replacing TextModel with GraphModel seems to solve the issue.

Pconti31, 2019-03-27:
Thanks for the report on XGraph.  Fix will be in Builder RC5
Paul--

DonpK, 2019-03-27:
I had a similar crash problem with the ProgressBar which I reported. I've installed  0.11.0-rc4 and no longer appear to have the same problem. Was the ProgressBar issue addressed in rc4?

Pconti31, 2019-03-29:
@DonpK Yes, the progress bar was fixed in rc4.   
I actually thought I had fixed XGraph too but that shows what thought gets you...
rc5 should be coming soon.
Paul--

pdupuy, 2019-06-25:
OS: Windows 7; GUIslice: 0.12.0; Builder: 0.11.0-rc4; Target: arduino
The image file selection dialog always opens in "Program Files (x86)\GUIsliceBuilder\arduino_res". When there are many images to add and their directory is far removed from the installation directory, adding them is onerous. Possible solutions: add an option to set "Default Image Source Directory"; remember the last directory in which an image was selected.

pdupuy, 2019-06-25:
OS: Windows 7; GUIslice: 0.12.0 (4769b20); Builder: 0.11.0-rc4; Target: arduino
Changed "Target's Image Directory" value in General project options is not persisted after saving, closing, and re-opening the project.

pdupuy, 2019-06-25:
OS: Windows 7; GUIslice: 0.12.0 (4769b20); Builder: 0.11.0-rc4; Target: arduino
When adding an Image Button, it is required to also choose a file for "selected" or "glow" state ("Choose your Button's Image when pressed"). When not using the glow feature, this seems to add an unnecessary step to the process (again picking the same file). Furthermore, the exported output has to be amended to include a line disabling glow for the image button:
`gslc_ElemSetGlowEn(&m_gui,pElemRef,false),`
Possible solution: Add a project option for default Button Image glow and a "Use Glow" property on Button Image instances. Set the "Use Glow" property to the default value on Button Image creation; permit changing it and selecting or removing (or changing) a glow image later.

pdupuy, 2019-06-25:
OS: Windows 7; GUIslice: 0.12.0 (4769b20); Builder: 0.11.0-rc4; Target: arduino
When saving or exporting a project file, a confirmation dialog ("Project saved..."; "Code generated...") is generated. This breaks development flow and causes what feels like an unnecessary action; when I save frequently (habitually), the confirmation interruption friction is heightened. Possible solutions: add a project option to "Disable Save confirmation dialog" and "Disable export confirmation dialog"; remove confirmations of success altogether and only message on failure.

pdupuy, 2019-06-25:
OS: Windows 7; GUIslice: 0.12.0 (4769b20); Builder: 0.11.0-rc4; Target: arduino

Deleting a Button Image does not remove the corresponding case in the `CbBtnCommon` callback function. ImageButton$1 was created and then deleted in the Builder (and it does not appear there); the exported code included a callback for it:

`bool CbBtnCommon(void* pvGui,void *pvElemRef,gslc_teTouch eTouch,int16_t nX,int16_t nY)
{
  gslc_tsElemRef* pElemRef = (gslc_tsElemRef*)(pvElemRef);
  gslc_tsElem* pElem = pElemRef->pElem;

  if ( eTouch == GSLC_TOUCH_UP_IN ) {
    // From the element's ID we can determine which button was pressed.
    switch (pElem->nId) {
//<Button Enums !Start!>
      case E_IMGBTN1:
        //TODO- Replace with button handling code
        break;
      case E_IMGBTN2:
        //TODO- Replace with button handling code
        break;`

ImpulseAdventure, 2019-06-26:
@pdupuy -- thank you very much for taking the time to file these detailed Builder bug reports! I have integrated them into our internal Builder issue list for review. Note that PaulC (Builder author) is in the midst of a major Builder code-gen rewrite (for 0.13.0) so some enhancements might be pushed to a later update.

Pconti31, 2019-07-01:
> OS: Windows 7; GUIslice: 0.12.0 ([4769b20](https://github.com/ImpulseAdventure/GUIslice/commit/4769b202bb5ef201c5844709f38976ba5a94031d)); Builder: 0.11.0-rc4; Target: arduino
>
> Changed "Target's [...]

Assigned bug# 77 and fixed in next release 0.13.0
problem only occured if user failed to press 'enter' key on the field before pressing  "'OK" button.

Pconti31, 2019-07-01:
> The image file selection dialog always opens in "Program Files (x86)\GUIsliceBuilder\arduino_res". >When there are many images to add and their directory is far removed from the installation [...]

Assigned bug#: 76 fixed in next release 0.13.0.
Will now remember last image directory accessed.

Pconti31, 2019-07-01:
> Deleting a Button Image does not remove the corresponding case in the `CbBtnCommon` callback function. ImageButton$1 was created and then deleted in the Builder (and it does not appear there); the exported code included a callback for it:

Assigned bug#80: fixed in next release of builder 0.13.0

Pconti31, 2019-07-01:
> When saving or exporting a project file, a confirmation dialog ("Project saved..."; "Code generated...") is >generated. This breaks development flow and causes what feels like an unnecessary action; [...]

Assigned bug#80:  Fixed in next release of builder
implemented a status bar instead of confirmation dialogs.

ImpulseAdventure, 2019-09-02:
Closing beta test thread given that Builder 0.13.0 was released.

We will keep track of issues with the Builder in its own companion repository. Please file any problem reports at the Builder repository noted below:
[GUIslice Builder - Issues](https://github.com/ImpulseAdventure/GUIslice-Builder-source/issues)

thanks!

closed 2019-09-02 by ImpulseAdventure