ImpulseAdventure/GUIslice #81: Support for Pop-UP Dialogs

Pconti31, 2018-11-25:
It would make it easy to pop up a standard dialog box that prompts users for a value or informs them of something.   I suggest four versions or styles.

- Confirm Dialog.  Asks a confirming question, like yes/no/cancel.
- Input Dialog. Prompt for some input.
- Message Dialog. Tell the user about something that has happened.
- Error Dialog. Report any error conditions.

Perhaps it could be built on top of issue #71 Support for sub-pages.
Note that might make issue #5 Create Keyboard or Keypad easier to implement.

ImpulseAdventure, 2019-02-27:
Update: Now that #71 has been implemented, the ability to support popup dialogs is now also available.

Example ex24 shows a example popup dialog on top of other page elements:
[image]

To create a popup dialog, simply make a "page" with the dialog and then call:
- `gslc_PopupShow(&m_gui,E_PG_ALERT,true);`

This will show a modal dialog (ie. prevents touch events outside of the dialog). If you prefer to continue supporting background touch events (ie. modeless dialog), you can pass "false" to the bModal parameter.

Note that slower Arduino devices may exhibit a visible page redraw following a PopupHide(). I will see if optimizations can be done later for these devices later.

Next, I plan to continue to integrate support for the data entry dialogs (as originally contributed by @Pconti31 ). Finally, I will investigate whether it is practical to create some "canned" dialog types, relieving the user from needing to construct these alert dialog boxes themselves.

As always, comments and suggestions welcome on this!

Lucy2003, 2021-08-08:
Maybe we can add a gslc_collect poniter and a result pointer (void*) to each page first,and then create and store "standard dialog" as a collect into the rom.Next,we provide some function for users,such as these:
```C
    bool gslc_msgbox(gui,page,const char* title,const char* msg,uint8_t* ret /*,uint8_t style*/); // "popup" a MsgBox but not wait for the result
    bool gslc_inputbox(gui,page,const char* title,const char* msg,char* ret);     // "popup" a InputBox but not wait for the result
    bool gslc_checkStdDialog(gui,page); //check for standard dialog result,if closed then return true
```
We would NOT wait for the result because of WDT soft reset.Users will need to process the dialog either in `loop()` or in callback function (e.g. a new callback `CB_StdDialog()`).Note that we will need to draw the "standard dialog" collect later so that it won't be covered by other elements.BTW,maybe we can process input events in ` gslc_checkStdDialog()` but  `gslc_update()` on some cases to increase speed?

When users need to popup a "standard dialog",what they need is:
1. Call `gslc_inputbox()` or `gslc_msgbox()`
2. Call `gslc_update()`
3. Wait for callback or call `gslc_checkStdPopup()` actively.