ImpulseAdventure/GUIslice #82: Support for MCUFRIEND_kbv

ImpulseAdventure, 2018-11-27:
Today, GUIslice supports a number of display drivers for Arduino including the *Adafruit* series (eg. *Adafruit-ILI9341*, etc.) that leverage **Adafruit-GFX**, as well as **TFT_eSPI** and **Adafruit-GFX-AS** (for *STM32*).

There also exists a number of inexpensive TFT displays manufactured by **mcufriend**, for which @prenticedavid has built the popular [MCUFRIEND_kbv](https://github.com/prenticedavid/MCUFRIEND_kbv) library (primarily targeting UNO shields).

If there is interest, I may consider adding support for these displays as it looks like the integration into GUIslice could be relatively straightforward. As always, the additional display driver support would only be included if selected through config, so it should not burden the core library.

Pconti31, 2018-11-27:
Actually, I implemented that six months ago.  I'm happy to share the code with you.
It just needs to be updated for 0.10.4 and your new touch parameters.

ImpulseAdventure, 2018-11-27:
That would be great, Paul!

prenticedavid, 2018-11-27:
It should be fairly straightforward to use MCUFRIEND_kbv or any other GFX compatible hardware library.

However everything is complicated for the user to configure.  Could I suggest that Arduino examples have meaningful names?  And that you provide a sketch that simply reads the Touch and displays it.

Surely there are some popular hardware combinations e.g. TFT_eSPI with Red ILI9341/XPT2046 display.
The User would feel comfortable with Bodmer's examples i.e. Touch calibration and graphics

MCUFRIEND shields come with simple Resistive Touch Panel.   The Touch pins are shared with the TFT pins.
There is a large range of TFT controllers and different layout of Touch pins.    The User can diagnose / calibrate with the MCUFRIEND_kbv examples.

Adafruit Displays come with different Touch panels.    Again,   there are examples.

David.

ImpulseAdventure, 2018-12-03:
Thanks David for the followup!

I have now renamed all examples to give them a meaningful name (#78).

That's a good suggestion to provide a simplified touch example, as that should help users determine whether the config was correct, the touch screen is responding and whether their calibration is accurate.

Regarding the configuration, in most cases once a standalone TFT_eSPI or Adafruit-GFX example has been confirmed to run, it should be a simple matter to copy the same pinout to the GUIslice config. I have a number of ideas in simplifying the config process and will be revising the guidance in the wiki shortly. I also plan on offering a number of specific CPU+display connection examples with wiring and config -- this should help users get up and running faster too.

As for integrating MCUFRIEND, Pconti31 has sent me a patch for GUIslice that added **MCUFRIEND_kbv** support. Once I have access to a new mcufriend display that works, I can begin testing and integrating the patch.

Thanks!

ImpulseAdventure, 2018-12-23:
mcufriend support has been added to PR #87

ImpulseAdventure, 2019-01-02:
Closed as merged in #87

closed 2019-01-02 by ImpulseAdventure