ImpulseAdventure/GUIslice #83: Error compiling in Arduino IDE (lto1.exe: internal compiler error)

mafckz1, 2018-12-02:
Hi.
I tried to compile half examples of GUIslice in Arduino IDE and every time there is an error compiling:

`lto1.exe: internal compiler error: in splice_child_die, at dwarf2out.c:4905`

Other sketches compile successfully.
(tried to compile on 2 different computers - clear windows 7-8.1 x64, Arduino 1.8.7, libraries and nothing more)

Here is the ArduinoIDE messages:
```
Arduino: 1.8.7 (Windows 7), Board: "Arduino/Genuino Uno"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Lenovo\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10807 -build-path C:\Users\Lenovo\AppData\Local\Temp\arduino_build_501431 -warnings=none -build-cache C:\Users\Lenovo\AppData\Local\Temp\arduino_cache_785428 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\Lenovo\Documents\Arduino\libraries\GUIslice\arduino_min\gslc_ex01_ard_min\gslc_ex01_ard_min.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Lenovo\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10807 -build-path C:\Users\Lenovo\AppData\Local\Temp\arduino_build_501431 -warnings=none -build-cache C:\Users\Lenovo\AppData\Local\Temp\arduino_cache_785428 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\Lenovo\Documents\Arduino\libraries\GUIslice\arduino_min\gslc_ex01_ard_min\gslc_ex01_ard_min.ino
Using board 'mega' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Build options changed, rebuilding all
Detecting libraries used...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10807 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\mega" "C:\\Users\\Lenovo\\AppData\\Local\\Temp\\arduino_build_501431\\sketch\\gslc_ex01_ard_min.ino.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10807 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\mega" "-IC:\\Users\\Lenovo\\Documents\\Arduino\\libraries\\GUIslice\\src" "C:\\Users\\Lenovo\\AppData\\Local\\Temp\\arduino_build_501431\\sketch\\gslc_ex01_ard_min.ino.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10807 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\mega" "-IC:\\Users\\Lenovo\\Documents\\Arduino\\libraries\\GUIslice\\src" "C:\\Users\\Lenovo\\Documents\\Arduino\\libraries\\GUIslice\\src\\GUIslice.c" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10807 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\mega" "-IC:\\Users\\Lenovo\\Documents\\Arduino\\libraries\\GUIslice\\src" "C:\\Users\\Lenovo\\Documents\\Arduino\\libraries\\GUIslice\\src\\GUIslice_drv_adagfx.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10807 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\mega" "-IC:\\Users\\Lenovo\\Documents\\Arduino\\libraries\\GUIslice\\src" "-IC:\\Users\\Lenovo\\Documents\\Arduino\\libraries\\Adafruit_GFX_Library" "C:\\Users\\Lenovo\\Documents\\Arduino\\libraries\\GUIslice\\src\\GUIslice_drv_adagfx.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10807 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\mega" "-IC:\\Users\\Lenovo\\Documents\\Arduino\\libraries\\GUIslice\\src" "-IC:\\Users\\Lenovo\\Documents\\Arduino\\libraries\\Adafruit_GFX_Library" "-IC:\\Users\\Lenovo\\Documents\\Arduino\\libraries\\Adafruit_ILI9341" "C:\\Users\\Lenovo\\Documents\\Arduino\\libraries\\GUIslice\\src\\GUIslice_drv_adagfx.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10807 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\mega" "-IC:\\Users\\Lenovo\\Documents\\Arduino\\libraries\\GUIslice\\src" "-IC:\\Users\\Lenovo\\Documents\\Arduino\\libraries\\Adafruit_GFX_Library" "-IC:\\Users\\Lenovo\\Documents\\Arduino\\libraries\\Adafruit_ILI9341" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\libraries\\SPI\\src" "C:\\Users\\Lenovo\\Documents\\Arduino\\libraries\\GUIslice\\src\\GUIslice_drv_adagfx.cpp" -o nul
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10807 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC [... cut ...]
Using library Adafruit_ILI9341 at version 1.2.0 in folder: C:\Users\Lenovo\Documents\Arduino\libraries\Adafruit_ILI9341 
Using library SPI at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SPI 
Using library Wire at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire 
Using library Adafruit_STMPE610 at version 1.0.1 in folder: C:\Users\Lenovo\Documents\Arduino\libraries\Adafruit_STMPE610 
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.
```

ImpulseAdventure, 2018-12-02:
Hi — thank you for the report!

This is apparently a bug in the compiler (**avr-gcc**) that is included with the latest Arduino IDE (1.8.7). Unfortunately, there is some code in the library that is triggering this particular compiler issue.

The current workaround is to either:
- **Option 1**: Remove the `-g` option in the platform.txt within your Arduino installation.
  - You should find the platform.txt file in the following path: `...\Arduino\hardware\arduino\avr\platform.txt`
  - You will find two rows labeled `compiler.c.elf.flags=` and `compiler.S.flags=`. In both lines you’ll see a `-g` parameter. Simply remove the `-g`, close and reopen the Arduino IDE and recompile.
  - The path above might be located in a different place depending on how you installed the Arduino IDE. It may require changing in one of the following locations:
    - `C:\Program Files (x86)\Arduino\hardware\...`
    - `C:\Users\{username}\AppData\Local\Arduino15\packages\arduino\hardware\avr\...`
- **Option 2**: Use an older version of Arduino IDE (e.g. 1.8.5)

I am currently working on tracking down what specific code construct in the library is triggering this issue so that no workaround will be necessary.

Sorry for the inconvenience in running into this issue. Please let me know if either of the above solutions works for you. In my testing (and others), removing “-g” was sufficient to avoid the problem.

Thanks!

DETAILS
- From what I have read on the net, this issue is triggered by the combination of the link-time optimization mode (-flto) and the debugger symbols (-g) in the compilation in recent versions of avr-gcc.
- Disabling debugger symbols (-g) is sufficient to work around the issue.

mafckz1, 2018-12-02:
Problem was with "Arduino AVR Boards v1.6.23" in last version of ArduinoIDE. 
After downgrading Boards version to 1.6.21, examples of GUIslice compile without errors.

Thank you!

ImpulseAdventure, 2018-12-13:
FYI -- I have now created a trimmed down testcase to demonstrate the compiler problem and submitted an issue to **ArduinoCore-avr** to investigate.
- https://github.com/ImpulseAdventure/Arduino-LTO-Error

My assumption is that their resulting guidance will be for people to roll back the **Arduino AVR Boards** to version 1.6.21, as noted above.

ImpulseAdventure, 2018-12-21:
Confirmed as issue in Arduino compiler toolchain. Upcoming version of Arduino IDE will resolve this, but for now the best guidance is to ensure **Arduino AVR Boards** version `1.21` is used. Closing the issue as this guidance has been integrated into the installation and configuration guides.

closed 2018-12-21 by ImpulseAdventure