ImpulseAdventure/GUIslice #92: Builder has fatal error dragging elements

rrroonn, 2019-01-08:
Firstly, great work.  Pleasing to see some vision in action.

WINDOWS 10 x64
Build 0.10.4-beta7
target arduino

ISSUE is that when I drag an element (of which I have two on the page) in the TreeView to change the z-order, a "fatal exception occurs".  I am trying to move the first element to the bottom.
Visually/functionally, when I start dragging, I get a line (drop point) at its current position, but I dont get a line after the last element.  If I drag slightly lower, I do get a line, but this is off-page.  If I release here, I get the fatal exception and the element has moved off the page (to top level) in the Tree View.

This makes it hard to get the Z order correct ;-)

thank you

[image]

[image]

[20190109_063125.txt](https://github.com/ImpulseAdventure/GUIslice/files/2738384/20190109_063125.txt)

[20190109_063114.txt](https://github.com/ImpulseAdventure/GUIslice/files/2738383/20190109_063114.txt)

Pconti31, 2019-01-09:
Its a known problem.  Its not supported to go below the last element on a page though I thought I had fixed the crash.   You need to move Text$2 above Box$1 to get your desired result.
Box$1 cannot move below Text$1 otherwise.

ImpulseAdventure, 2019-01-23:
Closing as I understand this has been fixed in the upcoming release of the Builder.

closed 2019-01-23 by ImpulseAdventure