ImpulseAdventure/GUIslice #94: Builder: Selecting component in Tree View should/could select item for editing

rrroonn, 2019-01-08:
It would be nice to be able to click on an item in the Tree View and have it bring that component into focus and show its properties.  This would be especially useful when components are not visible due to Z order.

An alternative would be to have a dropdown list of components (on current page?) in the Properties area.

thank you

Ron

rrroonn, 2019-01-08:
Funny, I restarted app and now this does work.  Not sure of what the conditions are to break this function.  Will keep and eye on it.

Pconti31, 2019-01-09:
Let us know if you find a way to reproduce this condition.

rrroonn, 2019-01-09:
That project that I uploaded for bad code generation has that issue on PG_MAIN - can't select anything though.  Maybe similar symptom but different cause.

ImpulseAdventure, 2019-01-23:
Closing as the selection feature is currently implemented and the treeview non-selection behavior does not appear to be reproducible at the moment. If it recurs, please feel free to add a followup comment and we can reopen for investigation. Thanks!

closed 2019-01-23 by ImpulseAdventure