ImpulseAdventure/GUIslice #95: Builder: Code generation error: MAX_ELEM_PG_MAIN_RAM was not declared

rrroonn, 2019-01-09:
Windows 10x64 Beta 7 arduino
Not sure how it got to that state - I thought maybe some code edits of mine.  So I saved as new project and generated code and error remains.

I have attached project file and generated souce code.  
[ron2.zip](https://github.com/ImpulseAdventure/GUIslice/files/2738995/ron2.zip)

Compiler Error Messages:
```
Arduino: 1.8.8 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

ron2:54:46: error: 'MAX_ELEM_PG_MAIN_RAM' was not declared in this scope

 gslc_tsElem                     m_asMainElem[MAX_ELEM_PG_MAIN_RAM];

                                              ^

ron2:55:49: error: 'MAX_ELEM_PG_MAIN' was not declared in this scope

 gslc_tsElemRef                  m_asMainElemRef[MAX_ELEM_PG_MAIN];

                                                 ^

C:\Users\rwbar\OneDrive\arduino\SecurityAlarm\gui\ron2\ron2.ino: In function 'bool CbBtnCommon(void*, void*, gslc_teTouch, int16_t, int16_t)':

ron2:121:32: error: 'E_PG_MAIN' was not declared in this scope

         gslc_SetPageCur(&m_gui,E_PG_MAIN);

                                ^

C:\Users\rwbar\OneDrive\arduino\SecurityAlarm\gui\ron2\ron2.ino: In function 'bool InitGUI()':

ron2:145:29: error: 'm_asMainElem' was not declared in this scope

   gslc_PageAdd(&m_gui,E_PG2,m_asMainElem,MAX_ELEM_PG_MAIN_RAM,m_asMainElemRef,MAX_ELEM_PG_MAIN);

                             ^

ron2:145:42: error: 'MAX_ELEM_PG_MAIN_RAM' was not declared in this scope

   gslc_PageAdd(&m_gui,E_PG2,m_asMainElem,MAX_ELEM_PG_MAIN_RAM,m_asMainElemRef,MAX_ELEM_PG_MAIN);

                                          ^

ron2:145:63: error: 'm_asMainElemRef' was not declared in this scope

   gslc_PageAdd(&m_gui,E_PG2,m_asMainElem,MAX_ELEM_PG_MAIN_RAM,m_asMainElemRef,MAX_ELEM_PG_MAIN);

                                                               ^

ron2:145:79: error: 'MAX_ELEM_PG_MAIN' was not declared in this scope

   gslc_PageAdd(&m_gui,E_PG2,m_asMainElem,MAX_ELEM_PG_MAIN_RAM,m_asMainElemRef,MAX_ELEM_PG_MAIN);

                                                                               ^

C:\Users\rwbar\OneDrive\arduino\SecurityAlarm\gui\ron2\ron2.ino: In function 'void setup()':

ron2:250:26: error: 'E_PG_MAIN' was not declared in this scope

   gslc_SetPageCur(&m_gui,E_PG_MAIN);

                          ^

exit status 1
'
MAX_ELEM_PG_MAIN_RAM' was not declared in this scope
```
Note,

I did have a 3 pages and I deleted "E_PG2" and I think that contributed to the problem. You can see that E_PG_MAIN should be declared here but instead E_PG2 is:

```
//<Enum !Start!>
enum {E_PG2,E_PG_PIN};
enum {E_BOX2,E_BTN10,E_BTN11,E_BTN12
      ,E_BTN13,E_BTN14,E_BTN15,E_BTN16
      ,E_BTN6,E_BTN7,E_BTN8,E_BTN9};
enum {E_FONT_TXT2};
//<Enum !End!>

```

Pconti31, 2019-01-09:
I will look into the issue.

rrroonn, 2019-01-09:
> The Page$1 - enum E_PG_MAIN was deleted. The Builder requires the first
> page to be these names.
> A future version will remove this dependency. The Builder does check to
> avoid a deletion of this page but not for renaming followed by deletion
> which might have caused the problem. If you can reproduce how you got into
> this state let us know.
> I have recovered as much of your project as possible in this zip file. the
> ron2B.ino complies without errors.
> ron2B.zip

Thank you vm. Appreciated.

Pconti31, 2019-01-28:
This bug is fixed in Builder 0.11.0-rc1.  However in fixing this bug I found another one that causes duplicate names for storage when users define more than two pages of screens.   
such as:

`gslc_tsElem              m_asExtraElem[MAX_ELEM_PG_EXTRA_RAM1];`
`gslc_tsElemRef           m_asExtraElemRef[MAX_ELEM_PG_EXTRA1];`
`gslc_tsElem              m_asExtraElem[MAX_ELEM_PG_EXTRA_RAM2]; <—duplicate `
`gslc_tsElemRef           m_asExtraElemRef[MAX_ELEM_PG_EXTRA2];   <—duplicate`
 
This is also fixed in 0.11.0-rc1 when its released.  
In the meantime the work-around is to simply rename

`m_asExtraElem[MAX_ELEM_PG_EXTRA_RAM2] as m_asExtraElem2[MAX_ELEM_PG_EXTRA_RAM2]`
`m_asExtraElemRef[MAX_ELEM_PG_EXTRA2]  as `m_asExtraElemRef2[MAX_ELEM_PG_EXTRA2];`

and modify 

`gslc_PageAdd(&m_gui,E_PG_EXTRA2 `

to use the renamed storage.

ImpulseAdventure, 2019-02-12:
Fixed in Builder 0.11.0-rc1. Closing.

closed 2019-02-12 by ImpulseAdventure