ImpulseAdventure/GUIslice #103: Support for RA8875 TFT/Touch Driver

gyundt, 2019-01-26:
Requesting adding support for the RA8875 TFT/Touch driver chipset. 

Breakout board and library examples provided by Adafruit (adafruit.com/product/1590)

I can test any beta code on the above breakout board with a 40pin 5' TFT display, running on an Arduino Uno/Teensy 3.6.

ImpulseAdventure, 2019-02-19:
Hi @gyundt --

I have now added initial support for the **RA8875** display driver. It compiles, but as I do not have a display that uses this driver, I would be interested in taking up your offer of assistance for testing it.

A few notes:
- I was able to locate two RA8875 display driver libraries: [Adafruit](https://github.com/adafruit/Adafruit_RA8875) and [Sumotoy](https://github.com/sumotoy/RA8875). For now, I have integrated Adafruit's library.
- For the initial testing, I have provided a display-only example config (`ard-adagfx-ra8875-notouch`) with no touch active. I understand that the RA8875 may include an integrated touch controller, though I am inclined to see if we can use the `simple` (resistive 4-wire touch handler) mode later.
- I have not yet attempted to test the GUIslice core on a **Teensy**, so will migrate support for that platform to a separate issue.

Please try the development branch: [WIP103-RA8875](https://github.com/ImpulseAdventure/GUIslice/tree/WIP103-RA8875) and let me know if your UNO can display `ex01_ard_basic` and `ex04_ardmin_ctrls` properly (no touch will be active yet).
- Within `src/GUIslice_config.h`, uncomment the `ard-adagfx-ra8875-notouch` line.

Thanks!

ImpulseAdventure, 2019-02-25:
Preliminary support for this feature has been merged into the master

@gyundt -- in case you find an opportunity to see whether this worked for you, please feel free to update this issue and I will re-open it.

closed 2019-02-25 by ImpulseAdventure

ImpulseAdventure, 2019-03-29:
@gyundt -- just a quick update that support for the integrated touch driver on RA8875 has now been added. Please see example config `ard-adagfx-ra8875-simple_int`.