ImpulseAdventure/GUIslice #104: Support for ILI9486 TFT LCD

au42bl, 2019-01-27:
Hello,

I recently purchased a touch screen module to go with my Arduino Mega 2560. The demo documentation was easy enough to follow, I uploaded a TFTpaint program demo, the GLUE demo, as well as a mock phone dialer program. My initial hopes were to utilize some drawing utilities in conjunction with the needed touch libraries in order to create a custom GUI for an automation project I am working on. Over the last couple of nights I have been exploring different display driver libraries for the Arduino and it would seem that GUIslice is exceptionally well-designed. However, there is no native support for my LCD. I am quite new to modifying source code and attempting to configure the library file correctly to match my screen has proven to be quite a challenge.

I will include the sample files for my LCD screen. This will provide information regarding library supports required to drive the display. Also, I will present all library files which I believe are necessary to drive the display, including the required raw GUIslice library files which must be configured accordingly to hardware of my LCD Screen. I will include a photo of the back of the module with the pins and descriptors as well. Lastly, I will provide the modified library files which I have edited to try and interface with the pins of my LCD screen and the GPIOs of my MEGA board.

This is the original raw GUIslice src zip repository. This is used in starting from scratch to configure for my LCD:

[GUIslice-master.zip](https://github.com/ImpulseAdventure/GUIslice/files/2800387/GUIslice-master.zip)

These are all the modified files and needed libraries to run MY LCD. There is a sketch included that should have all config.h aspects modified as needed to drive my display:

[myLCD files and examples.zip](https://github.com/ImpulseAdventure/GUIslice/files/2800400/myLCD.files.and.examples.zip)

After studying some of the test results of different models and settings provided via an Excel doc located at the bottom of this post, I looked at the source code for the libraries used in driving myLCD examples to find that Adafruit_TFTLCD.h is called to within the driver layer for Adafruit-GFX of GUIslice_drv_adagfx.cpp 

as per the following quote:

" Driver Layer for Adafruit-GFX
// =======================================================================

// GUIslice library
#include "GUIslice_drv_adagfx.h"

#include <stdio.h>

#if defined(DRV_DISP_ADAGFX)
  #include <Adafruit_GFX.h>
  #include <gfxfont.h>
#elif defined(DRV_DISP_ADAGFX_AS)
  #include <Adafruit_GFX_AS.h>
#endif

#if defined(DRV_DISP_ADAGFX_ILI9341)
  #include <Adafruit_ILI9341.h>
  #if (GSLC_SD_EN)
    #include <SD.h>   // Include support for SD card access
  #endif
  #include <SPI.h>
#elif defined(DRV_DISP_ADAGFX_ILI9341_8BIT)
  #include <Adafruit_TFTLCD.h>
  #if (GSLC_SD_EN)

...

can this be reconfigured as:

 Driver Layer for Adafruit-GFX
// =======================================================================

// GUIslice library
#include "GUIslice_drv_adagfx.h"

#include <stdio.h>

#if defined(DRV_DISP_ADAGFX)
  #include <Adafruit_GFX.h>
  #include <gfxfont.h>
#elif defined(DRV_DISP_ADAGFX_AS)
  #include <Adafruit_GFX_AS.h>
#endif

#if defined(DRV_DISP_ADAGFX_ILI9341)
  #include <Adafruit_TFTLCD.h>
  #if (GSLC_SD_EN)
    #include <SD.h>   // Include support for SD card access
  #endif
  #include <SPI.h>
#elif defined(DRV_DISP_ADAGFX_ILI9341_8BIT)
  #include <Adafruit_TFTLCD.h>
  #if (GSLC_SD_EN)

and would that allow myLCD to properly accessed by the Arduino? 

I know that it can be driven using the Adafruit_TFTLCD.h library. and Adafruit_GFX.h library. --both of which are embedded in GUIslice-- srcTouchScreen.h is also called out in myLCD example code but I do not recall whether it is embedded in GUIslice through reference in GUIslice.h

this is the guide with test results of different screens, boards, and parameters--from GUIslice data files
config guide:

[GUIslice_config_guide.xlsx](https://github.com/ImpulseAdventure/GUIslice/files/2800402/GUIslice_config_guide.xlsx)"

Here is the technical datasheets of myLCD:

[Technical Data about myLCD.zip](https://github.com/ImpulseAdventure/GUIslice/files/2800441/Technical.Data.about.myLCD.zip)

Presently, I don't have time to dig into this further. Working a lot and doing family. I would like to avoid purchasing a new display simply so that it is out-of-the-box compatible with GUIslice when I know that myLCD can be configured with the application. the GUIslice Builder cross-platform app looks incredible and I have already utilized it as a way to speed up the process of compiling code to my device. Adding support for this display could open the doors up for a lot of new development using GUIslice on other screens with the same hardware and pinout. My experience is currently limited and I would not be able to create a GUI so quickly were it not for the help of GUIslice. This platform is going to become my mainstay for navigating the user interface of my project. If nothing else, someone please help send me in the right direction. I have been through these files enough that my head is spinning and I'm at a loss for progress. SD access is not imperative, however, touch and graphic support are. Currently, my screen is white when the code compiles. I can only assume that something is not correct with the configuration files and I just feel as though I've been chasing my tail.
 
This is a picture of the back of myLCD:

[image]

Sincerely,

Au42

ImpulseAdventure, 2019-01-27:
Thank you @au42bl very much for the detailed supporting info, it was greatly appreciated!

Yes, I plan on adding support for your 8-bit parallel ILI9486 TFT shield to GUIslice. I had actually ordered an identical display to what you have shown, but unfortunately it arrived with a cracked screen, so I'm currently waiting on the replacement. I might be able to create a beta version for you to test out in advance of my test module arriving.

Thanks!

ImpulseAdventure, 2019-01-28:
That is correct! As Paul points out, the current release of GUIslice should already support this display using the **mcufriend** display driver (I just confirmed this last night with my broken display). However, I was not able to test the touch-to-display mapping for this particular shield until the replacement part arrives.

So, you should be able to use your display by doing the following:
- Ensure the supporting libraries have been installed via the Arduino Library Manager: **mcufriend_kbv** and **Adafruit_Touchscreen**
- Modify your `/src/GUIslice_config.h` to select (uncomment) the entry: `#include "../configs/ard-adagfx-mcufriend-simple.h`

As mentioned above, it is quite possible that the default touch mapping may be incorrect, so perhaps Paul may be able to indicate the touch config settings he found worked, ie:
- `ADATOUCH_X_MIN / X_MAX / Y_MIN / Y_MAX`
- `GSLC_TOUCH_ROTATE`

Note that I have completely revised the touch handling capability in the upcoming #98 , and it will include a calibration utility to detect all the settings you need.

Thanks!

Pconti31, 2019-01-28:
My settings won't work well for you but MCUFriend_kbv comes with example files.
 You Need to run TouchScreen_Calibr_native first to get your  touch screen settings.
The output it will look something like 
 ```
 const int XP = 6, XM = A2, YP = A1, YM = 7;
 const int TS_LEFT = 907, TS_RT = 136, TS_TOP = 942, TS_BOT = 139;
 ```
Mapping the results to GUIslice involves editing  configs/ard-adagfx-mcufriend-simple.h
Find the Calibration for "DRV_TOUCH_STMPE610 & DRV_TOUCH_SIMPLE" and map
 
 ```
 #define ADATOUCH_X_MIN {TS_LEFT value}
 #define ADATOUCH_Y_MIN {TS_TOP value}
 #define ADATOUCH_X_MAX {TS_RT value}
 #define ADATOUCH_Y_MAX {TS_BOT value}
 ``` 
 Then rename ard-adagfx-mcufriend-simple.h to GUIslice_config_ard.h and copy it to GUIslice/src

I hope that helps you out.
Paul

ImpulseAdventure, 2019-01-29:
Hi @au42bl -- if you can try this out and confirm whether it works, that would be great.

- Copy `configs/ard-adagfx-mcufriend-simple.h` to `configs/my_tft35.h`
- Modify `configs/my_tft35.h` by changing:
```c++
#define ADATOUCH_X_MIN 907
#define ADATOUCH_Y_MIN 942
#define ADATOUCH_X_MAX 136
#define ADATOUCH_Y_MAX 139
```
- Modify `/src/GUIslice_config.h` to add the following line: (ensure no other files in the `/configs` section are uncommented out)
```c++
#include "../configs/my_tft35.h`
```
Thanks!

au42bl, 2019-02-02:
To be clear:

I am supposed to go to the repository online and download the file **configs/ard-adagfx-mcufriend-simple.h** and rename to configs/my_tft35.h

then...

Modify configs/my_tft35.h by changing:
#define ADATOUCH_X_MIN 907
#define ADATOUCH_Y_MIN 942
#define ADATOUCH_X_MAX 136
#define ADATOUCH_Y_MAX 139

Modify /src/GUIslice_config.h to add the following line: (ensure no other files in the EXAMPLE CONFIGURATIONS section are uncommented)

uncomment everything?

#include "../configs/my_tft35.h"

did this. 

compile error

// FILE: test.ino
//
// GUIslice Builder Generated File
//
// ARDUINO NOTES:
// - GUIslice_config.h must be edited to match the pinout connections
//   between the Arduino CPU and the display controller (see ADAGFX_PIN_*).
//

#include "GUIslice.h"
#include "GUIslice_ex.h"
#include "GUIslice_drv.h"

#include <Adafruit_GFX.h>

// Note that font files are located within the Adafruit-GFX library folder:
//<Fonts !Start!>
//<Fonts !End!>

// ------------------------------------------------
// Defines for resources
// ------------------------------------------------
//<Resources !Start!>
//<Resources !End!>

// ------------------------------------------------
// Enumerations for pages, elements, fonts, images
// ------------------------------------------------
//<Enum !Start!>
enum {E_PG_MAIN,E_PG2,E_PG3};
enum {E_BTN1,E_BTN2,E_BTN4};
enum {E_FONT_TXT1};
//<Enum !End!>

// ------------------------------------------------
// Instantiate the GUI
// ------------------------------------------------

// Define the maximum number of elements per page
//<ElementDefines !Start!>
#define MAX_PAGE                3
#define MAX_FONT                1
#define MAX_ELEM_PG_MAIN        1
#define MAX_ELEM_PG_MAIN_RAM MAX_ELEM_PG_MAIN
#define MAX_ELEM_PG_EXTRA1      1
#define MAX_ELEM_PG_EXTRA_RAM1 MAX_ELEM_PG_EXTRA1
#define MAX_ELEM_PG_EXTRA2      1
#define MAX_ELEM_PG_EXTRA_RAM2 MAX_ELEM_PG_EXTRA2
[... 530 lines cut ...]
  #include "GUIslice_config_ard.h"
#elif defined(NRF52)
  #include "GUIslice_config_ard.h"
#elif defined(ARDUINO_STM32_FEATHER) || defined(__STM32F1__)
  #include "GUIslice_config_ard.h"
#elif defined(ARDUINO_ARCH_STM32)  // ST Core from STMicroelectronics
  #include "GUIslice_config_ard.h"
#elif defined(ARDUINO_ARCH_SAM)   // DUE
  #include "GUIslice_config_ard.h"
#elif defined(ARDUINO_ARCH_SAMD)   // M0_PRO
  #include "GUIslice_config_ard.h"
#else
  #error "Unknown device platform"
#endif

// -----------------------------------------------------------------------------------------

#ifdef __cplusplus
}
#endif // __cplusplus
#endif // _GUISLICE_CONFIG_H_

...

unsure.

ImpulseAdventure, 2019-02-02:
Hi @au42bl --

Thank you for supplying your config and code examples... very helpful!

The changes you made to the `my_tft35.h` file looked fine to me.
There were two issues that were likely causing your compiler error:
- `GUIslice_config.h`: We only wanted to add a single line `#include "../configs/my_tft35.h"` and leave everything else as-is (ie. uncommented). It seems that in your version all other include lines were uncommented. My apologies if my suggestion was unclear.
- `test.ino`: It looks like your generated file had the Builder issue that Paul had reported in https://github.com/ImpulseAdventure/GUIslice/issues/95#issuecomment-458170894 causing duplicate variables. This has been fixed in the upcoming Builder version.

I have fixed the above for you and attached as `files190202_fix1.zip`. Please unzip these files and replace the ones you had with these. On my side, these appear to compile OK.

[files190202_fix1.zip](https://github.com/ImpulseAdventure/GUIslice/files/2824695/files190202_fix1.zip)

Lastly, please ensure that you are using the latest release of GUIslice from GitHub.
- Refer to [Installing GUIslice from GitHub](https://github.com/ImpulseAdventure/GUIslice/wiki/Installing-GUIslice-on-Arduino#installing-from-github-source)

Hope that helps!

au42bl, 2019-02-03:
Calvin,

Firstly, I thank you tremendously for your help. I am working on overtime currently and have a family. Things are far from easy around here and you have really made this driver support issue come together relatively easy for me.

Second: WE HAVE GRAPHICS SUPPORT! 

Third: No touch support. So long as my GUI Builder sketch is drawn properly, I should be able to touch the buttons and cycle through pages 1-3 continuously. 1:2 2:3 3:1 and so on. No such luck.

I decided to rewrite the simple sketch and drop it back into my "test" folder to see if that might change something. Ran into that duplicate variable error. I'm not an experienced programmer so I don't quite know what all is happening there. My thoughts were that perhaps your test.ino sketch was different than my original. It looks as though the button placement does not match my saved file. On the .ino file you zipped, I have no touch support. Forgot about that error.

This is great news. However, the touch support is key in utilizing your platform as intended.

ImpulseAdventure, 2019-02-03:
That is great to hear that the display is working for you.

As for the touch support on your 4-wire resistive display, we need to ensure that both the pin settings and the calibration settings are matched to your display. The latest development version of GUIslice includes some [diagnostic sketches](https://github.com/ImpulseAdventure/GUIslice/wiki/Configure-Touch-Support) to do this, but for now, please paste in the output from David Prentice's **MCUFRIEND_kbv** library [TouchScreen_Calibr_native](https://github.com/prenticedavid/MCUFRIEND_kbv/blob/master/examples/TouchScreen_Calibr_native/TouchScreen_Calibr_native.ino) sketch. Have the serial monitor window open before you start.

Regarding the duplicate variable, I am including examples below that show how you can fix it (in advance of the updated Builder being available). Please see the text that has been added in dark red:

[image]
 
[image]
 
[image]

I am also attaching a slightly modified version of the "fix1" from the one I included in my previous update. This one makes the buttons advance to the next pages, which I believe was the intent of your original code.

[files190202_fix2.zip](https://github.com/ImpulseAdventure/GUIslice/files/2825226/files190202_fix2.zip)

Cal

au42bl, 2019-02-03:
Very good, sir.

I will be reviewing these changes and soon moving into programming with GUISlice. Correct on your assumption of page-advance. Not sure if I had not selected that within the builder. I will address these issues and be back in touch. Thank you.

au42bl, 2019-02-03:
I am actually confused about what you have instructed me to do with the MCUFRIEND_kbv library.

ImpulseAdventure, 2019-02-03:
Hi -- the [MCUFRIEND_kbv](https://github.com/prenticedavid/MCUFRIEND_kbv) library includes a sketch that will test and report both the pin settings and also the calibration settings suitable for your display. So, just open the serial monitor in the IDE, select **File → Examples → MCUFRIEND_kbv → TouchScreen_Calibr_native** and then Upload. Follow the directions and then copy & paste the resulting output into this issue, I can convert those settings into values that we can use in your `my_tft35.h` GUIslice configuration. Hope that helps clarify the recommendation.

au42bl, 2019-02-03:
Oh, yes that makes good sense. I understand that.

au42bl, 2019-02-03:
This worked. So, now how to translate this touch support to GUIslice integration?

au42bl, 2019-02-03:
TouchScreen.h GFX Calibration
Making all control and bus pins INPUT_PULLUP
Typical 30k Analog pullup with corresponding pin
would read low when digital is written LOW
e.g. reads ~25 for 300R X direction
e.g. reads ~30 for 500R Y direction

Testing : (A2, D8) = 21
Testing : (A3, D9) = 31
Diagnosing as:-
XM,XP:  (A2, D8) = 21
YP,YM:  (A3, D9) = 31
ID = 0x9486

ImpulseAdventure, 2019-02-03:
Thanks... was there any more output from the mcufriend calibration sketch? Usually after reporting the diagnostic lines you posted it then presents you a series of crosshair targets to touch. After clicking on all 8 targets it should output the calibration settings.

Once we have your calibration settings, please post them here. I (or you) could then convert them into GUIslice config according to my documentation on [Importing Calibration from mcufriend_kbv](https://github.com/ImpulseAdventure/GUIslice/wiki/Touch-Calibration-Import#mcufriend_kbv-calibration).

au42bl, 2019-02-03:
Sorry, was pushing it last night. Falling over trying to get this up and running so I can start development of source and integration of peripherals. Here I have to calibration settings. 

MCUFRIEND_kbv  ID= x9486   320 x 480

PORTRAIT CALIBRATION 320 x 480
x = map(p.y, LEFT=521, RT=499, 0, 320)
y = map(p.x, TOP=552, BOT=564, 0, 480)

Touch Pin Wiring XP=8 XM=A2 YP=A3 YM=9

LANDSCAPE CALIBRATION 480 X 320
x = map(p.y, LEFT=552, RT=564, 0, 480)
y = map(p.x, TOP=499, BOT=521, 0, 320)

That is the rest of the information. 

I have a feeling that we are very near the end of this case. As soon as I am able to achieve touch support, that is it. The process has been simplified with your help but still is very in-depth. A lot of small changes.

For now, I have provided you the information. In the event that I am unable to configure the settings immediately, at least the information is here now.

au42bl, 2019-02-03:
Cant seem to copy Serial Monitor output on calibration settings. Plugged them into config file still no luck. Here is updated file. 

#ifndef _GUISLICE_CONFIG_ARD_H_
#define _GUISLICE_CONFIG_ARD_H_

// =============================================================================
// GUIslice library (example user configuration #???) for:
//   - CPU:     Arduino UNO (ATmega328P)
//              Arduino Mega2560 (ATmega2560)
//   - Display: mcufriend (various)
//   - Touch:   Simple Analog (Resistive)
//   - Wiring:  Uno/MEGA shield
//              - Display pinout defined by mcufriend_kbv library
//              - Touch pinout:
//                  CPU     Touch
//                  ----    -------
//                  A1      YP / Y+
//                  A2      XM / X-
//                  7       YM / Y-
//                  6       XP / X+
//
//   - Example display:
//     - 
//
// DIRECTIONS:
// - To use this example configuration, include in "GUIslice_config.h"
//
// WIRING:
// - The pinout configuration may need to be modified to match your wiring
//
// =============================================================================
// - Calvin Hass
// - https://github.com/ImpulseAdventure/GUIslice
// =============================================================================
//
// The MIT License
//
// Copyright 2016-2019 Calvin Hass
//
// Permission is hereby granted, free of charge, to any person obtaining a copy
// of this software and associated documentation files (the "Software"), to deal
// in the Software without restriction, including without limitation the rights
// to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
// copies of the Software, and to permit persons to whom the Software is
// furnished to do so, subject to the following conditions:
//
// The above copyright notice and this permission notice shall be included in
// all copies or substantial portions of the Software.
//
// THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
// IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
// FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
// AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
// LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
// OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN
// THE SOFTWARE.
//
// =============================================================================
// \file GUIslice_config_ard.h

// =============================================================================
// User Configuration
// - This file can be modified by the user to match the
//   intended target configuration
// =============================================================================

#ifdef __cplusplus
extern "C" {
#endif // __cplusplus

[... 150 lines cut ...]

  #define GSLC_USE_FLOAT      0   // 1=Use floating pt library, 0=Fixed-point lookup tables

  #define GSLC_DEV_TOUCH ""
  #define GSLC_USE_PROGMEM 1

  #define GSLC_LOCAL_STR      0   // 1=Use local strings (in element array), 0=External
  #define GSLC_LOCAL_STR_LEN  30  // Max string length of text elements

  // -----------------------------------------------------------------------------
  // Debug diagnostic modes
  // -----------------------------------------------------------------------------
  // - Uncomment any of the following to enable specific debug modes
  //#define DBG_LOG           // Enable debugging log output
  //#define DBG_TOUCH         // Enable debugging of touch-presses
  //#define DBG_FRAME_RATE    // Enable diagnostic frame rate reporting
  //#define DBG_DRAW_IMM      // Enable immediate rendering of drawing primitives
  //#define DBG_DRIVER        // Enable graphics driver debug reporting

  // =============================================================================

#ifdef __cplusplus
}
#endif // __cplusplus
#endif // _GUISLICE_CONFIG_ARD_H_

au42bl, 2019-02-03:
> Thank you @au42bl <https://github.com/au42bl> for the additional info.
>
> If you can send me a quick email to guislice@gmail.com I should be able
> to help narrow down the config for you, as the output from your
> MCUFRIEND_kbv calibration looks a bit surprising. Thanks!

Calvin,

Thank you. I have acknowledged your request to be in touch via email. What
information may I provide you with to help aid in resolving this matter. I
noticed your mytft_36.h file calibration outputs and pin configs did not
match the serial output calibration information I was given during test. I
reconfigured the file with the given information and yet no change. At this
point, I don't know where the problem lies. Am willing to start from basics
and gather everything, put it the way it should be, save it and recompile.
All the information should be available now, the challenge seems to be
properly adjusting the config files. I have this afternoon to correct the
matter. Thank you for your patience.

ImpulseAdventure, 2019-02-03:
Thank you - I will send you a followup email so we can get it working for you.

With displays that I already have on-hand & tested, it should be a simple matter of copying over the appropriate example config. In this particular case I don't have access to the matching hardware to test (my replacement 3.5" shield is still waiting to arrive), so it can sometimes be more challenging. I'll give you some recommendations in the email and then followup later here with the results. thx

au42bl, 2019-02-03:
I have forwarded you a compressed folder with the header files as they
exist currently on my end. If you need more files please let me know. I am
willing to start over with the proper files and reformat them once again
from the start. Now that I know what config files are critical in the
matter and how to obtain the screen information from my device, it should
be a simple matter, as you say. Perhaps I have confused some settings
throughout this process or transcribed something incorrectly. Starting over
may be a good thing now that all the information should be available. At
this point, it should be no harder than going through the provided
information, ensuring correct entry into the config files, and eliminating
any files or information that doesn't pertain. Thanks again for your time
and help. It has been a pleasure working with you.
Brandon L. Luckadoo | CETa
*Associate Certified Electronics Technician, ETA-I*

Brandon L Luckadoo - LinkedIn
<https://www.linkedin.com/in/bluckadoo>Brandon L Luckadoo - Indeed
<https://resumes.indeed.com/resume/0eb97ce9956f6cba>

au42bl, 2019-02-03:
I have forwarded you a compressed folder with the header files as they
exist currently on my end. If you need more files please let me know. I am
willing to start over with the proper files and reformat them once again
from the start. Now that I know what config files are critical in the
matter and how to obtain the screen information from my device, it should
be a simple matter, as you say. Perhaps I have confused some settings
throughout this process or transcribed something incorrectly. Starting over
may be a good thing now that all the information should be available. At
this point, it should be no harder than going through the provided
information, ensuring correct entry into the config files, and eliminating
any files or information that doesn't pertain. Thanks again for your time
and help. It has been a pleasure working with you.

ImpulseAdventure, 2019-02-14:
Meant to followup earlier... I had understood from au42bl that the revised configuration provided by email was working correctly with his ILI9486 display, and all that remained was optimizing the calibration since his resistive overlay hardware had one corner that was not very responsive. With hardware scenarios like this I think the easiest solution is to offer an alternate calibration target location (ie. for this, relocating the calibration point, eg. with `TARGET_MARGIN` might be an option).

I have now created an example config for this particular shield and checked it into the repository as:
`/configs/ard-shld-generic1_35_touch.h`

Thanks again au42bl for all the helpful debug info!

closed 2019-02-14 by ImpulseAdventure