ImpulseAdventure/GUIslice #106: windows 7 32bit

finder0009, 2019-02-06:
Request to support windows 7 32bit

ImpulseAdventure, 2019-02-07:
Hi @finder0009 -- can you confirm for me what functionality you are looking for within Windows 7 32-bit?
- For Arduino targets, GUIslice can be built on the Arduino IDE under Windows (should include Windows XP and up), LINUX or Mac OS X using various TFT graphics drivers.
- For LINUX targets (eg. Raspberry Pi), GUIslice can be built under LINUX using the SDL graphics driver.
- It may be theoretically possible for me to also support a Windows target using the SDL graphics driver, but I'm not sure if that is what you are asking for?

thanks!

finder0009, 2019-02-07:
Hi ,thank you for your response.
the installer support windows 64bit system while mine is 32bit(windows7 32bit) so after installing the program it didn't run. is there any way to run it on my system

Pconti31, 2019-02-07:
I'm the author of the Builder.  I'm sorry but I have no plans to support 32-bit systems.

Now that said it still MIGHT be possible for you to run the builder. 

In theory, if you have installed a Java 1.8 JRE 32-bit on windows 7.  The builder could run on it since its in a java byte code which runs in JRE.  Java byte code generated by java compiler does not change between 32-bit and 64-bit, using the same version of java. (well as long as the app keeps under 4gb memory requirements).

Hence, you can safely use a java application compiled using 64 bit JDK on 32 bit JRE.

So you need to install a Java 1.8 32-bit JRE first.  If you have Arduino IDE you can point
JAVA_HOME to its version,  say, "C:\Program Files\Arduino\java" 
otherwise you need the oracle version:
jre-8u202-windows-i586.exe
available at:
[https://www.oracle.com/technetwork/java/javase/downloads/jre8-downloads-2133155.html](url)

Make sure you have JAVA_HOME environment variable defined pointing to the java's installed folder.
See our wiki for the older install instructions:
[https://github.com/ImpulseAdventure/GUIslice/wiki/Builder-_-Installation-_-Windows-_-old](url)

Now, download the builder-linux64-0.10.4-beta7.tar.gz version and unpack it in a folder of your choice.  Give write access on each unpacked folder and the GUIsliceBuilder directory.

create a GUIslice.bat file inside the GUIsliceBuilder directory.
The GUIslice.bat file should have these lines:
@echo off
set CLASSPATH=.;%JAVA_HOME%\lib
start "" /min "%JAVA_HOME%\bin\javaw" -jar builder-0.10.4-beta7.jar

Now you should be able to start the builder by clicking on the GUIslice.bat file.

Remember I said this might work, no promises.

finder0009, 2019-02-08:
Thanks alot. I'll try that hoping that it will work. Thank you again it is really a great work.

Pconti31, 2019-02-08:
Cal seems to have over done his edits if by post.
You MUST NOT use builder-linux64-0.10.4-beta7.zip
but use
builder-linux64-0.10.4-beta7.tar.gz
instead.  The zip simply contains the install exe which can't run on 32-bit machines
while the linux tar bar has everything you need unpacked.
You just can't use the install.sh script and must move the builder-0.10.4-beta7.jar up into the parent folder GUIsliceBuilder.  I have modified my comments above for this detail.

ImpulseAdventure, 2019-02-08:
Paul: Oops... yes, indeed I had inadvertently swapped in the `win64` release instead of the `linux64` release (as I had wrongly assumed that was a typo). I have corrected (hopefully correctly!) the URLs again, as the links in your recent edit were leading to a 404. Sorry for any confusion and thank you for clarifying the suggested use of the LINUX release for the Windows 32bit operation, as that may not have been obvious :)

finder0009, 2019-02-09:
Hi friends. I want to thank you very much for your patience and cooperation. every think went just fine. I have both arduino and java jdk1.8.0_131. my home java is pointed to C:\Program Files\Java\jdk1.8.0_131
so i followed the steps explained by you with one modification on the bat content : 
@echo off
set CLASSPATH=.;%JAVA_HOME%\lib
start "" /min "%JAVA_HOME%\javaw" -jar builder-0.10.4-beta7.jar
and finally I had the GUIsliceBuilder running on my windows7 32bit system
thanks alot for your help.

finder0009, 2019-02-09:
> class java.lang.ArrayIndexOutOfBoundsException: 14
builder.models.WidgetModel.initProp(WidgetModel.java:131)
builder.models.TextBoxModel.initProperties(TextBoxModel.java:118)
builder.models.TextBoxModel.<init>(TextBoxModel.java:96)
builder.widgets.TextBoxWidget.<init>(TextBoxWidget.java:53)
builder.widgets.WidgetFactory.createWidget(WidgetFactory.java:185)
builder.views.Toolbox.createWidget(Toolbox.java:192)
builder.views.Toolbox.actionPerformed(Toolbox.java:280)
javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
java.awt.Component.processMouseEvent(Component.java:6539)
javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
java.awt.Component.processEvent(Component.java:6304)
java.awt.Container.processEvent(Container.java:2239)
java.awt.Component.dispatchEventImpl(Component.java:4889)
java.awt.Container.dispatchEventImpl(Container.java:2297)
java.awt.Component.dispatchEvent(Component.java:4711)
java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4904)
java.awt.LightweightDispatcher.processMouseEvent(Container.java:4535)
java.awt.LightweightDispatcher.dispatchEvent(Container.java:4476)
java.awt.Container.dispatchEventImpl(Container.java:2283)
java.awt.Window.dispatchEventImpl(Window.java:2746)
java.awt.Component.dispatchEvent(Component.java:4711)
java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
java.awt.EventQueue.access$500(EventQueue.java:97)
java.awt.EventQueue$3.run(EventQueue.java:709)
java.awt.EventQueue$3.run(EventQueue.java:703)
java.security.AccessController.doPrivileged(Native Method)
java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
java.awt.EventQueue$4.run(EventQueue.java:733)
java.awt.EventQueue$4.run(EventQueue.java:731)
java.security.AccessController.doPrivileged(Native Method)
java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

finder0009, 2019-02-09:
(TextBox) button in the gui builder  indicates a fatal error message producing a logo file whose content is:

`class java.lang.ArrayIndexOutOfBoundsException: 14
builder.models.WidgetModel.initProp(WidgetModel.java:131)
builder.models.TextBoxModel.initProperties(TextBoxModel.java:118)
builder.models.TextBoxModel.<init>(TextBoxModel.java:96)
builder.widgets.TextBoxWidget.<init>(TextBoxWidget.java:53)
builder.widgets.WidgetFactory.createWidget(WidgetFactory.java:185)
builder.views.Toolbox.createWidget(Toolbox.java:192)
builder.views.Toolbox.actionPerformed(Toolbox.java:280)
javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
java.awt.Component.processMouseEvent(Component.java:6539)
javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
java.awt.Component.processEvent(Component.java:6304)
java.awt.Container.processEvent(Container.java:2239)
java.awt.Component.dispatchEventImpl(Component.java:4889)
java.awt.Container.dispatchEventImpl(Container.java:2297)
java.awt.Component.dispatchEvent(Component.java:4711)
java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4904)
java.awt.LightweightDispatcher.processMouseEvent(Container.java:4535)
java.awt.LightweightDispatcher.dispatchEvent(Container.java:4476)
java.awt.Container.dispatchEventImpl(Container.java:2283)
java.awt.Window.dispatchEventImpl(Window.java:2746)
java.awt.Component.dispatchEvent(Component.java:4711)
java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
java.awt.EventQueue.access$500(EventQueue.java:97)
java.awt.EventQueue$3.run(EventQueue.java:709)
java.awt.EventQueue$3.run(EventQueue.java:703)
java.security.AccessController.doPrivileged(Native Method)
java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
java.awt.EventQueue$4.run(EventQueue.java:733)
java.awt.EventQueue$4.run(EventQueue.java:731)
java.security.AccessController.doPrivileged(Native Method)
java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
`

ImpulseAdventure, 2019-02-09:
Hi - thanks for reporting this. Yes, there was a bug in Builder v0.10.4-beta7’s TextBox component, but it has been fixed in the upcoming release.

For reference, this and other Builder open issues have been noted in 
https://github.com/ImpulseAdventure/GUIslice/issues/80#issuecomment-458825941

ImpulseAdventure, 2019-02-09:
Glad to hear that the Builder is running on Windows 7 32bit for you with Paul’s recommendation. If you find any additional issues, please feel free to open a new ticket. Thanks!

closed 2019-02-09 by ImpulseAdventure