ImpulseAdventure/GUIslice #107: Builder install on Mac

Pconti31, 2019-02-10:
Support Builder install on Mac.
If anyone that has a MAC could document the exact steps they take during an install and the issues that occur it would allow us to create an install.sh script for MACs.
This is the only report so far and its from Vitor Henrique and it's a big help.
Thanks Vitor!

For the builder, to get the location of java on mac you need to do a "/usr/libexec/java_home -v" it will spit out something like: "/Library/Java/JavaVirtualMachines/jdk-9.0.1.jdk/Contents/Home" 

For me it was not clear if I had to put the full path for java with the /bin/java or the home folder. I guess people can get confused, but I tried until got it working. Another thing is that on mac usually shortcuts for applications, or installations are placed on "/Users/{USER_NAME}/Applications/" .

And another important thing to let people know is that the created shortcut is linked to the installation path, so if someone downloads it to the download folder, run it from there, he can not delete it from the download folder.

Actually, the "shortcut" itself does not work on mac, I'm running the application from the terminal, by executing "/Users/vitormhenrique/Applications/GUIsliceBuilder/GUIslice.sh"

I'm guessing, because it's a java app, you don't really need to install it, just downloading and running the terminal command would work.

Pconti31, 2019-02-15:
The install for macos has been completed

ImpulseAdventure, 2019-02-18:
Builder install for Mac OS/X was included in the latest library release (0.11.0).

closed 2019-02-18 by ImpulseAdventure