ImpulseAdventure/GUIslice #108: Touch input X/Y swapped with ILI9884, XPT2046 and TFT_eSPI

thedevleon, 2019-02-12:
I'm using the latest GUISlice with an ILI9844 display (and XPT2046) running via TFT_eSPI on a ESP32.
All the examples from TFT_eSPI work correctly and the touch_calib works too and I can draw on the screen after the calibration and the dots are exactly where I touch the display.
The values from the calibration are even similar to the ones already defined by `TFT_ESPI_TOUCH_CALIB`.

However, when I run the `diag_ard_touch_test` sketch, while I'm getting dots on the screen, they are not where I'm touching the screen. 
When I touch from the middle to the right, the dots will draw from the middle to the top. 
If I touch from the middle to the left, the dots will draw from the middle down.

Seems to me that X and Y are simply swapped. How can I adjust for this in the GUIslice_config_ard.h?
I tried changing GSLC_ROTATE and GSLC_TOUCH_ROTATE to different values, however the result was always the same and nothing changed. What can I do?

thedevleon, 2019-02-12:
I've made some progress. If I set GSLC_ROTATE == GSLC_TOUCH_ROTATE == 1 as well as tft.setRotation(1); in the TFT_eSPI touch_calib sketch, I get the behavior explained above.

However, if I set GSLC_ROTATE == GSLC_TOUCH_ROTATE == 0 and tft.setRotation(0); in the TFT_eSPI and use those calibrations values for TFT_ESPI_TOUCH_CALIB, touch works as intended, but ONLY in Rotate=0 and Rotate=2 in the `diag_ard_touch_test` sketch.
For Rotate=1 and Rotate=2 the touch dots on the screen are weirdly offset a little bit up and left. Touching in the middle will draw the pixels in a top, left position.

ImpulseAdventure, 2019-02-13:
Hi @thedevleon --

Thank you very much for reporting this, and taking the time to identify the behavioral differences between the rotation modes. As you correctly noted, I just completed a major update to the touch & orientation code (in #105), and it is likely that there were a few loose ends on the TFT_eSPI mode side as the majority of the recent testing focused on the ADAGFX mode.

I have now revised the repository to address a couple issues I had found in the TFT_eSPI mode, so if you are able to re-download the latest, that would be great.

Just for background, could you share a link to the touch display you are currently using?
- Despite having 30+ displays on-hand, I'm not sure if any of my ones with an XPT2046 controller are compatible with the TFT_eSPI display library, so I may be unable to fully test this combination at the moment, though will order a module for testing ASAP.

### Recommendations
Please try the following:
- Re-download the latest repository code. Ensure `XPT2046_Touchscreen` has been installed via the Library Manager (though I assume you may have already done so).
- Select the Example Config: `esp-tftespi-default-xpt2046` by uncommenting this line in `GUIslice_config.h`
- Run the GUIslice `diag_ard_touch_calib`. Do touch presses track properly in all 4 orientations?
- Copy the config output of the calibration sketch into SECTION 4B of your `configs/esp-tftespi-default-xpt2046.h` (refer to [Configuring touch for XPT2046](https://github.com/ImpulseAdventure/GUIslice/wiki/Configuring-Touch-for-XPT2046)).
- Run the GUIslice `diag_ard_touch_test` sketch. Do touch presses track properly?
- If anything doesn't appear to work, please:
  - Attach the output of your `diag_ard_touch_calib` sketch to this issue
  - Indicate what appears to happen (as you did here already)
  - Indicate if any other modifications were made to the Example Config

### Background Details

As for your observations:
- In `GUIslice_config.h`, the new [recommended setup](https://github.com/ImpulseAdventure/GUIslice/wiki/Configure-GUIslice-for-ESP8266-&-ESP32) is that one selects an example config by un-commenting one of the supplied configurations (eg. `configs/esp-tftespi-default-xpt2046.h`). Was one of these selected or was this file left unmodified?
  - If none was selected, then the old `GUIslice_config_ard.h` would have been loaded, but this is no longer recommended. The new example configs should be much cleaner and easier to understand (plus have removed options that aren't applicable, such as `GSLC_TOUCH_ROTATE`). For the guidance above, I have *assumed* that you might not have been using the example configs, but please correct me if I'm wrong!
- Note that there are two XPT2046 touch controllers available for TFT_eSPI mode.
  - For ease of configuration & calibration, I recommend we use the **XPT2046_Touchscreen** library instead of the TFT_eSPI's integrated XPT2046 mode. The example config I have listed here will do that automatically. Since we are not using the TFT_eSPI's touch controller, we should ensure it is disabled in Bodmer's library (`#define TOUCH_CS` commented out in TFT_eSPI's `User_Setup.h`)

When you referred to the TFT_eSPI "touch_calib", did you mean GUIslice's `diag_ard_touch_calib` or TFT_eSPI's `Touch_calibrate` sketch?

Thanks!

thedevleon, 2019-02-13:
Thanks for the incredibly detailed answer, I really appreciate your work.

A bit more info about the previous config:
- I'm using a [ER-TFT035-6](https://www.buydisplay.com/default/serial-spi-3-5-inch-tft-lcd-module-in-320x480-optl-touchscreen-ili9488) display
- I had made my configurations inside the `GUIslice_config_ard.h`, I must have missed the example configs!
- I was using the touch integrated into TFT_eSPI and not the XPT2046_Touchscreen driver, thus I was unable to run the `diag_ard_touch_calib` sketch.

Things I done so far:
- Downloaded the latest GUISlice
- Uncommented the `esp-tftespi-default-xpt2046` config, adjusted the `XPT2046_CS` and commented `TOUCH_CS` in User_Setup.h to disable the touch from TFT_eSPI.
- Ran the `diag_ard_touch_calib` sketch, touch now works really well
- Put the calibration values into `esp-tftespi-default-xpt2046.h`
- I then ran `diag_ard_touch_test` and everything works correctly, and touch is much smoother. With the integrated touch driver of TFT_eSPI, the touch points were spaced quite far apart. Now they are pretty much denser and form almost a closed line.

Thanks very much!

ImpulseAdventure, 2019-02-13:
Excellent... thank for the detailed followup! Glad to hear the example config for TFT_eSPI with XPT2046 is working nicely for you.

If you could attach your final settings (ie. `XPT2046_CS`, `X/Y_MIN/MAX` values) to this issue, that would be a useful reference for me to see how your particular display's touch overlay is configured.

I will order one of the **EastRising** displays for future testing purposes; thanks for the link.

Regarding BMP from SPIFFS, I created a new issue #110 to capture the request.

Thanks!

closed 2019-02-13 by ImpulseAdventure

thedevleon, 2019-02-13:
My configuration is as follows:
```
  #define XPT2046_CS     25

  #define ADATOUCH_X_MIN    3835
  #define ADATOUCH_Y_MIN    140
  #define ADATOUCH_X_MAX    285
  #define ADATOUCH_Y_MAX    3940
```

Note that when you get the display, keep in mind that SDI, SDO and SCL from the TFT and XPT2046 are on separate Pins for some reason, even though the TFT and the XPT2046 have their own chip selects. To use them on a single SPI bus, just connect the SCL and SDI from the TFT and XPT2046 together, but make sure only the SDO from the XPT2046 is connected to MISO.
The TFT will always output data, even if the TFT CS is low, and will confuse the XPT2046 and make it unusable. It took me a lot of digging to figure out this issue.
[Bodmer mentions this in his library too](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L27), maybe you can add it to the wiki.

dimafe6, 2019-03-21:
@ImpulseAdventure @thedevleon Thank you very much! I have the same problem and you have helped me a lot!

ImpulseAdventure, 2019-03-22:
Glad to hear that it works for you, Dmytro!