ImpulseAdventure/GUIslice #110: Request: Load images from SPIFFS

ImpulseAdventure, 2019-02-13:
Feature request from @thedevleon in #108 :

> is loading bmp images from spiffs supported with the ESP32? I know that fonts can already be loaded via the SPIFFS (as seen in ex19_ard_smooth), but what about images?

Loading BMP from SPIFFS is not currently supported (SD being the primary source). It should be possible to add support for SPIFFS, ideally extending `gslc_DrvDrawBmp24FromSD()`. I certainly welcome PRs if you're inclined, otherwise I can take a deeper look at it later.

thedevleon, 2019-02-13:
I'm currently busy with exams, but in about a week I could spend some time on it.
It should be rather simple, since all the methods are already there.
Just include this at the top:
```
#define FS_NO_GLOBALS // Avoid conflict with SD library File type definition
#include <FS.h>
#include "SPIFFS.h"
```

And then you can use the methods [described here](http://esp8266.github.io/Arduino/versions/2.0.0/doc/filesystem.html), they are the same for the ESP8266 and ESP32.
There is also an example sketch [here](https://github.com/espressif/arduino-esp32/blob/master/libraries/SPIFFS/examples/SPIFFS_Test/SPIFFS_Test.ino).

ImpulseAdventure, 2019-02-18:
Agreed. While it should be fairly easy to write an equivalent DrvDrawBmp24FromSD() for SPIFFS, it would be less straightforward to integrate if one wasn’t as familiar with the architecture.

As for JPEG rendering, that would be a possibility, especially leveraging the tile-oriented decoding from Bodmer’s library. Keep in mind that rendering JPEGs could be quite a bit slower than rendering BMP due to the need for decompression. If your application could support server-side decoding of the JPEG images into BMP before the ESP, that may be worth considering. Thank you for passing along the references — that’s helpful!

I’ll take a look at this, though it may be a bit before I can get started.

Lucy2003, 2020-02-11:
There is a library [TFT_eFEX](https://github.com/Bodmer/TFT_eFEX) that can draw BMP and JPEG form SPIFFS. Just need to provide a pointer to the TFT _ eSPI object (m_disp).

ImpulseAdventure, 2020-02-11:
Thank you very much @Lucy2003 for pointing out this option!

Pconti31, 2020-03-10:
I have added jpeg and bmp image reading from the SPIFFS file system to my local fork
in branch WIP-SPIFFS.  A sample app has also been added inside examples/arduino/ex44_bld_spiffs.ino.  Since I use the Adruino IDE I also used
the tools->ESP32 Sketch data upload which requires the images be inside a "data" sub folder.
The support uses TFT_eSPI, TFT_eFEX, and the JPEGDecoder libraries.
I will issue a pull request.  
GUIslice API adds a new image type GSLC_IMGREF_FMT_JPG and you access it like so:
```
 #define IMG_LOGO                  "/logo.jpg"
 // Create E_ELEM_IMAGE_LOGO using Image 
  pElemRef = gslc_ElemCreateImg(&m_gui,E_ELEM_IMAGE_LOGO,E_PG_MAIN,(gslc_tsRect){50,13,219,60},
    gslc_GetImageFromFile(IMG_LOGO,GSLC_IMGREF_FMT_JPG));
  gslc_ElemSetFillEn(&m_gui,pElemRef,false);
```
I also added support inside The Builder version 0.13.b020
If your Target Platform is set to "arduino TFT_eSPI" you can select jpg images and the builder
will correctly set the loading.

ImpulseAdventure, 2020-03-15:
Support for this feature has now been merged #206 
Thanks to @Lucy2003 for pointing out the driver and @Pconti31 for adding support.

closed 2020-03-15 by ImpulseAdventure