ImpulseAdventure/GUIslice #125: Arduino DUE Compatability

Uup115, 2019-03-17:
Hello,

Has anyone successfully ported this library to Arduino Due?

I can successfully compile and upload the examples to DUE.  However, most of the time, nothing renders on the LCD.  Example "ex04_ard_ctrls" renders nothing but a couple of small circles.  I've spent nearly a day trying to find the problem.  Sadly, no success.

Does anyone have insight as to how to make this work for Ardiuno DUE?  If yes, please help.

Best Regards

ImpulseAdventure, 2019-03-17:
Hi @Uup115  — thanks for highlighting an issue with Due operation. I don’t have a Due so haven’t had an opportunity to test with one yet, so it is likely there are some compatibility adjustments needed. I will order one right away.

It is good to hear that you are seeing evidence of the radio buttons on the display as that means the basic operations are working.

(1) First, I understand that the Due’s Serial port operation may differ from other targets. We should probably add the following in your setup examples after the `Serial.begin(9600);` line to be sure: `while (!Serial);`

(2) Can you confirm for me:
- What display model (link to the product page would be ideal)
- What config settings you are using? (Version, display and touch modes as reported by GUIslice initialization on the Serial Monitor at startup).
- Assume no errors are reported via the Serial Monitor at startup? (Ie. After the GUIslice initialization messages)

(3) Assuming you are running an ILI9341 display, presumably the Adafruit_ILI9341 `graphicstest` is running properly?

(4) Can you confirm if GUIslice ex01 and ex02 work ok?

I will put in an order for a **Due** so that I can ensure we have them covered in future testing too.

Thanks!

Uup115, 2019-03-17:
Hello Calvin,

I really appreciate your quick and enthusiastic response.  Thanks!

In response to your questions/requests:

1.  **while (!Serial);**  //has no effect
2.  LCD model with shield:  [https://www.buydisplay.com/default/serial-spi-arduino-5-inch-tft-lcd-touch-shield-ra8875-for-mega-due-uno](url) 
Presently, I don't have touch enabled.  Taking one step at a time...
Serial monitor output (no errors reported):  
> GUIslice version [0.11.2]:
> - Initialized display handler [ADA_RA8875(SPI-HW)] OK
>- No touch handler enabled
3.  Adafruit RA8875 test sketches, "buildtest" and "textmode" both successfully render LCD output. 
 Touch feature is also functional.
4.  GUIslice examples, "ex01_ard_basic", "ex02_ard_btn_txt", "ex01_ardmin_basic", and "ex02_ardmin_btn_txt" ALL compile and upload successfully.  However, nothing renders on the LCD.  Serial monitor output is the same, with no errors (see above #2 serial output).

Thanks for your help!

Best Regards

DonpK, 2019-03-17:
Hello Uup115,

As I'm sure Cal will comment, he and I have been in discussion about using GUISlice with the same Buydisplay.com 5" TFT LCD display you're working with.  I, and probably others, look forward to his findings and your efforts as well,
   Don

ImpulseAdventure, 2019-03-17:
Thank you very much for providing the extra details. The issue is more likely associated with the **RA8875** support than the Due. While preliminary support was added for the Adafruit RA8875 driver (in #103 ), I don’t believe it has been tested on hardware yet.

As Don points out, I have already ordered the same EastRising display in order to test & ensure the RA8875 driver is working smoothly, but it may still be a week or two before it arrives.

In the meantime, if you would like to send me a quick email to guislice@gmail.com, I can followup with some test builds that might help us track down the changes we need to get it going for you.

Thanks!

Uup115, 2019-03-18:
Hello Calvin and Don,

I am more than happy to help with testing... whatever I can do to help.  Will follow up with an email.

Best Regards.

ImpulseAdventure, 2019-03-19:
Just a quick update: thanks to testing by @Uup115 and some minor initialization updates, the basic RA8875 display primitive operations are now working with GUIslice (per branch *WIP125-RA8875*). However, additional updates will be required in order to get text rendered correctly (since the RA8875 handles built-in fonts differently from other GFX drivers, especially in "text mode").

At this time it does not appear that the Arduino DUE target has introduced a compatibility issue, but I still intend to add this device to my test suite in the future just in case.

Interesting pointer regarding the calibration... it seems that the buydisplay supplied library is actually an old version from [sumotoy/RA8875](https://github.com/sumotoy/RA8875).

ImpulseAdventure, 2019-03-21:
Update: Latest repo confirmed to be working on **Due** + **RA8875** using the RA8875’s built-in (internal ROM) font. This font is slightly larger than the Adafruit-GFX default font so users may need to adjust some of the text fields in the existing GUIslice examples accordingly (ie. apply a scale factor to coordinates / dimensions).

The next step would be to add support for custom Adafruit-GFX fonts, as well as text justification. From a testing perspective, it may be more practical to defer these updates until after I have the RA8875 **EastRising** display hardware on hand (might still take a couple weeks for its arrival).

Thanks again @Uup115 for the initial testing on this hardware!

ImpulseAdventure, 2019-03-29:
Update: I now have a **5" RA8875 EastRising** display (link: [BuyDisplay.com](https://www.buydisplay.com/default/serial-spi-arduino-5-inch-tft-lcd-touch-shield-ra8875-for-mega-due-uno)) here for testing.
Support has now been added for the integrated touch driver (Adafruit_RA8875).
Example config: `ard-adagfx-ra8875-simple_int`.

This should theoretically work for the 4.3" (480x272), 5" (800x480) and 7" (800x480) displays.

Uup115, 2019-03-29:
Hello Calvin,

Great!  Touch is working nicely.  :-)   Thanks!

Best Regards

ImpulseAdventure, 2019-03-29:
Glad to hear it!

I have now also updated the repo with support for the Adafruit-GFX fonts, including both the default font and custom fonts. The GUIslice examples should now run properly by default (without any scaling necessary).

I will integrate support for the faster (hardware-accelerated) RA8875 ROM fonts in a later update.

The above is using the DRV_DISP_ADA_RA8875 display driver (based on Adafruit_RA8875). I will likely also support the Sumotoy/RA8875 library in a later update, which introduces a number of enhancements and also supports dynamic rotation (not supported by the Adafruit driver)

closed 2019-03-31 by ImpulseAdventure