ImpulseAdventure/GUIslice #126: GUIslice Builder

finder0009, 2019-03-18:
I think  GUIslice_config.h  and related files should be generated automatically not modifies by user .that is to choose the board(arduino) and the lcd from within the builder including assigning pins and then binding commands to each widget and finally generating the code file . that what makes the builder a real user friendly interface not another programing language to be learnt. thank you and best wishes

ImpulseAdventure, 2019-03-19:
Interesting suggestion on the config generation, @finder0009 !

This has actually been something I have contemplated in the past, as it would be nice to have some "configuration wizard" to help establish the necessary configurations according to a list of available display shields, devices and options.

I have considered a few different approaches:
a) Create a web page that enables users to auto-generate their config files
b) Create a standalone utility that generates the config files
c) Integrate the config generation into the GUIslice builder
d) Enhance the existing [TFT Configuration Guide](https://github.com/ImpulseAdventure/GUIslice/wiki/Display-Config-Table) and "example configs"

Option (c) would probably be the most seamless for users, but it does mean introducing a whole new section of the Builder's operation that is likely outside of the scope of the work that Paul has been doing so far on the Builder.

Option (a) is something I could probably put together if there was enough interest. While it wouldn't be an "offline" generator, I expect that the initial config generation would not need to be done that often by users. Once the config has been generated, the user could tweak later as needed.

Option (d) is the approach taken right now. I would like to improve the legibility of the [TFT Display Config Table](https://github.com/ImpulseAdventure/GUIslice/wiki/Display-Config-Table) to make it easier to use (GitHub wiki is somewhat limiting in its table rendering). I would be inclined to enhance this table first before developing a standalone tool.

I'm open to ideas, so please feel free to share your thoughts.

DonpK, 2019-03-19:
I have a couple of initial thoughts. Given the variety of boards and displays available, it would seem that the specific configuration information has to come from the user. This fact would tend point away from Option (d) because there are just too many board/displays for the Builder/GUISlice team to keep up with and include in the TFT Display Config Table. For example, the display I'm currently using is not in the Table.

Perhaps the [Wik](https://github.com/ImpulseAdventure/GUIslice/wiki/GUIslice-Config-Location)i could be improved to provide more detailed assistance for the user. Options a) and b) look to the user to provide information, but I suspect would again require a good deal of work on the part of the Builder/GUISlice team.

neu-rah, 2019-03-19:
we all face the same problems, maybe we should talk more :D

ImpulseAdventure, 2019-03-20:
Thanks for sharing your comments on this, Don!

@neu-rah -- you're right... I am sure you have learnt a lot from your experiences developing numerous adapter layers for the [ArduinoMenu](https://github.com/neu-rah/ArduinoMenu) library. No doubt bodmer and prenticedavid have also spent a good deal of time on config issues too. Unlike other environments, the Arduino IDE is pretty limiting when it comes to sketch build / config, so people end up devising "clever" workarounds :)

I was thinking it might be worth stepping back to see what problems we'd like to solve before we get into the implementation details...

### User challenges
For users attempting to configure and implement a GUI, there are a number of areas that are likely to present a challenge. I have listed a few that I *assume* could be challenging (this applies to all libraries, not just GUIslice):

- (1) Identifying the display and determining which display driver & touch driver ICs are integrated (Just look at the vast efforts by a few extremely helpful individuals in the Arduino forum assisting new users).
- (2) Selecting an appropriate library to drive the display, given the controller ICs.
- (3) How to configure the display driver & touch driver (is it hardware SPI, I2C, etc.) according to the selected libraries
- (4) For breakout boards, what is a valid/preferred pinout to use for a given MCU
- (5) What special features / modes should be enabled for the GUI
- (6) For certain touch-enabled displays, having to identify the correct resistive touch signals and calibrating
- (7) How to develop code to drive the display using a GUI library
- (8) Designing the GUI layout
- (?) Anything else I missed?

One of GUIslice’s primary objectives was to simplify (3) by creating a single configuration set that applies across all display & touch drivers and (7) by creating a common API framework across all driver libraries. The GUIslice Builder attempts to streamline (7) further by auto-generating the boilerplate GUI framework code and enabling a visual drag & drop editor (8).

With the move to leverage “example configurations” and the accompanying “TFT config table”, the hope was to offer some assistance on (1) and (2) by providing a set of configurations for display shields and some popular breakout boards.

### Config generator
I had contemplated the idea of some “generator” tool that could take the existing “config tables” (with additional expansion), and then, through a set of choices for the user, guide the creation of a custom config file. This would contrast from the current approach of identifying a matching “example config” according to the TFT config table.

As Don points out, it would be unrealistic to maintain a full list of possible displays / boards... We can classify displays into shields and breakout boards. In the case of shields, I think it might be feasible to have a reasonably comprehensive list (expanded by community input). However, for breakout boards, it would only be practical to track the popular boards.

### Wiki documentation
Don -- I agree that there are definitely areas in the wiki that could be revised. Perhaps I can open up a separate issue to track areas that people think deserve more attention.

### Next steps
Returning to the original question: **which step(s) listed above do you feel could benefit from improved ease-of-use?**

Thanks!

ImpulseAdventure, 2019-04-19:
Tagging as *Open Future Enhancement*. If there is renewed interest in exploring further enhancements in this area, I would be happy to reopen the issue.

closed 2019-04-19 by ImpulseAdventure