ImpulseAdventure/GUIslice #130: Touchscreen support with 9341 8-bit parallel on ESP32

benton-anderson, 2019-04-08:
I'm trying to get touchscreen support on the ESP32 on my 9341 8-bit parallel UNO shield with your GUI builder. I can get the GUI to display, but touch doesn't work. These three issues make it ambiguous to me whether parallel will work with TFT_eSPI and your library:

https://github.com/ImpulseAdventure/GUIslice/issues/61
https://github.com/Bodmer/TFT_eSPI/issues/106
https://github.com/Bodmer/TFT_eSPI/issues/245

I'm looking through GUIslice_config.ard.h and it has the 9341 8-bit parallel option, but nowhere to define the digital pins. Any advice is appreciated.

ImpulseAdventure, 2019-04-08:
Hi Ben —

Thanks for providing the details and a link to the shield; that’s very helpful!

It looks like your shield will use a 4-wire (simple/analog) touch driver. To your point, I see now that we are missing an example config (listed in GUIslice_config.h) for the TFT_eSPI display driver with the 4-wire touch driver. I will add this as “esp-tftespi-default-simple” and notify when it is ready.

If you are able to use Bodmer’s TFT_eSPI library with your 8-bit ILI9341 then we should be able to get GUIslice also working with it and also add the 4-wire resistive touch. The 8bit data interface will be defined by the TFT_D0...TFT_D7 defines in the TFT_eSPI library’s User_Setup.

To ensure I have the details correct, I had a couple questions:
- 1: When you say that you can get the GUI to display, did you mean with GUIslice running the examples (eg. ex02)?
- 2: if so, what GUIslice_config.h option were you using? (Eg. esp-tftespi-default-notouch?)
- 3: In your TFT_eSPI library, what config are you using? (Eg. Setup14_ILI9341_Parallel.h?)

Thanks!

ImpulseAdventure, 2019-04-08:
Hi @benton-anderson -- I have now added an example config for **TFT_eSPI** + **4-wire touch**:
Example config: `esp-tftespi-default-simple`

Notes:
- You will need to modify Section 4A in the config to adjust your pinout to match your shield connections. I selected some arbitrary ESP32 ADC-enabled pins for XM/YM/XP/YP, but would be interested in hearing what pins your shield requires.
- Could you clarify for me if you are using an adapter to interface between the ESP32 and the UNO shield you provided a link to? If so, providing a link to the adapter would be useful too.
- The 4-wire resistive touchscreen will likely require calibration (`diag_ard_touch_calib`). The calibration results would be saved into Section 4B of the config.

In this update, I also altered the touch initialization for ESP32 to reduce the ADC resolution to 10-bit in order to provide compatibility with the Adafruit_Touchscreen library (thanks to @lukasfischer83 identifying this in https://github.com/adafruit/Adafruit_TouchScreen/issues/15 ). I haven't attempted to test this yet, so would be interested in hearing if this change enables us to use Adafruit_Touchscreen on the ESP32 (by default, it is normally not compatible).

thx

benton-anderson, 2019-04-08:
Thank you for the prompt reply! I will update and answer your questions ASAP.  I'm very excited to hear I can use your amazing GUI builder.

ImpulseAdventure, 2019-04-10:
Thank you Ben for the update.

### ESP32 + 4-wire fix
The fix for the ESP32 + Adafruit_Touchscreen (4-wire) mode has been pushed to the repo -- thanks for catching this! That combination was only just added but still not in the test regression hence it was overlooked; now it has been included. The latest GUIslice should not require mods to lines 1075 & 1081.

### Touch testing
You had mentioned that the **mcufriend_kbv** library's `Touch_shield_new` (ie. without using GUIslice) is not properly responding to touch on your display. Ideally, that should be resolved before attempting to use any touch functionality in GUIslice. Did you mean that your display correctly tracks touch presses in the right side of the display, but nowhere else (ie. a broken touch overlay), or that touch presses seem to map only to the right side of the display? (ie. a calibration issue)

Putting aside touch for the moment, let's focus on using the display-only `esp-tftespi-default-notouch` mode.

### Display testing
I assume your current TFT_eSPI `Setup14_ILI9341_Parallel.h` is working on your DOIT Esp32 DevKit + display with TFT_eSPI's `TFT_graphicstest_one_lib` example?

With regards to `ex02`, let's start by confirming `ex01` first (as it doesn't depend on touch).
- Can you confirm that GUIslice ex01 compiles OK without error (assuming you have the 1075/1081 fix above, or downloaded the latest GitHub repo code)
- When you run ex01, do you see any error messages in the serial monitor window? Please copy the output you see to this issue.

thanks!

benton-anderson, 2019-04-10:
The touch functions worked on my Mega a few days ago and I doubt it has broken since then (I will retest on the mega to be sure). I calibrated the display on the Mega using the mcufriend calibrate sketch, and I copied over the same 130-ish/900-ish values into the ESP32-connected setup because I assumed the calibration values were screen-specific. And because the touches mapped only to the one side of the screen, I assumed that meant it was a voltage issue, because the Mega does 5V and the ESP pins do 3.3V. 

But now that I think of it, when testing the working side of the screen with the mcufriend touch_shield_new sketch, the read-out of x= and y= values goes from x=~200 to x=~580, and y=~800 to y=~2500 (those values appear when using the values of ~130/~900 entered into the sketch). Based on those odd numbers, this also strikes me as a voltage level issue, not something that calibration can fix...? I'm unsure.

I understand that this question is outside of the scope of your software, so I'd be happy to post this on the Arduino forums or on the relevant tft_eSPI/mcufriend/adafruit_touchscreen github pages. I just need to find out which one because I'm not even sure what library uses which dependencies anymore! 

I will work on your display tests as soon as possible. Thanks for your help!

ImpulseAdventure, 2019-04-11:
Thanks for the extra details.

I suspect the difference you observed in the **mcufriend_kbv** library’s calibration utility touch handling between the **ATmega2560** and the **ESP32** is probably to do with the ESP’s default 12-bit ADC resolution versus the AVR’s 10-bit. The **Adafruit_Touchscreen** depends on the MCU’s ADC sampling of the 4-wire resistive overlay and assumes a 10-bit range.

To prove this out, I would suggest adding the following line in the `setup()` for your mcufriend_kbv calibration test (when running on your ESP32):
`analogReadResolution(10);`

### DETAILS
If you simply ported the ATmega calibration values to the ESP32 and didn’t change the ESP’s ADC resolution, then I would expect the combination of the Adafruit_Touchscreen library and the calibration remapping logic to result in a compressed & clipped remapping output, leading to a relatively small region in one corner of the display (though I think you mentioned one side). So, the mapping function that previously saw an input range of `130..900` may now see an input range of `520..3600`.

The Adafruit_Touchscreen library (latest release 1.0.1) doesn’t naturally support ESP32, in part due to this difference in ADC resolution. When adding support for this hardware combination, I had added an override for the ESP32 ADC resolution to 10-bit to address this specifically.

So, if you were to use the latest GUIslice (eg. `diag_ard_touch_test`) with your ATmega-derived calibration values on the ESP32, you might be able to see a more complete mapping range (though it sounds like you also have a display issue that would need to be addressed first).

Hope that helps!

benton-anderson, 2019-04-12:
TFT_graphicstest_one_lib works. Ex02 and ex01 both confirmed to work using default-notouch. Changing to default-simple and running ex01 makes it compile and upload just fine, but the screen is white. 

The mcufriend touch_shield_new sketch works perfectly with analogReadResolution(10). The mcufriend Touchscreen_Calbr_native sketch does not detect any touches with analogReadResolution(10). 

default-simple and diag_ard_touch_test compiles and uploads but displays a white screen.

ImpulseAdventure, 2019-04-12:
Thanks for the data points. Good to see `esp-tftespi-default-notouch` is working and that the **mcufriend_kbv/touch_shield_new** works with the `analogReadResolution(10)` update.

As mentioned earlier, the 4-wire Adafruit_Touchscreen library doesn't natively support ESP32, but I still think we should be able to get it working.

I plan to wire up a similar display setup to double-check the config myself, but in order to rule out an initialization error, could you paste in the output from the Serial Monitor when you run ex01 with `tft-tftespi-default-simple` on your ESP32?

thx

benton-anderson, 2019-04-12:
GUIslice version [0.11.2.45]:
- Initialized display handler [TFT_eSPI] OK
- Initialized touch handler [SIMPLE(Analog)] OK

ImpulseAdventure, 2019-04-12:
I have wired up a **SparkFun ESP32 Thing** and selected the `tft-tftespi-default-simple` config, but haven't managed to reproduce the white screen issue yet. Usually that occurs if the display driver failed to initialize, but we already confirmed in tft-tftespi-default-notouch that the display init can be successful.

Can you confirm for me:
- We have updated Section 4A of `esp-tftespi-default-simple` (ie. ADATOUCH_PIN_*) to match your analog touch pin config (ie. XP=12, XM=15, YP=33, YM=13)
- We have disabled (commented-out if it exists) any `TOUCH_CS` in the **TFT_eSPI**'s `Setup14_ILI9341_Parallel.h` (assuming this is the config you are selecting with User_Setup_Select).

If the above are confirmed, then it would be helpful to try the following two tests: (quitting the Arduino IDE before each test)
- Test 1) Uncomment DBG_TOUCH in Section 10 your `tft-tftespi-default-simple.h`. Upload ex02. Despite the white screen, touch the display and see if anything is reported to the Serial Monitor after the Initialization message.
- Test 2) Replace the `GUIslice/src/GUIslice_drv_tft_espi.cpp` file with this attached [GUIslice_drv_tft_espi-mod1.zip](https://github.com/ImpulseAdventure/GUIslice/files/3071753/GUIslice_drv_tft_espi-mod1.zip) version (after unzipping it). Rerun ex02 and check to see if the display is still white. Note that the initialization should report "SIMPLE(Analog)-skip" with this modified version.

To get you up and running faster, it may be easiest to just send me a quick email to guislice@gmail.com so I can attach a revised build for you. thx!

benton-anderson, 2019-04-13:
Confirmed the ADATOUCH_PIN_*, and found no TOUCH_CS in Setup14. 

Restarted Arduino IDE and ran Test 1) and touched the screen in opposite corners to get this: 

```
 GUIslice version [0.11.2.45]:
- Initialized display handler [TFT_eSPI] OK
- Initialized touch handler [SIMPLE(Analog)] OK
DBG: remapX: (829,120,923,0,239)
DBG: remapY: (136,37,904,0,319)
DBG: PreRotate: x=211 y=36
DBG: RotateCfg: remap=1 nSwapXY=1 nFlipX=0 nFlipY=1
DBG: Touch Press=1893 Raw[829,136] Out[36,28]
Trk:                                           (36,28) P=1893 : TouchDown
DBG: Touch End  =0 Raw[829,136] *****
DBG: remapX: (829,120,923,0,239)
DBG: remapY: (136,37,904,0,319)
DBG: PreRotate: x=211 y=36
DBG: RotateCfg: remap=1 nSwapXY=1 nFlipX=0 nFlipY=1
DBG: Touch Press=0 Raw[829,136] Out[36,28]
Trk:                                           (36,28) P=0 : TouchUp
DBG: remapX: (259,120,923,0,239)
DBG: remapY: (847,37,904,0,319)
DBG: PreRotate: x=41 y=298
DBG: RotateCfg: remap=1 nSwapXY=1 nFlipX=0 nFlipY=1
DBG: Touch Press=3497 Raw[259,847] Out[298,198]
Trk:                                           (298,198) P=3497 : TouchDown
DBG: Touch End  =0 Raw[259,847] *****
DBG: remapX: (259,120,923,0,239)
DBG: remapY: (847,37,904,0,319)
DBG: PreRotate: x=41 y=298
DBG: RotateCfg: remap=1 nSwapXY=1 nFlipX=0 nFlipY=1
DBG: Touch Press=0 Raw[259,847] Out[298,198]
Trk:                                           (298,198) P=0 : TouchUp 
```

Restarted Arduino IDE and ran test 2) with your new .cpp file to get:

```
GUIslice version [0.11.2.45]:
- Initialized display handler [TFT_eSPI] OK
- Initialized touch handler [SIMPLE(Analog)-skip] OK

```

ImpulseAdventure, 2019-04-17:
Update:
- I recreated Ben's observations in using ESP32 + TFT_eSPI + Adafruit_TouchScreen + 8-bit parallel ILI9341. The following are some preliminary findings.
- Even without GUIslice, I observe that the Adafruit_TouchScreen read process can sometimes conflict on the ESP32 with the 8-bit parallel TFT (using the TFT_eSPI display driver) when sharing certain pins between the 4-wire touch overlay and the parallel TFT interface. The result is generally a white screen.
- On the mcufriend display (with 8080-style 8-bit parallel interface) that I tested with, the YP touch pin is shared with the LCD_WR control signal.
- Looking at my oscilloscope traces, it appears that the Adafruit_TouchScreen sets YP high during the Y-axis read, and the pin sharing triggers a write strobe (WRX assertion) on the TFT. If LCD_CS is asserted (active low), it would appear that the ILI9341 controller may mistakenly latch a write command, leading to a white screen.
- I am currently looking to see if there are suitable workarounds to enable users to utilize the 4-wire Adafruit_TouchScreen with the 8-bit parallel TFT and TFT_eSPI. I can successfully utilize the mcufriend_kbv display library, so I am exploring if adaptations may be required for TFT_eSPI mode.

Previously, we had marked the Adafruit_TouchScreen as not being compatible with ESP32 mode. With this investigation, I am hoping that we might be able to offer this combination successfully.

ImpulseAdventure, 2019-04-23:
Some followup details:
- Bodmer has recently confirmed that the TFT_eSPI library does not support touch control on these UNO-style shields
- I have now updated GUIslice to include workarounds to support the use of TFT_eSPI with some of these 8-bit shields with 4-wire touch, but fundamentally it won’t work for all displays
- I have successfully tested a mcufriend 8-bit parallel display with ESP32 + TFT_eSPI + 4-wire analog touch (pin sharing) with these workarounds though the variability in the shields may mean this success will not be universal.

As the original problem appears to exist outside of GUIslice, I will close the issue. My recent updates to provide some compatibility workarounds currently exists in a test branch (WIP130-ESP_4wire) and will be merged soon into the master.

closed 2019-04-23 by ImpulseAdventure

hemangjoshi37a, 2024-05-21:
has this been solved yet ? please help me with white screen someone.