ImpulseAdventure/GUIslice #131: Example 1 does not compile and asks for touch library

rtek1000, 2019-04-11:
Hello,

Firstly, congratulations for this project!

I tried to compile the ex01_ard_basic.ino, but it presented error:

> GUIslice\src\GUIslice_drv_adagfx.cpp:120:33: fatal error: Adafruit_STMPE610.h: No such file or directory

But, the heading says:
```
// - Example 01 (Arduino):
//   - Display a box
//   - No touchscreen required <---------------------------------------
```

I found a library: https://github.com/adafruit/Adafruit_STMPE610

But, the library heading says:
"This is a library for the Adafruit STMPE610 Resistive touch screen controller breakout"

It would be great if the project could be compiled without the need for other libraries that apparently should not be being solicited.

Thank you.

ImpulseAdventure, 2019-04-11:
Hi @rtek1000 -- thank you for reporting this!

As described in the header, example **ex01** doesn't require a touchscreen or any touch controller installed.

However, the default GUIslice configuration for Arduino is set up for the common **ILI9341** display and **STMPE610** resistive touch controller combination, resulting in the error message you have seen. I can see why this may have been confusing.

Thanks to you pointing this out, I have now updated the latest repository so that the default Arduino config will use **ILI9341** but disable touch.

Note that, unlike certain Arduino libraries, some configuration is required in GUIslice to match your particular MCU device and display combination. Please refer to [Configuring GUIslice](https://github.com/ImpulseAdventure/GUIslice/wiki/Configuring-GUIslice) for details. Basically, you'll need to identify what type of display you have and then uncomment the appropriate "example config" in the `GUIslice_config.h` file.

If you retry the latest version, you should no longer see any attempt to load a touch driver, though there will still be an assumption that you have Adafruit_ILI9341 installed. If you have a different display, then you will need to select the appropriate driver in the GUIslice_config.

ImpulseAdventure, 2019-04-11:
Closing as default GUIslice_ard_config now disables touch.

closed 2019-04-11 by ImpulseAdventure