ImpulseAdventure/GUIslice #133: Can not change project folder choice (Failure)

rtek1000, 2019-04-11:
Hi,

When you try to save the project, a folder selection window has opened but it does not work, you can not select any other location to save the project (top folder, or any other shortcut such as My Computer).

If I try to create a folder in the current directory, a fatal error message appears ("Failure").

[image]

ImpulseAdventure, 2019-04-11:
Thanks for reporting the Builder save dialog behavior.

In recent versions of the Builder (after 0.11.0-beta4), I understand that the Builder project save dialog was altered to default to the Arduino libraries folder, and enable the creation / selection of any subfolders *below* that. If I recall correctly, Paul (the Builder developer) had implemented it this way to help ensure that round-trip edits between the Builder and the Arduino IDE would be seamless and minimize the chance for accidental overwrites. I'll let him clarify it better as I believe this was an important decision point.

As for creating sub-folders, I just tested it again on my side and didn't observe the error. I assume you have write permission to this folder? If you could email the crash log to `guislice@gmail.com` we should be able to track it down for you.

thanks!

rtek1000, 2019-04-11:
Hi,

I think I've already found the cause of this error,

I am using Windows 10 x64, my Java is x64.

I tried to install the GUI again, but this time I selected the Arduino IDE's Java folder (1.8.9).

And I also selected the C: \ folder for the projects, now I have more freedom, but I still can not choose D: \ and this is horrible. If I needed to select an external drive (pendrive) could no longer save. For it is another letter of unity.

Note: The Arduino IDE is ok, everything normal.

Thank you.

rtek1000, 2019-04-11:
Hi,

I have made use of B4A, a type of code generator for Android, which generates full app, the B4A is well mature already.

This developer has also got another application that works with the Arduino, if it is called B4R, maybe you can talk and join forces, it seems the application of B4R does not have the graphical part, (precisely the strength of B4A).

Take a look: https://www.b4x.com/b4r.html

I commented there in the B4R forum about this project, I hope it will make sure that you join forces, the graphic part for Arduino was what was missing!

https://www.b4x.com/android/forum/threads/no-designer-how-are-tft-screens-made-do-you-know-guislice.104755/

DonpK, 2019-04-11:
I asked previously about user selection of prj/ino folder locations. I received the following response from Paul:

`As long as its in or a sub directory of your registered project folder.  `

What I have done is open the General Tab of the Builder Edit Preferences window and enter the path to my  top-level folder which I call "Arduino" as the new Project Directory. All of my Arduino sketches are various sub-folders under the "Arduino" folder. 

When I start a new sketch/project, I first give a new folder the desired sketch name, say "Blink". This is a sub-folder under my Project Directory folder, "Arduino". When I start a new sketch/project, I open the "Choose your project folder" window in Builder. The "Folder name:" will default to the Project Folder set up in the Edit Preferences window, in my example, "Arduino".   I change this to the new sketch/project folder I just set up, "Blink".

This seems to work okay. I'd be glad to hear if there are problems with this method.

Pconti31, 2019-04-12:
@ritek1000 First I'm open to changing the folder chooser but final say is with Cal.
As for your issue what makes you think you can't use your pen drive?
Open Builder, edit->options
change project directory from say
c:\users\paul\documents\arduino	
to your pen drive I used for a test
f:/foobar
created and compiled a test project (Builder-0.11.0-rc4) it created 
f:/foobar/testpendr/testpendr.prj, gen code
f:/foobar/testpendr/testpendr.ino, compiled and ran fine.

Paul--

ImpulseAdventure, 2019-04-13:
Thank you Don & Paul for the comments regarding the folder selection.

I would definitely be interested in revising the folder chooser as I think it may be more restrictive than necessary (I suspect I might have inadvertently implied more limitations in earlier discussions than intended!).

In my opinion, the important goals would be to ensure we can support easy round-trip edits (Builder <-> Arduino IDE) and minimize the chance of any accidental overwrites (due to lack of file coherency). I believe the only restriction this implies is that we save new Builder projects to a subfolder with the same name as the Builder project (and any generated sketch file) so as to avoid the IDE creating the folder and relocating the sketch. Do we have any other requirements beyond that? For example, a Builder project named `MyCount` would be saved in `MyCount/MyCount.prj`. Theoretically, this subfolder (eg. `MyCount`) could be placed anywhere on the computer / external drive, though defaulting to the Arduino IDE sketch folder might be a convenient default. Hopefully this change wouldn’t be too much of a hassle; I think it would greatly increase flexibility for users.

rtek1000 — thank you for highlighting B4A / B4R and your thoughts on it; I wasn’t aware of the framework and will take a look.

rtek1000, 2019-04-13:
Hi,

I've tried several softwares, since the 2000s, I was confused by this because in the GUI installation, it asks for a project folder.

But if the user can change the folder after the GUI is installed, it helps.

I think it would be less confusing if it worked similarly to Visual Basic 6, Visual Studio, or SharpDevelop (V4.4), or even B4A (which closely resembles the style of VB). Where the project folder is selected at the time of the first save operation, or at the time of new project creation.

I use a government program that makes use of an unfriendly interface to perform the backup, where the only way to navigate between folders is to open folder by folder, as if it were to navigate in Windows regedit, nor has it an option to "paste" the path I prefer. If the user has to open the settings to select a different folder for each project, I think it follows the same pattern: Not very friendly.

I still have not understood the idea of choosing a project folder during the installation. I think it's bad for the user to limit this way, maybe it could just "remind the user to avoid saving in the wrong place".

Thank you.

[image]

[image]

[image]

DonpK, 2019-04-14:
Adding to the last several comments, I tried the following two approaches. I added some questions & comments afterwards.

In Builder "Edit Preferences" window General tab, "Project Directory" is set to _C:\Users\Arduino_
**----Approach 1: Start and Save an ino sketch file first, then save Builder file second.----**
1. Started new sketch and named and saved the ino to a level below "_C:\Users\Arduino_". This automatically created a folder of the same name as the sketch ino file.
2. Opened Builder and added an element.
3. Saved Builder to folder created by Arduino IDE in Step 1. Pressed "Generate Code" which opened a message box indicating success.
4. Opened sketch. Builder had not added any element code.

**---Approach 2: Start and Save Builder file first, then save ino sketch file second.---**
1. Since there is no option in Builder to assign a file name to the .prj file, I first set up a folder using Windows Explorer with the desired project/sketch name as a subfolder of _C:\Users\Arduino_.
2. Created an element with Builder.
3. Used the Builder "Choose your project folder" window to locate the folder I set up in Step 1 and saved the .prj file.
4. Builder created a sketch of the same name and successfully added element code to the sketch ino file.
//-----Comments---
- I understand that for technical reasons all sketches with Builder code must be subfolders of the "Project Directory" folder in Preferences. Somehow I think this requirement needs to be made clearer since, in general, most applications allow the user to place folders and files anywhere they choose.
- I realized from the SimpCount example in the Wiki Walkthrough that Approach 1 apparently won't work, that is, the user can't start with a ino sketch file then added a Builder prj file. Perhaps some sort of error message: "NOTE: Builder project folder and file must be created first. Builder will then create a sketch file of the same name."
- If Approach 2 must be used - Builder first, sketch second - perhaps the Builder Save as Project... window could allow the user to assign the desired file name which would create a .prj file in a folder of the same name. Isn't this the way the Arduino IDE works?

Pconti31, 2019-04-16:
@DonpK I've removed the restriction that you must create projects in "projects folder" in release
builder rc5.  The need to create sub folders will still exist.  The windows installer will still ask
for a project directory.   The file chooser has to open up somewhere after all.  Don't like the starting
spot you can change it later.

The reason for creating sub folders first is due to experience with builder betas 1-4.
They allowed project ".prj" files to be created first, anywhere you wanted.
The problem came up later.   compile the generated .ino file and arduino refused unless it was allowed
to place it inside a sub folder of the same name.  Agree and the .ino file is moved into the new
sub folder.  It gets compiled and runs and eveything is fine.  Unless you want to go back to the builder
and make some changes.   The ".prj" file isn't copied by the Arduino IDE so it's now one level up from your ".ino" file.   Open builder, find your ".prj" file open it and make some change, generate code
and it's now created a new ".ino" file above the new sub folder Arduino IDE created.  You now have two
".ino" files.  

My idea of the builder was to make things simple and idiot proof as possible  for a new user not to target senior software engineers.   So I made my choices with that in mind.

rtek1000, 2019-04-16:
Hi,

Ok thanks for commenting,

[Off topic]:
I would like to suggest this same kind of possibility to create graphical part, for Python Tk and for Processing.

(Are different projects, but similar to B4A (Android), B4J (Java), B4I (IOS). It can be GUIslice for Processing, GUIslice for Python Tk).

For Python Tk, maybe it's off course, but can turn out to be a very accessed project, mainly by the Raspberry Pi board community. And Processing is directly related to Arduino project.

See:
[Create Button in Processing]

> Button.
> 
> (Goal: Click on one of the colored shapes in the center of the image to change the color of the background.)
> 
>  
> int rectX, rectY;      // Position of square button
> int circleX, circleY;  // Position of circle button
> int rectSize = 90;     // Diameter of rect
> int circleSize = 93;   // Diameter of circle
> color rectColor, circleColor, baseColor;
> color rectHighlight, circleHighlight;
> color currentColor;
> boolean rectOver = false;
> boolean circleOver = false;
> 
> void setup() {
>   size(640, 360);
>   rectColor = color(0);
>   rectHighlight = color(51);
>   circleColor = color(255);
>   circleHighlight = color(204);
>   baseColor = color(102);
>   currentColor = baseColor;
>   circleX = width/2+circleSize/2+10;
>   circleY = height/2;
>   rectX = width/2-rectSize-10;
>   rectY = height/2-rectSize/2;
>   ellipseMode(CENTER);
> }
> 
> void draw() {
>   update(mouseX, mouseY);
>   background(currentColor);
>   
>   if (rectOver) {
>     fill(rectHighlight);
>   } else {
>     fill(rectColor);
>   }
>   stroke(255);
>   rect(rectX, rectY, rectSize, rectSize);
>   
>   if (circleOver) {
>     fill(circleHighlight);
>   } else {
>     fill(circleColor);
>   }
>   stroke(0);
>   ellipse(circleX, circleY, circleSize, circleSize);
> }
> 
> void update(int x, int y) {
>   if ( overCircle(circleX, circleY, circleSize) ) {
>     circleOver = true;
>     rectOver = false;
>   } else if ( overRect(rectX, rectY, rectSize, rectSize) ) {
>     rectOver = true;
>     circleOver = false;
>   } else {
>     circleOver = rectOver = false;
>   }
> }
> 
> void mousePressed() {
>   if (circleOver) {
>     currentColor = circleColor;
>   }
>   if (rectOver) {
>     currentColor = rectColor;
>   }
> }
> 
> boolean overRect(int x, int y, int width, int height)  {
>   if (mouseX >= x && mouseX <= x+width && 
>       mouseY >= y && mouseY <= y+height) {
>     return true;
>   } else {
>     return false;
>   }
> }
> 
> boolean overCircle(int x, int y, int diameter) {
>   float disX = x - mouseX;
>   float disY = y - mouseY;
>   if (sqrt(sq(disX) + sq(disY)) < diameter/2 ) {
>     return true;
>   } else {
>     return false;
>   }
> }

Source: https://processing.org/examples/button.html

[Create Button in Python Tk]

> from tkinter import *
> 
> class Window(Frame):
> 
>     def __init__(self, master=None):
>         Frame.__init__(self, master)        
>         self.master = master
> 
>         # widget can take all window
>         self.pack(fill=BOTH, expand=1)
> 
>         # create button, link it to clickExitButton()
>         exitButton = Button(self, text="Exit", command=self.clickExitButton)
> 
>         # place button at (0,0)
>         exitButton.place(x=0, y=0)
> 
>     def clickExitButton(self):
>         exit()
>         
> root = Tk()
> app = Window(root)
> root.wm_title("Tkinter button")
> root.geometry("320x200")
> root.mainloop()

Source: https://pythonbasics.org/tkinter-button/

Pconti31, 2019-04-16:
@rtek1000  Before getting into Python I would like to stress that constructive criticism is always welcome. In my earlier post I was only trying to say is that my goal is to keep things as simple for users as possible without needlessly imposing restrictions on how they work.   That is the lens I use for any changes I make.

Now for Python.   I think it would be fairly easy to supply difference language mappings to the builder.
It's actually designed that way.   I'm just not sure that I could add any value to what is already out there.
Now I haven't done much with python over the years so maybe someone could educate me.
Doing a quick google search shows at least two decent GUI builders for Python.

ActiveState GUI Builder (using grid layout)
[http://spectcl.sourceforge.net/](url)
and the faq and how to get source:
[http://aspn.activestate.com/ASPN/Mail/Message/komodo-announce/3355346](url)

PAGE v3.0 by Stewart Allen (using placer layout)
[http://page.sourceforge.net/](url)

I'm sure there are more that would be revealed with a more detailed search.
Of course, again I don't really know the strengths or weaknesses of what is available.

ImpulseAdventure, 2019-04-23:
@rtek1000 -- thanks for the suggestions regarding alternate targets for the Builder. The way Paul designed it makes it quite flexible in generating different output formats.

I am going to close this issue as the primary topic/concern regarding the ability to change the Builder project folder has been addressed (and included in the upcoming Builder 0.11.0-rc5). If you would like to continue the other discussions, please open a separate issue. thanks!

closed 2019-04-23 by ImpulseAdventure