ImpulseAdventure/GUIslice #134: Generated code requests library Adafruit_ILI9341

rtek1000, 2019-04-11:
Hi,

After generating the code, when attempting to compile, an error occurred, reporting the missing Adafruit_ILI9341 library.

> GUIslice\src\GUIslice_drv_adagfx.cpp:53:34: fatal error: Adafruit_ILI9341.h: No such file or directory

It might be interesting for a person to review each step, so users will not report so many errors.

Thank you.

rtek1000, 2019-04-11:
After downloading the requested library, it presented another error:

> ...GUIslice\src\GUIslice_drv_adagfx.cpp:53:0:
> 
> ...\Documents\Arduino\libraries\Adafruit_ILI9341/Adafruit_ILI9341.h:140:38: error: expected ')' before 'busWidth'
> 
>          Adafruit_ILI9341(tftBusWidth busWidth, int8_t d0, int8_t wr, int8_t dc,

rtek1000, 2019-04-11:
I would like to suggest that the code be generated in a more transparent way so that the user can use his own driver, ILI9341 is just one of the many being found.

At the moment I am with an ILI9327, Open Smart of 3.2 inches (I also have an ILI9341 of 2.4) and the great majority of the codes found in the internet, that are of the generic type can be used easily in different drivers, as long as the main library (Adafruit, mcufriend_kbv etc) works with to similar syntax.

Thank you.

rtek1000, 2019-04-11:
I'm also using uC STM32 because the Arduino (Uno/Mega) is very slow with these displays.

Thank you.

rtek1000, 2019-04-11:
P.S.:

I'm not using stm32 to try to compile the generated code, nor the examples.

I tried to compile with the one board.

Thank you.

ImpulseAdventure, 2019-04-11:
Hi --

Regarding the compilation message: `Adafruit_ILI9341.h: No such file or directory`, this will be shown if your config (in `GUIslice_config.h`) hasn't been updated to [match your particular display driver](https://github.com/ImpulseAdventure/GUIslice/wiki/Configure-GUIslice-for-Arduino). The GUI library needs to ship with a default; the relatively common **Adafruit_ILI9341** was chosen. I understand you are currently testing with an **ILI9327**, which will need a different display driver (I'll come back to this later).

The compilation message: `Adafruit_ILI9341.h:140:38: error: expected ')' before 'busWidth'` suggests to me that your Arduino IDE may be using an old version of Adafruit-GFX. Adafruit recently (1 month ago) made a breaking change to the Adafruit_ILI9341 library that requires Adafruit_GFX version 1.4.0 or later. Please try updating your Adafruit-GFX library (eg. to 1.4.8) and this error should go away.

GUIslice supports a large number of displays (I have purchased approx 40 so far for testing), but I have not yet encountered / added support for an ILI9327.

If you could clarify for me the following, I may be able to add support for your setup:
- **What display shield** are you using? (please provide a link)
  - I am assuming that your display is comparable to this one: [OPEN-SMART 3.2 TFT Shield for UNO/MEGA](https://www.dx.com/p/open-smart-touch-screen-expansion-shield-w-touch-pen-for-arduino-2080038#.XK7U_6R7mUk)
  - I have already added support for the OPEN-SMART 2.2" TFT Shield, but the 3.2" requires a different display driver.
- **What display driver library** are you using? (please provide a link to the Arduino IDE / GitHub repo)
- **What touch driver library** are you using? (please provide a link)
  - I assume from the schematic of the above display shield that you have a 4-wire resistive touch overlay and can probably use the Adafruit_Touchscreen driver.

So for the moment you might want to proceed with your 2.4" ILI9341 on the UNO until I have determined if we can support the particular ILI9327 library you require. If you have a MEGA, that would be a much better option as the UNO is extremely limited in SRAM (meaning we would need to use the low-memory `arduino_min` examples and limit the GUI complexity to fit).

GUIslice also supports the much-more-capable STM32F103, but I haven't added as many display configurations yet (recent focus has been on other targets). The STM config guide/doc still needs revision. I'd be happy to expand the STM32 configs to address the display once we have the UNO/MEGA compilation going for you.

Lastly, I appreciate your comment regarding configuration transparency. If you are using a display that is already supported by mcufriend_kbv, then you should be able to enable an example config like `ard-shld-mcufriend_4wire` and not worry about exactly what display chipset is being used. Ultimately, GUIslice leverages external display and touch libraries, so we still need some configuration to select an appropriate library to call upon.

Thanks again for all the feedback!

rtek1000, 2019-04-11:
Hi,

It's this same kind, I found an image identical to mine, it's attached

The Google drive files on your link seem to be from where I got mine.

The developer of the MCUfriend library helped me set up a TFT, so I managed to make this 3.2 work with STM32.

[image]

rtek1000, 2019-04-11:
I also have, Mega 2560, DUE, Nano, Mini Pro, ESP32, ESP8266, but not UNO (it's expensive here)

rtek1000, 2019-04-11:
Hi,

It's attached (TFT+Touch):

[STM32F103C8T6 (BluePill)+ Open-Smart ILI9327 special 3.2 inch]

Note: original code for Open-Smart ILI9327 special 3.2 inch do not support Arduino DUE, only Mega & Uno.

Thank you.

```
// IMPORTANT: Adafruit_TFTLCD LIBRARY MUST BE SPECIFICALLY
// CONFIGURED FOR EITHER THE TFT SHIELD OR THE BREAKOUT BOARD.
// SEE RELEVANT COMMENTS IN Adafruit_TFTLCD.h FOR SETUP.
//by Open-Smart Team and Catalex Team
//catalex_inc@163.com
//Store:   http://dx.com
//Demo Function: Display graphics, characters (modified)

#include <stdint.h>
#include "TouchScreen_STM.h" // TouchScreen_STM

// These are the pins for the shield!
#if !defined (__STM32F1__)
#define YP A1  // must be an analog pin, use "An" notation!
#define XM A2  // must be an analog pin, use "An" notation!
#define YM 7   // can be a digital pin
#define XP 6   // can be a digital pin
#else
#define YP PB1  // must be an analog pin
#define XM PB0  // must be an analog pin
#define YM PB15   // can be a digital pin
#define XP PB14   // can be a digital pin
#endif

#define MINPRESSURE 10
#define MAXPRESSURE 1000

// For better pressure precision, we need to know the resistance
// between X+ and X- Use any multimeter to read it
// For the one we're using, its 300 ohms across the X plate
TouchScreen ts = TouchScreen(XP, YP, XM, YM, 300);
#define Orientation 1

int x_pos, y_pos, z_pos;

#include <Adafruit_GFX.h>    // Core graphics library
//#include <Adafruit_TFTLCD.h> // Hardware-specific library
#include "MCUFRIEND_kbv.h"
MCUFRIEND_kbv tft;

//***********************************************//
// If you use OPEN-SMART TFT breakout board                 //
// Reconmmend you to add 5V-3.3V level converting circuit.
// Of course you can use OPEN-SMART UNO Black version with 5V/3.3V power switch,
// you just need switch to 3.3V.
// The control pins for the LCD can be assigned to any digital or
// analog pins...but we'll use the analog pins as this allows us to
//----------------------------------------|
// TFT Breakout  -- STM32F103C8 - MEGA shield
// GND                -- GND
// 5V                 -- 5V
// CS                 -- PB3
// RS/CD (XM)         -- PB0
// WR    (YP)         -- PB1
// RD                 -- PB4
// RST                -- RESET
// LED                -- GND
// DB0                -- PB8
// DB1                -- PB9
[... 113 lines cut ...]
  tft.fillRect(mid_x + 40, mid_y + 20, 100, 20, BLACK);
  tft.setCursor(mid_x + 40, mid_y + 20);
  tft.print(p.y);

  tft.setCursor(mid_x + 0, mid_y + 40);
  tft.print("z =");
  tft.fillRect(mid_x + 40, mid_y + 40, 100, 20, BLACK);
  tft.setCursor(mid_x + 40, mid_y + 40);
  tft.print(p.z);

  tft.setCursor(mid_x + 0, mid_y + 60);
  tft.print("X =");
  tft.fillRect(mid_x + 40, mid_y + 60, 100, 20, BLACK);
  tft.setCursor(mid_x + 40, mid_y + 60);
  tft.print(x_pos);

  tft.setCursor(mid_x + 0, mid_y + 80);
  tft.print("Y =");
  tft.fillRect(mid_x + 40, mid_y + 80, 100, 20, BLACK);
  tft.setCursor(mid_x + 40, mid_y + 80);
  tft.print(y_pos);
}
```

[OpenSmart32ILI9327.zip](https://github.com/ImpulseAdventure/GUIslice/files/3067336/OpenSmart32ILI9327.zip)

ImpulseAdventure, 2019-04-12:
Thank you very much for providing the extra details!

On the basis of what you have supplied, I have now created a configuration that might get us closer to your particular setup: uncomment `stm32-adagfx-mcufriend-simple` in GUIslice_config.h.

It would be great if you could try compiling and running the following examples with the above config. Please paste any output that you see reported from the Serial Monitor.
- ex01
- diag_ard_touch_calib

I noticed in your above example that a **TouchScreen_STM** was used instead of Adafruit's original version. I have not yet examined the TouchScreen_STM to see what might have changed in it. I have confirmed that the above new config compiles, but note that I have not yet tested it on an STM32 device, so it is quite possible further edits are required.

thanks!

rtek1000, 2019-04-13:
Thank you! I'll try that.

rtek1000, 2019-04-13:
Hi,

Can not compile:

> In file included from C:\Users\user\Documents\Arduino\libraries\GUIslice\src\GUIslice_drv_adagfx.cpp:126:0:
> 
> C:\Users\user\Documents\Arduino\libraries\Adafruit_TouchScreen/TouchScreen.h:51:12: error: 'RwReg' does not name a type
> 
>    volatile RwReg *xp_port, *yp_port, *xm_port, *ym_port;
> 
>             ^
> 
> C:\Users\user\Documents\Arduino\libraries\Adafruit_TouchScreen/TouchScreen.h:52:3: error: 'RwReg' does not name a type
> 
>    RwReg xp_pin, xm_pin, yp_pin, ym_pin;
> 
>    ^

It looks like this library is not transparent.

Maybe I need to use another library,

'TouchScreen STM.h' instead 'TouchScreen.h'

Thank you.

rtek1000, 2019-04-13:
I modified the GUIslice_drv_adagfx.cpp file: to use 'TouchScreen_STM.h' instead 'TouchScreen.h'

Now ex01 can compile!

But it did not show any serial output.

Only after adding an initial delay of 500 ms, in the first line of the sub setup, there was serial output, maybe this is related to the bootloader I'm using, to upload via USB:

```
void setup()
{
  delay(500); // <------------------
  
  gslc_tsElemRef*  pElemRef = NULL;

  // Initialize debug output
  Serial.begin(9600);
```

(Bootloader: generic_boot20_pc13.bin/generic_boot20_pc13_fastboot.bin from STM32duino-bootloader)

Serial output:

> - Initialized display handler [ADA_MCUFRIEND] OK
> - Initialized touch handler [SIMPLE(Analog)] OK

Thank you.

rtek1000, 2019-04-13:
diag_ard_touch_calib

Serial output:

> - Initialized display handler [ADA_MCUFRIEND] OK
> - Initialized touch handler [SIMPLE(Analog)] OK
> 
> === Touch Calibration ===
> 
> CALIB: Config defaults: XMin=150 XMax=910 YMin=108 YMax=934 RemapYX=0
> - MCUFRIEND ID=0xFFFFD3D3 
> CALIB: Averaging mode: BASIC

Note: in both codes nothing appeared on the display.

Thank you.

rtek1000, 2019-04-13:
diag_ard_touch_calib (adding an initial delay of 500 ms)

Serial output:

> GUIslice version [0.11.2.52]:
> - Initialized display handler [ADA_MCUFRIEND] OK
> - Initialized touch handler [SIMPLE(Analog)] OK
> 
> === Touch Calibration ===
> 
> CALIB: Config defaults: XMin=150 XMax=910 YMin=108 YMax=934 RemapYX=0
> - MCUFRIEND ID=0xFFFFD3D3 
> CALIB: Averaging mode: BASIC

ImpulseAdventure, 2019-04-13:
Good to hear your results!

I noticed a couple things from your update:
- **RwReg**: Your STM32 platform doesn't define a STM32 macro (`#define`) that matches the ones built-in to the Adafruit_TouchScreen library (by default they only detect `ARDUINO_STM32_FEATHER` in TouchScreen.h line 14 of version 1.0.1), whereas I believe your device defines `__STM32F1__`. You might be able to update line 17 of your **Adafruit_TouchScreen/TouchScreen.h** to include your STM32 platform define as follows:
  - FROM:`#if defined(ARDUINO_FEATHER52) || defined(ESP32)`
  - TO: `#if defined(ARDUINO_FEATHER52) || defined(ESP32) || defined(ARDUINO_ARCH_STM32) || defined(__STM32F1__)`
  - I have updated the `stm32-adagfx-mcufriend-simple.h` header to explain this step.
- HOWEVER, as mentioned in my previous update, I think your `TouchScreen_STM.h` is likely to have other fixes that are necessary for your platform, so your plan to use that was good.
- The `MCUFRIEND ID=0xFFFFD3D3` line indicates that `0xD3D3` is returned from your particular display. Looking at the mcufriend sample code you [provided earlier](https://github.com/ImpulseAdventure/GUIslice/issues/134#issuecomment-481994767), we see that David's library overrides the MCUFRIEND ID to be `0x9381` in this case.
  - Therefore, please download the updated `stm32-adagfx-mcufriend-simple`, where you will see the following line has been added:
  - `//#define DRV_DISP_ADAGFX_MCUFRIEND_FORCE  0x9481 // Optionally override the MCUFRIEND initialization ID`
  - Uncomment this line to force the MCUFRIEND initialization to match your particular display.

To summarize:
- Download the updated `/configs/stm32-adagfx-mcufriend-simple.h`
- Uncomment the MCUFRIEND_FORCE in line 80 of Section 1, which should set the ID to `0x9481`
- Continue to use `TouchScreen_STM.h`

Let me know if that helps!

rtek1000, 2019-04-13:
After some adaptations, this presented a more serious error:

> cc1plus.exe: out of memory allocating 65536 bytes
> 
> exit status 1
> Error compiling for board Generic STM32F103C series.

I had to modify your code to operate with the local files ("lib.h") instead of global (<lib.h>).
(I'm not sure I could do it this way)

GUIslice_drv_adagfx.cpp:
```
  #elif defined(DRV_DISP_ADAGFX_MCUFRIEND)
    #include "MCUFRIEND_kbv.h"
```

diag_ard_touch_calib.ino:
```
// ARDUINO NOTES:
// - GUIslice_config must be edited to match the pinout connections
//   between the Arduino CPU and the display controller (eg. ADAGFX_PIN_*).
//

// Import optional libraries
//#if defined(DRV_DISP_ADAGFX_MCUFRIEND)
  // Import MCUFRIEND to support ID reporting
  #include "MCUFRIEND_kbv.h"
  extern MCUFRIEND_kbv m_disp;
  int16_t m_nMcuFriendId = -1;
//#endif

#include "GUIslice.h"
#include "GUIslice_drv.h"
```

Thank you.

rtek1000, 2019-04-13:
Sorry!

I did not see the previous post!!!

rtek1000, 2019-04-13:
Ok, done (Uncomment this line to force the MCUFRIEND)

New error:

> diag_ard_touch_calib:29:4: error: #error "Calibration testing requires a touch driver but DRV_TOUCH_NONE was set"
> 
>    #error "Calibration testing requires a touch driver but DRV_TOUCH_NONE was set"
> 
>     ^
> 
> diag_ard_touch_calib:46:4: error: #error "Calibration only supported for resistive touch displays"
> 
>    #error "Calibration only supported for resistive touch displays"
> 
>     ^
> 
> exit status 1
> #error "Calibration testing requires a touch driver but DRV_TOUCH_NONE was set"

rtek1000, 2019-04-13:
Sorry, but would not it be safer to create a "tree" to avoid errors in the main code?

ImpulseAdventure, 2019-04-13:
Hi --

From the errors that you just posted, it looks like the `/src/GUIslice_config.h` file (read by Arduino IDE when including `GUIslice.h`) is the default unedited version. The default `GUIslice_config.h` looks to see if any of the example configs have been uncommented, and if not, it loads GUIslice_config_ard.h which defines `DRV_TOUCH_NONE` and `ILI9341`. You will want to make sure that your GUIslice_config.h has uncommented the example config include line `stm32-adagfx-mcufriend-simple.h`.

It should be cautioned that the **STM32F103** tends to be tight on program memory after importing MCUFRIEND_kbv, Adafruit_TouchScreen, Adafruit-GFX, SPI and the GUI. The ATmega2560 and ESP32 generally have a lot more flexibility. I believe I have seen reports of MCUFRIEND_kbv working with the ESP32, but I'm not certain if that also applies to his OPEN-SMART TFT config.

As for your suggestion of the *create a "tree" to avoid errors in the main code*, can you clarify what you are recommending?

thanks!

rtek1000, 2019-04-13:
> As for your suggestion of the create a "tree" to avoid errors in the main code, can you clarify what you are recommending?

I have seen some repositories with this "tree", sorry but I do not have very solid knowledge about this, I think it would be safer to talk to another developer.

> https://github.com/jeelabs/esp-link
> https://github.com/jeelabs/esp-link/tree/master/html
> https://github.com/jeelabs/esp-link/tree/master/serial
> https://github.com/jeelabs/esp-link/tree/master/esp-link

I found this link, please if you can, check if it is related:
[The Three Trees]
https://www.youtube.com/watch?v=1GFIZFXGMXY

I found this link, but I think it's outdated (2016):
https://code.tutsplus.com/courses/git-basics-branches

rtek1000, 2019-04-13:
Sorry, but I need to review the IDE installation, something went wrong.

[image]

ImpulseAdventure, 2019-04-13:
Thanks... I see that you were recommending that we utilize git feature branches.

Whenever I make changes to the GUIslice core library, I almost always create a feature branch for that very reason (if you look at the pull request history alone, you'll see nearly 50 of them, plus an equal number of local branches that were never pushed to the remote repository).

For this particular feature request, I did not create a feature branch because we weren't touching any of the main library code. The primary changes were localized to the new config file I created for you (`stm32-adagfx-mcufriend-simple.h`), so the expectation was that it should not impact any other users. I don't think any errors would have been injected into the main code.

I agree that in general, any library changes should be in a branch rather than the master.

thanks for digging up the references

ImpulseAdventure, 2019-04-13:
Hi @rtek1000 — as you appear to be encountering other issues specific to your system, I think it may be more efficient to followup initially via email (guislice@gmail.com), where I would be happy to assist you with your IDE issue. I can then post a summary here later with any changes that are likely to be relevant to other users. Thx

rtek1000, 2019-04-13:
Hi,

After restarting Windows 10 the Blink sketch was recompiled without errors to the UNO and STM32 boards.

I was able to make the sketch compile without the local files, I no longer need to use "lib.h", only <lib.h>.
(I needed to replace only the file mcufriend_shield.h)

The next step is to reinstall GUIslice, and reconfigure the files, as nothing is still appearing on the display.

Thank you.

ImpulseAdventure, 2019-04-13:
Good to hear your Arduino environment is working again!

Thank you for attaching your edited mcufriend_shield and the OPEN-SMART sketch. I would like to see your other locally-edited mcufriend library files for completeness and had a number of clarifying questions that I think may be better suited to resolve over email. If you don't mind, please send a quick message to guislice@gmail.com and I'd be happy to get this going for you. Thx.

ImpulseAdventure, 2019-04-23:
Hi @rtek1000  — if you are still interested in getting your setup operational, my recommendation would be to start with your ATmega2560 or ESP32 to ensure that the library is configured properly for the Open-Smart display (since much more testing has been done with them). Once we have confirmed your display works properly with the library, we can then adapt any remaining adjustment for your STM32 if desired. Thanks

ImpulseAdventure, 2019-04-27:
For continued debug of your setup, please feel free to update this issue and I can reopen it. Thx

closed 2019-04-27 by ImpulseAdventure

rtek1000, 2019-04-28:
Hi, thanks, I was having some hardware issues. my computer was badly aligned on the heat sink, so the processor was operating very slowly. Sometimes he would hang up without warning. I am currently resolving other pending services. As soon as possible I want to re-test this GUI better.

rtek1000, 2019-04-28:
Hi,

I would like to know if you know the Nextion platform (https://nextion.itead.cc/editor_guide/).

I believe that if it is possible to define open source hardware, it can be a great alternative to this Nextion, which sells the most than the other compatible ones.

They use the STM32 chip to control the display.

Maybe in the STM32duino forum you can find people interested in helping to define hardware. What do you think of this idea?

rtek1000, 2019-04-28:
The idea would be hardware of a controller that can be configured for the TFT and touchscreen available in the local market of each user. The SSD1963 chip already exists as an example. Because STM32 was already combined with the Arduino IDE, it would certainly be more simpler to use the same STM32 to control the display without the need for an SSD1963 (or ILI9341 etc), I believe a display library would be designed to enable STM32 as a controller, and the user could add other necessary functions.

Can I post the idea there in the STM32duino forum? Or do you prefer to do it on your own?