ImpulseAdventure/GUIslice #135: Wrong display width/height with ST7735

amyspark, 2019-04-21:


ImpulseAdventure, 2019-04-21:
Hi — thanks for the report. I will be able to test more thoroughly tomorrow evening but in the interim I have a couple suggestions:

- 1) What size & resolution display are you using? For example, if you have the 1.44” 128x128 display, could you try initializing DRV_DISP_ADAGFX_ST7735_INIT to INITR_144GREENTAB?

- 2) Can you confirm that the Adafruit_ST7735 example graphicstest runs properly? If so, what changes were required to have it run?

Thanks!

ImpulseAdventure, 2019-04-21:
From your update I see that you have a 1.8” display: I understand from the Adafruit example that `INITR_BLACKTAB` might be a more appropriate initialization mode to use. I would be interested to know if that improves the resolution settings.

amyspark, 2019-04-21:
Hey @ImpulseAdventure,

1: it's a 1.8'', 128x160 SPI TFT display, green tab.
2: Yes, it runs properly, with both `INITR_GREENTAB` and `INITR_BLACKTAB`. Even the text is wrapped properly. (I only changed `TFT_RST` to 0 and `TFT_DC` to 9 so as to match your library.)

[image]

3: Same as above, the Adafruit GFX `graphicstest` runs perfectly.
4: All of the examples run successfully, but it seems that the library doesn't realise it's a small screen. See this picture (this is with `GSLC_ROTATE` set to 0):

[image]

ImpulseAdventure, 2019-04-21:
Thank you for the details & photos! It looks like GUIslice is working properly, though that example requires a larger display.

In general, most examples are typically designed to place GUI elements on the common 320x240 display size and thus were not intended to “auto-size” to smaller resolutions.

That said, in light of your posting, I think I should create an example that targets smaller display resolutions (eg. 128x128), suitable for a ST7735.

ImpulseAdventure, 2019-04-22:
Hi @amyspark -- I have now added example `ex40_ard_ctrls_small` to the repository. This example was designed to fit on small displays such as those supported by **ST7735** (eg. 128x128). Hopefully this should enable you to confirm the library operation.

amyspark, 2019-04-22:
I'll test later with a bigger Arduino (the new example exhausts my Uno's memory). Thanks for the example @ImpulseAdventure !

closed 2019-04-22 by amyspark

ImpulseAdventure, 2019-04-23:
Thanks for the followup. Extensive debug messaging is enabled by default in GUIslice. If you are running on a low-memory device like the UNO, you can disable error messaging by setting `#define DEBUG_ERR 0` in Section 5 of the example config. Using a device with more memory would naturally offer greater flexibility.

yahay05, 2024-06-13:
Hi [brightproject](https://github.com/brightproject).
I also had the same issue and solved it by changing  ST7735_TFTWIDTH_128  and ST7735_TFTHEIGHT_160 variables in **Adafruit_ST77xx.h** library file.
The default values are 128 and 160. I increased each according to my screen. (131,161).
/
#define ST7735_TFTHEIGHT_160 161