ImpulseAdventure/GUIslice #136: Teensy + Audio Adaptor

pdupuy, 2019-04-22:
With this hardware:
- Teensy 3.2
- Audio Adaptor
- TJCTM24024-SPI (2.4" TFT) 

And these libraries:
- Adafruit_ILI9341 (1.3.6)
- Adafruit_GFX (1.4.10)
- XPT2046_Touchscreen (5ccb951)
- GUIslice (4b1b5a7)

All "arduino" examples work in Arduino 1.8.7, with these exceptions:
- ex03_ard_btn_img (haven't tested required SD card)
- ex06_ard_callback (haven't tested required SD card)
- ex16_ard_touch_hnd (untested)
- diag_ard_touch_detect ("only intended to run in DRV_TOUCH_ADA_SIMPLE mode")
- ex19_ard_smooth ("This example only works on ESP8266 / ESP32")

When the following changes are made:

**GUIslice_config.h: 84** 
` #include "../configs/ard-adagfx-ili9341-xpt2046.h"`
 

**GUIslice_config.h: 123**
Got `#error "Unknown device platform"` so I just commented out the entire Default section for now:
```
// Import user configuration depending on device type
// #if defined(__linux__)
[...]
// #endif
```

**ard-adagfx-ili9341-xpt2046.h: 81**
Using pins in ["Teensy 3.x Audio Board" Connections column](https://www.pjrc.com/store/display_ili9341_touch.html):

```
  // For shields, the following pinouts are typically hardcoded
  #define ADAGFX_PIN_CS       21    // Display chip select
  #define ADAGFX_PIN_DC       20     // Display SPI data/command
  #define ADAGFX_PIN_RST      255     // Display Reset

  // Display interface type
  #define ADAGFX_SPI_HW       1     // Display uses SPI interface: 1=hardware 0=software
```

**ard-adagfx-ili9341-xpt2046.h: 120** 

```
  // Touch bus & pinout
  #define XPT2046_CS     8
```

**GUIslice_drv_adagfx.cpp: 314**
Must use these SPI.set methods as described in [Usage with Audio Board Connections](https://www.pjrc.com/store/display_ili9341_touch.html):

```
    #if defined(DRV_DISP_ADAGFX_ILI9341) || defined(DRV_DISP_ADAGFX_ILI9341_STM)
      SPI.setMOSI(7);
      SPI.setSCK(14);
      m_disp.begin();
```

**ard-adagfx-ili9341-xpt2046.h: 174** 
Enabled all optional features:

```
  #define GSLC_FEATURE_COMPOUND       1   // Compound elements (eg. XSelNum)
  #define GSLC_FEATURE_XGAUGE_RADIAL  1   // XGauge control with radial support
  #define GSLC_FEATURE_XGAUGE_RAMP    1   // XGauge control with ramp support
  #define GSLC_FEATURE_XTEXTBOX_EMBED 1   // XTextbox control with embedded color
  #define GSLC_FEATURE_INPUT          1   // Keyboard / GPIO input control
```

**ard-adagfx-ili9341-xpt2046.h: 130**   
Set values from calibration utility:

```
  // Calibration settings from diag_ard_touch_calib:
  // DRV_TOUCH_XPT2046:
  #define ADATOUCH_X_MIN    230
  #define ADATOUCH_X_MAX    3819
  #define ADATOUCH_Y_MIN    183
  #define ADATOUCH_Y_MAX    3665
  // Certain touch controllers may swap X & Y coords
  #define ADATOUCH_REMAP_YX 0
```

ImpulseAdventure, 2019-04-23:
Excellent... thank you @pdupuy for the initial work on adapting GUIslice for the Teensy!

I will integrate native support for the [Teensy 3](https://www.pjrc.com/teensy/) into GUIslice, and in addition to **Adafruit_ILI9341**, I intend to support the inclusion of a Teensy-optimized display driver (eg. **ILI9341_T3**).

ImpulseAdventure, 2019-04-27:
**Teensy 3.2** now confirmed operational using the above TFT with both the **Adafruit_ILI9341** as well as a patched **ILI9341_t3** library. Support has been added for both the default Teensy breakout connections in addition to the alternate breakout pinout to support the Teensy audio adapter. Touch support is provided by the XPT2046 driver. GUIslice was enhanced to support multiple font modes to enable both Adafruit-GFX and Teensy fonts.

The above enhancements are present in the [WIP136-Teensy](https://github.com/ImpulseAdventure/GUIslice/tree/WIP136-Teensy) branch.

The remaining work is to confirm which ILI9341_t3 fork to use as the basis for GUIslice support and complete testing for the ILI9341_t3 fonts. In order to support the text justification in the GUI, some enhancements (primarily text width/height APIs) need to be rolled into the t3 library.

My intention is to submit a PR for these enhancements into the [PaulStoffregen/ILI9341_t3](https://github.com/PaulStoffregen/ILI9341_t3) library, but if they are not integrated, then we may use our own fork.

pdupuy, 2019-04-28:
Working with the same hardware as above and these libraries (in PlatformIO):
- ILI9341_t3 (57d0141)
- XPT2046_Touchscreen (5ccb951)
- WIP136-Teensy (e82d629)

In PlatformIO, with these changes, it works for me:

**GUIslice_config.h: 89**
Commented #include lines for the new ILI9341_t3 configs are missing; added:
`#include "../configs/teensy-adagfx-ili9341_t3-xpt2046-audio.h"`

**teensy-adagfx-ili9341_t3-xpt2046-audio.h: 187**
Enabled SD as I need to use the Audio Adaptor SD card:
`#define GSLC_SD_EN    1`

**main.cpp:**
To use the ILI9341_t3 fonts:
```
#include <font_Arial.h>
[...]
// Load Fonts
// ------------------------------------------------
// //<Load_Fonts !Start!>
if (!gslc_FontAdd(&m_gui,E_FONT_ARIAL14,GSLC_FONTREF_PTR,&Arial_14,1)) { return false; }  
gslc_FontSetMode(&m_gui,E_FONT_ARIAL14,GSLC_FONTREF_MODE_1);
```
Amazingly easy -- thanks Cal!

pdupuy, 2019-04-29:
I'm seeing unused variable warnings and am patching with `__attribute__((unused))` to eliminate the compile noise:
```
lib\GUIslice\src\GUIslice_drv_adagfx.cpp:320:8: warning: unused variable 'bInitOk' [-Wunused-variable]
lib\GUIslice\src\GUIslice_drv_adagfx.cpp:566:13: warning: unused variable 'nTxtLen'
lib\GUIslice\src\GUIslice_drv_adagfx.cpp:658:13: warning: unused variable 'nColBgRaw' 
lib\GUIslice\src\GUIslice_drv_adagfx.cpp:1079:14: warning: unused variable 'nFileSize'
lib\GUIslice\src\GUIslice_drv_adagfx.cpp:1085:14: warning: unused variable 'nHdrSize'
lib\GUIslice\src\GUIslice_drv_adagfx.cpp:1063:21: warning: unused variable 'startTime'
lib\GUIslice\src\elem\XTextbox.c:357:21: warning: variable 'pBox' set but not used
lib\GUIslice\src\elem\XTextbox.c:375:21: warning: unused variable 'pBox'
```

There is also this warning:
```
lib\GUIslice\src\GUIslice_drv_adagfx.cpp: In function 'bool gslc_DrvDrawImage(gslc_tsGui*, int16_t, int16_t, gslc_tsImgRef)':
lib\GUIslice\src\GUIslice_drv_adagfx.cpp:1181:38: warning: format '%X' expects argument of type 'unsigned int', but argument 3 has type 'const unsigned char*' [-Wformat=]
sprintf(addr,"%04X",sImgRef.pImgBuf);
```

ImpulseAdventure, 2019-04-30:
Great... thank you very much for testing out the Teensy updates and providing the thorough details!

I have now merged in the changes from the WIP136-Teensy branch into master. I also integrated a few of the loose ends that you noted above, but feel free to let me know if you happen to catch anything else.

### ILI9341_t3 usage
Note that the use of the ILI9341_t3 optimized driver has a few differences from other Adafruit-GFX drivers, so some minor updates are required when t3 fonts are in use (most of which @pdupuy noted above). For example, GUIslice Builder generated code and certain examples may require some small adjustments. For details, please refer to: [Using ILI9341_t3 on Teensy](https://github.com/ImpulseAdventure/GUIslice/wiki/Install-ILI9341_t3-for-Teensy)

closed 2019-05-01 by ImpulseAdventure