ImpulseAdventure/GUIslice #137: Integrated display & controller

ImpulseAdventure, 2019-04-29:
@rtek1000 — I moved your comments to a new issue:
> The idea would be hardware of a controller that can be configured for the TFT and touchscreen available in the local market of each user. The SSD1963 chip already exists as an example. Because STM32 was already combined with the Arduino IDE, it would certainly be more simpler to use the same STM32 to control the display without the need for an SSD1963 (or ILI9341 etc), I believe a display library would be designed to enable STM32 as a controller, and the user could add other necessary functions.

Interesting idea... I haven’t looked at the particular hardware you referenced earlier, but from what I understand of the partitioning you indicated, one would specify an integrated GUI device (TFT with tightly-coupled MCU for display offload), driven by a host controller via some form of serial interface. Is that correct?

### Objective
What do you think might be the primary goals/advantages one would be looking for in using an integrated display with remote serial UART operation? Having the display driver offload would certainly free up cycles in the host / application processor, enabling resource-constrained host MCUs to be used. It would also give the user full flexibility in selecting whatever host MCU they wish, rather than being tied to the particular platforms that the given display drivers support. Would these be the main benefits, or are there other aspects worth considering?

Clearly there would be some tradeoff in performance and flexibility; the question is whether the above would add enough value to be useful for certain users?

Feel free to post your thoughts/questions to get input from elsewhere if you think there might be interest.

The following are just a few quick thoughts that come to mind...

### Display-side control
I believe that it could be possible to create a thin driver adaptation layer (“display serial driver”) that would execute on the TFT MCU, listen to serial commands, call GUIslice APIs and feed back responses to the serial port. Any user device with a serial port (eg. baseline Arduino, Raspberry Pi, etc.) could drive the special commands over the serial interface and manage the application code, respond to user input, request page changes, etc.

### Feature impact
To consider a feature like this, my startpoint would be to ensure it had little impact to the GUIslice core code. My impression is that such a serial driver could indeed be created independently without much impact on the existing code. A single file include for the display serial driver (on the TFT MCU) could provide much of the `setup()` and `loop()` function content needed to bridge between the serial interface and the existing GUIslice APIs. To keep things simple I would probably assume that GUI elements would be created ahead of time on the display MCU, not created dynamically (at run-time) via the serial interface, though immediate graphics primitive rendering could be done dynamically. Would that be too restrictive?

### Host-side control
Optionally, one could also create another adaptation layer (“host serial driver”) that runs on the host to convert between simplified GUIslice-like API function calls and the serial commands. These APIs would probably include a user-friendly parameter (perhaps string-based?) reference to the existing GUI elements, and be used to transfer touch events, element value updates, page transitions, etc.

With such a framework, the user could create a GUI using the GUIslice Builder, compile the TFT MCU code (perhaps the Builder would include a flag for a “serial driver” code generation mode), leaving the user to be free to develop the application code on whatever host processor they selected.

### Advanced Usage / Future Proofing
One possibility with this approach is that advanced users could add their own special commands to the display serial driver, and leverage any custom code they want to place on the TFT MCU (eg. drive other GPIOs, I2C, SPI devices, etc.) Basic users could simply stick to the built-in GUI operations.

### Hardware selection
From a display module perspective, one would have flexibility in selecting an off-the-shelf MCU (such as ESP32 or STM32) and a cheap touchscreen, as GUIslice offers quite a bit of flexibility on the hardware side, though I’d imagine that one would want to propose a few tested combinations that are likely to deliver a suitable performance/price tradeoff.

### Serial port options
Given the creation of the such a framework, it seems possible that one could use alternate interfaces to the typical wired serial UART (eg. WiFi, XBee, Bluetooth) assuming the MCU offered the option.

Thoughts?

rtek1000, 2019-04-29:
Hi,

All of these issues raised are relevant, but I think at the first moment, a single hardware must be defined to prevent the user from having many options and eventually overloading software support with various hardware models.

I think it would be a good idea to set up a low cost hardware, (among market options, not specifically using any generic hardware to emulate what a dedicated hardware does) but that has the ability to be used directly, not specifically being controlled by another external device such as Arduino.

Note that the most interesting are large displays, such as those of 7 inches, or perhaps 4.3 inches, since the smaller ones usually already have controllers like the ILI9341. I think it's not worth emulating an ILI9341 for a 2.4-inch display, for example.

Some chips STM32 have an internal hardware block called LTDC, which is dedicated to interfacing with the TFT display.

There is a "discovery" card that already has a display:
[image]

Source:
https://www.st.com/en/evaluation-tools/32f429idiscovery.html

This document describes this integration of the STM32 chip with the TFT display:

[image]

Source:
https://www.st.com/content/ccc/resource/technical/document/application_note/group0/25/ca/f9/b4/ae/fc/4e/1e/DM00287603/files/DM00287603.pdf/jcr:content/translations/en.DM00287603.pdf

I have found a board that can be a starting point:

[image]

Source:
https://www.ebay.com/itm/STM32F429IGT6-Development-Board-Cortex-M4-32-bit-STM32F429-Core-BoardAU/323226501354?hash=item4b41cd00ea:g:KZUAAOSwWMFa4XD9

The most interesting display I found was a 7-inch 800x480, is similar to those found on most low-cost Android tablets.

[image]

https://www.ebay.com/itm/7-inch-LCD-Display-Screen-Panel-AT070TN90-AT070TN92-800-480-50-pins-TFT/173873482805?hash=item287baacc35:g:Pj0AAOSw7aNcsEA3

rtek1000, 2019-04-29:
I found a board that already has the connector for the display, but it is for 40-pin display. I asked the seller which display she supports,

[image]

https://www.ebay.com/itm/STM32-Development-Board-Cortex-M4-Small-System-Board-STM32F429IGT6-Core-Board-Z/254165281946?hash=item3b2d6e6c9a:g:nZ0AAOSwQKdcjFjo

Some 40-pin displays also have the built-in touchscreen:

[image]

https://www.ebay.com/itm/40-Pin-7-Inch-TFT-LCD-Module-Touch-Screen-Panel-800-480-High-Resolution/180845680581?hash=item2a1b3e17c5:g:QVoAAOSwGUBZ5vmW

ImpulseAdventure, 2019-04-29:
Thank you @rtek1000 for sharing the thoughts behind your project idea, and clarifying that the scope includes "setting up a low cost hardware". 

As the scope is more expansive and initial discussions would likely focus on the hardware (component selection / feature set / board development), I recommend that you create a new repository & wiki for your overall project. That way all of the relevant details & discussions can be maintained with the right audience and gain the best visibility as this repo is primarily focused on the GUI library side.

Naturally, it would be great to see a new open source hardware platform, so if GUIslice looks like it could be a good fit for the GUI driver, please feel free to continue the API-oriented discussions here.

Thanks again for the ideas!

rtek1000, 2019-04-29:
Hello,

Yes, I understand that hardware is not the object of this repositorium, that's why I commented of posting on the stm32duino site to see if anyone has the time available to move this idea. Although the cost of the hardware does appear a little high at first, we should note that the performance compared to an Arduino Due + an SSD1963 is much higher. For the Arduino Due to load an image of 800x480 in the display with the SSD1963, it takes about 27 seconds.

I'll post to the forum and see what the users say.

Thank you.

Thank you.

rtek1000, 2019-04-29:
I found a post a bit old already, but I think it can be a good starting point:

https://www.stm32duino.com/viewtopic.php?f=3&t=1445&p=54260#p54259

Thank you.

ImpulseAdventure, 2019-05-27:
Thanks again @rtek1000 for sharing your ideas on the project. I will mark this discussion as "Open for Future Enhancement". If you have continued thoughts later on how GUIslice could assist with your project, please feel free to post the ideas and I'd be happy to adjust GUIslice if needed. Thanks!

closed 2019-05-27 by ImpulseAdventure