ImpulseAdventure/GUIslice #140: whitescreen elegoo

KolyaKollega, 2019-05-14:
Ola, i would love your software if i can get it to work :(
i'm use the elegoo 2.8 touch tft with an ardu mega.
if i create a gui, export the code and upload it, everything is fine, no error while compiling. but the screen stays white -__-
i changed the config to ili9341-simple.h and the serial monitor also says everything is ok...

ImpulseAdventure, 2019-05-15:
Hi KolyaKollega —

Before using the Builder to create a GUI, it’s important that we make sure the graphics driver is correctly matched to your display. The white screen usually indicates that the graphics driver is not set correctly.

For GUIslice, I think we could probably start with the following config: “ard-shld-mcufriend” (Note that this will not enable touch, as I’d like to ensure the display is operational first).

Can you confirm that the mcufriend_kbv library examples works on your display? (Ie. Before integrating GUIslice)

### DETAILS:
I assume you have an Elegoo 2.8” TFT shield with an 8-bit interface and a resistive 4-wire touchscreen.

Elegoo provides their own drivers, but let’s try using David Prentice’s mcufriend_kbv for your display instead.

Please do the following:
- Uninstall any hacked mcufriend_kbv library that Elegoo may have provided (move/delete the directory from the Arduino Libraries folder)
- From the Arduino IDE Library Manager, install “mcufriend_kbv”

Now, try running the mcufriend_kbv examples:
- diagnose_TFT_support
- graphicstest_kbv

Do these work on your display? If you can paste in the output from the diagnose_TFT_support to this issue, that would be helpful.

Thanks!

KolyaKollega, 2019-05-15:
using the ard-shld-mcufriend config solve the white screen problem, big thanks, now i just need to find out which touch-handler to use.
gib thanks mate!

ImpulseAdventure, 2019-05-15:
Great... glad to hear the display driver is working well!

I have now created a new config that should work with the **Elegoo 2.8" TFT** including touch support, based on the shield pinout I have seen described elsewhere: `/configs/ard-shld-elegoo28_res`

Please update your copy of GUIslice with the latest versions of these files from the repository:
- `/configs/ard-shld-elegoo28_res.h`
- `/src/GUIslice_config.h`

As your touchscreen requires calibration, you should run the following sketch to adjust the parameters for your display:
- `/examples/arduino/diag_ard_touch_calib.ino`

Hopefully the touch works correctly for you with the new config.
If possible, please post the results from your calibration sketch so that I can update the config in the repo.

Thanks!

KolyaKollega, 2019-05-17:
jup everythings works fine now, thx mate!

````GUIslice version [0.11.3.6]:
- Initialized display handler [ADA_MCUFRIEND] OK
- Initialized touch handler [SIMPLE(Analog)] OK

=== Touch Calibration ===

CALIB: Config defaults: XMin=150 XMax=920 YMin=120 YMax=940 RemapYX=0
- MCUFRIEND ID=0x9341 
CALIB: Averaging mode: BASIC

Capture TL: X=859 Y=133 Cnt=23, Z=(265..339)
Capture TR: X=192 Y=134 Cnt=25, Z=(283..313)
Capture BL: X=840 Y=891 Cnt=30, Z=(281..345)
Capture BR: X=177 Y=889 Cnt=33, Z=(317..392)
SegDetect: Normal display (FlipX)
CALIB: Result(adj) XMin=879 XMax=153 YMin=107 YMax=915

CALIB: Rotate
Capture TL (Rotated): X=168 Y=126 Cnt=24, Z=(321..435)
CALIB: Detected normal rotation

Recommended calibration settings are shown below.
- Copy and paste these lines into your GUIslice config file
  over top of the existing ADATOUCH_X/Y_MIN/MAX settings.

---
  // DRV_TOUCH_ADA_SIMPLE [240x320]: (MCUFRIEND ID=0x9341) (XP=8,XM=56,YP=57,YM=9) 
  #define ADATOUCH_X_MIN    879
  #define ADATOUCH_X_MAX    153
  #define ADATOUCH_Y_MIN    107
  #define ADATOUCH_Y_MAX    915
  #define ADATOUCH_REMAP_YX 0
---
````

ImpulseAdventure, 2019-05-17:
Thank you @KolyaKollega for the confirmation. I have also updated the default calibration in `ard-shdl-elegoo_28_res` according to the values you observed in your display.

closed 2019-05-17 by ImpulseAdventure

mvan127, 2022-04-07:
@KolyaKollega @ImpulseAdventure Hi, I'm newer to coding for arduino and this conversation was a bit tricky for me to follow. Was there code released for a 2.8 inch Elegoo screen with an arduino mega? I have the same set up and am getting a white screen after the sketch is successfully compiled and uploaded.