ImpulseAdventure/GUIslice #141: Errors in builder

jmcastillejo, 2019-05-19:
First of all I want to congratulate you for your project, a long time ago I was looking for something like that to develop designs on small screens.

I have been testing the library and I have several problems, the first is about the bulder, does not perform well the import code of the examples, and therefore does not generate the code well

On the other hand, I can not get the touch panel to work, it's an XPT2046, with the library https://github.com/PaulStoffregen/XPT2046_Touchscreen, and the internal one that uses TFT_eSPI can not exchange the XY axes

I will be grateful for the comments

Regards

ImpulseAdventure, 2019-05-19:
Hi Jose — thanks!

Before using the Builder, we should make sure that the basic library examples are working properly on your hardware.

A couple questions can help us get it going for you:
1) Can you send a web link to your display that you are using?
2) What MCU do you have? (eg. ESP32, etc.)
3) When you run example ex02 (using PaulStoffregen/XPT2046) is the display working but not the touch?
4) When you open the Serial Monitor and run example ex02, please copy the output initialization messages to this issue.

Thanks!

jmcastillejo, 2019-05-19:
1.- The display is a SPI ILI9488 with XPT2046 touch controller: https://www.aliexpress.com/item/3-5-inch-TFT-LCD-Module-with-Touch-Panel-ILI9488-Driver-320x480-SPI-port-serial-interface/32985467436.html?spm=a219c.10010108.1000001.12.4a48798274WU5t&isOrigTitle=true

2.- The MCU is a ESP8266F 4Mb (Nodemcu)

3.- using PaulStoffregen/XPT2046 the MCU crashed reset loop

The display and touch works perfectly with TFT_eSPI library and examples

ImpulseAdventure, 2019-05-20:
Thanks for the extra details.

Just to confirm:
- When you ran the PaulStoffregen/XPT2046 **TouchTest** example (**without GUIslice**), did it run correctly?
- When you run GUIslice with config `esp-tftespi-default-notouch` on example **ex02**, does it display correctly?
- When you run GUIslice with config `esp-tftespi-default-xpt2046_int` on example **diag_ard_touch_test**, does it respond to touch presses? If so, do they appear to be in the wrong locations?

thanks

jmcastillejo, 2019-05-20:
Yes, the PaulStoffregen/XPT2046 TouchTest example run correctly.

Yes, the esp-tftespi-default-notouch on example ex02 run correctly.

The esp-tftespi-default-xpt2046_int on example diag_ard_touch_test respond to touch presses, but appear to be in the wrong locations, the x and y axes seem to be changed

ImpulseAdventure, 2019-05-20:
Ok good. In that case it just looks like the touch calibration hasn’t been set yet.
Assuming you have run TFT_eSPI’s calibration utility (Touch_calibrate), we need to copy the output “calibration code” to TFT_ESPI_TOUCH_CALIB in section 4B of your config.

Please refer to [Configuring XPT2046-TFT_eSPI](https://github.com/ImpulseAdventure/GUIslice/wiki/Configuring-Touch-for-XPT2046-(TFT_eSPI)). Thanks.

jmcastillejo, 2019-05-20:
I already did that before starting the thread, but there is no change when modifying the calibration values, as well as the #define ADATOUCH_REMAP_YX

jmcastillejo, 2019-05-22:
Hi Calvin,

I solved the problem by forcing the rotation variable in the header file
where TFT_eSPI manages the graphic rotations, specifically for my screen is
the ILI9488_Rotation.h

Enclosed code in case it is of your help:

  // This is the command sequence that rotates the ILI9488 driver
coordinate frame

  writecommand(TFT_MADCTL);
  // rotation = m % 4;
  rotation = 3; // fix bug GSLC_ROTATE
  switch (rotation) {
   case 0: // Portrait
     writedata(TFT_MAD_MX | TFT_MAD_BGR);
      _width  = TFT_WIDTH;
      _height = TFT_HEIGHT;
     break;
   case 1: // Landscape (Portrait + 90)
     writedata(TFT_MAD_MV | TFT_MAD_BGR);
      _width  = TFT_HEIGHT;
      _height = TFT_WIDTH;
     break;
   case 2: // Inverter portrait
     writedata(TFT_MAD_MY | TFT_MAD_BGR);
      _width  = TFT_WIDTH;
      _height = TFT_HEIGHT;
     break;
   case 3: // Inverted landscape
     writedata(TFT_MAD_MX | TFT_MAD_MY | TFT_MAD_MV | TFT_MAD_BGR);
      _width  = TFT_HEIGHT;
      _height = TFT_WIDTH;
     break;
  }

ImpulseAdventure, 2019-05-23:
Thank you very much for identifying a workaround!

After examining it in more detail, I have determined that the primary issue appears to be that TFT_eSPI’s integrated XPT2046 touch driver does not currently support dynamic rotation. The touch rotation correction is done once during calibration (`calibrateTouch()` / `setTouch()`), but not again after `setRotation()` is called. I have filed an issue in the TFT_eSPI repo to see if this functionality might be considered. Compounding this is the default use of calibration `rotation=1` instead of the native portrait orientation (`rotation=0`).

In light of the existing TFT_eSPI capabilities, I think the most robust solution may be to bypass TFT_eSPI’s touch calibration / rotation logic and use its raw uncorrected touch results instead. GUIslice can then manage the calibration and dynamic rotation like it does for other resistive touch drivers (ie. from `diag_ard_touch_calib`).

I will start evaluating the changes necessary to address this alternate approach, but in the meantime you should be able to use your workaround (as it effectively disables dynamic rotation). Thanks!

ImpulseAdventure, 2019-05-24:
I have now integrated changes to the TFT_eSPI internal touch driver mode (config: `esp-tftespi-default-xpt2046_int`). The revised implementation uses the raw touch coordinates from TFT_eSPI's XPT2046 driver and leaves the calibration and dynamic rotation to GUIslice.

With this update, you should be able to run the `diag_ard_touch_calib` utility in the TFT_eSPI integrated XPT2046 touch driver mode. If you would like to give this a try, please download the latest GUIslice repo (includes a revised config file). Thx.

ImpulseAdventure, 2019-06-01:
I believe the enhancements that were made to GUIslice's TFT_eSPI touch handler mode should greatly improve the compatibility and functionality in line with the other driver modes. Dynamic rotation should also operate properly.

If you continue to experience any other troubles, please feel free to update this issue and I'll reopen it, or create a new one.

Thanks again for the report and workaround!

closed 2019-06-01 by ImpulseAdventure