ImpulseAdventure/GUIslice #146: esp8266 reboot with Adafruit GFX fonts

jmcastillejo, 2019-06-05:
Hi Cal!

When I use de fonts of Adafruit_GFX library the esp8266 reboot:

GUIslice version [0.11.3.7]:
- Initialized display handler [TFT_eSPI] OK
- Initialized touch handler [TFT_eSPI(XPT2046)] OK

Soft WDT reset

>>>stack>>>

ctx: cont
sp: 3ffffba0 end: 3fffffc0 offset: 01b0
3ffffd50:  00000000 000081f6 00000020 0000000c  
3ffffd60:  40244d24 4010429c 3ffef9b4 feefeffe  
3ffffd70:  00000001 401034bf 3ffed330 00000090  
3ffffd80:  3ffeec1c 0000001e 00000078 402031bb  
3ffffd90:  0078001e 00180054 00000000 feefeffe  
3ffffda0:  00000072 00000001 00000021 0000ffe0  
3ffffdb0:  0078001e 00180054 00000000 3ffeea90  
3ffffdc0:  00000078 0000001e 3ffeeb3c 40201b54  
3ffffdd0:  3ffeeb1c 3ffe858c 00000008 0000ffff  
3ffffde0:  00000000 3ffeec1c 00000000 00000030  
3ffffdf0:  0078001e 00180054 0000ffff 0000ffff  
3ffffe00:  0022ff00 3ffeeb1c 00000000 00000000  
3ffffe10:  3ffeea88 00000000 00000054 00000018  
3ffffe20:  3ffeea96 00000000 00019555 00000000  
3ffffe30:  40104cdc 00019555 3ffeedec 00000000  
3ffffe40:  00000001 3ffeea88 3ffeeb3c 40201c08  
3ffffe50:  00000001 00000002 3ffeea88 00000000  
3ffffe60:  3fffff00 3ffeeafc 3ffeeb3c 402027c4  
3ffffe70:  3ffffec0 00000258 3ffeec1c 00000001  
3ffffe80:  00000000 3ffeeafc 00000000 402027f9  
3ffffe90:  00000001 00000002 3ffeea88 00000000  
3ffffea0:  00000001 00000002 3ffeeafc 00000000  
3ffffeb0:  3ffeeb3c 00000001 3ffeeb3c 40203619  
3ffffec0:  00000000 014001e0 00000000 3ffeeb40  
3ffffed0:  00000002 00000001 3ffeeb3c 4020286c  
3ffffee0:  00000001 00000002 3ffeeafc 00000000  
3ffffef0:  00000000 014001e0 3ffeeafc 3ffeed94  
3fffff00:  00000000 3ffeeb3d 3ffeeb3c 402028e4  
3fffff10:  00000001 00000002 3ffeeafc 00000000  
3fffff20:  3ffeeb3f 00000001 00000000 00000001  
3fffff30:  3fffdad0 00000000 3ffeeb3c 3ffeed94  
3fffff40:  00000000 00000001 3ffeeb3c 40202add  
3fffff50:  00000000 00000000 00000000 40202350  
3fffff60:  3fffdad0 00000000 3ffeeb3c 402015bb  
3fffff70:  00000001 00000000 0000ed3c 00000000  
3fffff80:  3fff0000 00000000 3ffeeb3c 402015f8  
3fffff90:  3fffdad0 00000000 3ffeed64 40202e30  
3fffffa0:  feefeffe feefeffe 3ffeed64 402079ec  
3fffffb0:  feefeffe feefeffe 3ffe8518 401004d9  
<<<stack<<<

any idea?

regards

ImpulseAdventure, 2019-06-06:
Thanks Jim for reporting this with the version & stack trace!

I have recreated the issue on an ESP8266. It appears that TFT_eSPI will crash during the font rendering if we have included Adafruit_GFX directly into the sketch (which loads the FreeFonts), rather than letting TFT_eSPI load its own copy of the FreeFont files. I have seen reports of compatibility issues between ESP8266 and PROGMEM that may be related.

In the meantime, the following is working for me -- in your sketch, please try changing the `#include` from:
```c++
  #include <Adafruit_GFX.h> // REMOVE
  #include "Fonts/FreeSans12pt7b.h"  // REMOVE
  ...
  if (!gslc_FontSet(&m_gui,E_FONT_BTN,GSLC_FONTREF_PTR,&FreeSans12pt7b,1)) { return; }
```
to:
```c++
  #include "TFT_eSPI.h" // ADD
  ...
  if (!gslc_FontSet(&m_gui,E_FONT_BTN,GSLC_FONTREF_PTR,&FreeSans12pt7b,1)) { return; }
```

BTW -- you appear to be running a moderately old version of GUIslice (from mid-May). Once we have confirmed whether the above addresses the font issue, you may want to consider rolling forward to the latest version as it has also added improvements to the TFT_eSPI's integrated touch handling mode that you are using (as noted in #143).

thx

jmcastillejo, 2019-06-06:
I have made the changes and it also works for me, thank you!

ImpulseAdventure, 2019-06-07:
Thank you Jose for confirming that it works!
I will continue to see if I can identify the source of conflict in the original include.

ImpulseAdventure, 2019-08-10:
A number of the examples have now been updated to selectively include the appropriate font/libraries. This change should avoid the noted incompatibility between TFT_eSPI and Adafruit-GFX libraries.

closed 2019-08-10 by ImpulseAdventure