ImpulseAdventure/GUIslice #148: mcufriend ILI9320 8-bit parallel on stm32 bluepill

mcbp223, 2019-06-17:
I am trying the [0.12.0](https://github.com/ImpulseAdventure/GUIslice/releases/tag/v0.12.0) library examples  on a mcufriend 2.4" TFT with resistive touchscreen + stm32 bluepill, but controls don't get painted. Here is what works: 
- uno + mcufriend_kbv
- mega + mcufriend_kbv
- mega + GUIslice, all examples work, including touchscreen
- bluepill + mcufriend_kbv works, including touchscreen

On the combo bluepill + GUIslice I am working with ex02_ard_btn_txt, with a config file based on stm32-adagfx-mcufriend-simple.h, where I have modified for touchscreen to match the hardwired shield, as following: 
```C++
  #define ADATOUCH_PIN_YP     PA6
  #define ADATOUCH_PIN_XM     PA7
  #define ADATOUCH_PIN_YM     PB7
  #define ADATOUCH_PIN_XP     PB6
```
I have also activated the Debug diagnostic modes located by the end of the config file.
In gslc_DrvInit I have added some debug printing for m_disp.readID(), and it correctly reads 0x9320.
In the sketch I have sprinkled some drv calls, such as 
```C++
  // test driver
  gslc_DrvDrawFillRect(&m_gui, (gslc_tsRect){120,100,100,60}, GSLC_COL_BLUE);
```
and they *do* get painted.
The touchscreen seems to work, as it does print debug info, and when I click in the region where the button should be painted, it does call the callback.
The [previous release](https://github.com/ImpulseAdventure/GUIslice/releases/tag/v0.11.2) painted the controls on the screen after some minor tweaking with defines, but there was no stm32 support at the time, and I couldn't get the touchscreen to work with it either.

Any idea what to try next, to get the controls painted?

ImpulseAdventure, 2019-06-17:
Hi Mike --

Thanks for providing detailed info regarding what you have tested!

As the STM32 + mcufriend combo utilizes a number of config options/customization outside GUIslice, I'd like to recreate your particular setup. 

Could you confirm:
- Which STM32 core you are using?
- What bluepill pinout is used for LCD_D0..D7 / LCD_RST / LCD_CS / LCD_RS / LCD_WR / LCD_RD
- When you say the controls don't get painted: do the controls get drawn once and then not updated after touch presses, or is the screen left white / black? 

Lastly, if you could attach a ZIP file with your any of your modified mcufriend config files, that would be great. For example:
- mcufriend_kbv/utility/mcufriend_special.h
- mcufriend_kbv/utility/mcufriend_shield.h

thx

mcbp223, 2019-06-17:
Hi,
I have tried with both STM32F103C8T6 and the CS32 clone compiled for both 64kB/128kB, on blue pill board, and behavior is the same.
The display used is a 2.4" mcufriend shield for UNO, with the default MCUFRIEND_kbv wiring found in `mcufriend_shield.h,` by the line 446 (look for "Uno Shield on BLUEPILL"):

	LCD_RST	PB9
	LCD_CS	PB8
	LCD_RS	PB7	touch YM
	LCD_WR	PB6	touch XP
	LCD_RD	PB0 *	so PB5 can be used for SD card MOSI

	LCD_D0	PA0
	LCD_D1	PA1
	LCD_D2	PA2
	LCD_D3	PA3
	LCD_D4	PA4
	LCD_D5	PA5
	LCD_D6	PA6	touch YP
	LCD_D7	PA7	touch XM

I am using MCUFRIEND_kbv current version [2.9.8](https://github.com/prenticedavid/MCUFRIEND_kbv/releases/tag/v2.9.8) unmodified, it detects the graphic chip by reading it's ID.

The controls don't get painted I think, although there is some flickering when I use the touchscreen. The screen is initialized to gray, and it flickers between white and gray on touchscreen activity, but kind of randomly. If I call Update only once (by setting m_bQuit = true after the first call in loop) I can figure I didn't miss any erasure if the controls were painted once.

ImpulseAdventure, 2019-06-18:
Thanks very much for the added details. I have recreated this setup locally (though with an 0x9341 display).

## Summary
From what I have seen so far, I suspect that this might not be an issue specific to GUIslice -- instead I think it may be bringing to light a known incompatibility between **Adafruit_TouchScreen** library and the pin-sharing parallel 4-wire touch displays (common for **mcufriend**). This is the combination selected by the config: `stm32-adagfx-mcufriend-simple`.

That said, I would be keen to look into adding in any necessary driver selection or workarounds in GUIslice to support this particular setup.

## Details
### Debug output
Just to rule out the existence of any simple errors, we should ensure that we set `DEBUG_ERR` to `2` in your config. This enables all of the major error messages via the serial monitor.

### Pin-sharing issue
Your particular setup is utilizing an mcufriend parallel display with 4-wire touch handling provided by **Adafruit_TouchScreen**. It is important to note that Adafruit's 4-wire resistive touchscreen library (Adafruit_TouchScreen) is not generally compatible with many of the mcufriend displays as it fails to restore pin state after reading the touch interface. Since these mcufriend displays often share the touch pins with the TFT pins, corruption is common.

### Config w/o touch
I have now uploaded an STM32 config (`stm32-adagfx-mcufriend-notouch`) that provides a no-touch option. I believe it may work better on your setup and would be interested in hearing if it paints the controls properly for you.

Once we have confirmed that the above "notouch" config works for you, we can then proceed to address the touch option.

Enabling the touch handler (`stm32-adagfx-mcufriend-simple`) may result in the issues you observed due to the potential conflict between the Adafruit_TouchScreen library & the pin-sharing TFT (as the "`-simple`" config leverages the default Adafruit_TouchScreen library).

### mcufriend w/ touch
You had mentioned that mcufriend_kbv + TouchScreen worked in your testing. This is a good sign as it indicates that we should be able to enable the same combo for GUIslice too.
- Were you running mcufriend_kbv/TouchScreen_Calibr_native or Touch_shield_new? (Did you modify the example other than to update the XP/YP/etc. pinout?)
- I am able to get the mcufriend_kbv + TouchScreen to work correctly if I modify Touch_shield_new to use David's hacked version of the Adafruit_TouchScreen library (TouchScreen_kbv) included in the examples/Touch_Calibr_native directory.

For reference, David's hacked version of the Adafruit_TouchScreen library adds a few compatibility fixes, namely:
- Saved pin state after touch reads
- Reduced ADC resolution from 12 bits to 10 bits

Incidentally, David points out the following in his latest "mcufriend_how_to.txt" regarding the Bluepill:
```
Maple core:  use Touchscreen_kbv library
STM Core: regular Touchscreen libraries should be ok.
```
My compilation is currently using the Maple core, so this could lead to a difference in behavior versus your setup if you are using the STM core.

### Adding STM32 + TouchScreen workarounds for GUIslice
I can take a look to see what might be involved in either adding a workaround for the STM32 to continue using the original Adafruit_TouchScreen or possibly adding support for David's hacked version of this library.

The best solution of all would be if I could find a fork of the Adafruit_TouchScreen library that has addressed some of these compatibility issues (with ESP32 & STM32) and then support this option in GUIslice. If anyone is aware of one, please let me know. Otherwise, I might proceed to develop one myself.

That said, it is possible that there is an altogether different issue you're observing within the GUI, but given my prior experience with Adafruit_TouchScreen on these mcufriend displays, I'd like to rule out the above potential compatibility issue first.

thanks!

mcbp223, 2019-06-18:
Ok you are perfectly right regarding the conflicts due to pin-sharing between the touchscreen and display, I have defined DRV_TOUCH_NONE instead of DRV_TOUCH_ADA_SIMPLE, and now everything gets painted correctly, and I really like the speed of example 06 on stm32 versus 8-bit atmel.

The combo that worked for me was with David's hacked touchscreen_kbv version, hacked a little by me too to use the same calibration settings between uno, mega and bluepill, by scaling from 12-bit to 10-bit. I would suggest to David to promote his version from example level to library level, since it is the one that works, and/or to make a PR to add stm32 and 12-bit ADC compatibility to the Adafruit library itself, since the differences aren't that big.

Regarding setting DEBUG_ERR to 2, my bluepill hangs with the green LED rapidly blinking when calling delay() in gslc_PageRedrawGo(), I had to comment that line out; have you encountered this too?

ImpulseAdventure, 2019-06-18:
Great!
Yes -- I had assumed that you may have been using David's hacked **TouchScreen_kbv**.

I agree that a good option would be to see if he might be open to package up his version (so that it is available via the Arduino IDE library manager). I have intended to create an equivalent lib for quite some time (as we ran into the same issues with ESP32 support some time ago). I'll check in with David to see what his preference might be. Ideally, I'd like to fork from the latest Adafruit version (David's version predates a couple recent upstream commits).

With regards to the Adafruit lib PRs: it doesn't look like functional PRs are being integrated into the upstream as their scope is likely limited to the displays that they currently support. One will note issues / PRs specific to the pin-sharing and ADC resolution have been pending for a couple years.

As for the DEBUG_ERR=2, there shouldn't be delays in the code unless `DBG_REDRAW` has been enabled. The DBG_REDRAW mode is purely for debugging the library's redraw process and not intended for general usage. If you currently have DBG_REDRAW enabled, try disabling it and see if DEBUG_ERR=2 mode might work better for you.

thx

mcbp223, 2019-06-18:
I have tried a crude test in `GUIslice_drv_adagfx.cpp` with
```C++
  #include <stdint.h>
  #if 0
    #include <TouchScreen.h>
  #else
    #include <TouchScreen_kbv.h>
    using TouchScreen = TouchScreen_kbv;
    using TSPoint = TSPoint_kbv;
  #endif
```
but that brought me back to no controls painted, so there is more to restoring pin state or avoiding conflicts than just replacing the library.

Regarding DEBUG_REDRAW yes I have mistakenly tought you were referring to it when I was in a hurry to post the good news, but I have activated both DEBUG_ERR=2 and DEBUG_REDRAW, just that I have to comment out the delay() call, otherwise the bluepill curiously freezes right there.

ImpulseAdventure, 2019-06-19:
With a similar quick hack to `GUIslice_drv_adagfx.cpp` (and copying `TouchScreen_kbv.*` into `/Arduino/libraries/GUIslice/src/`), I can get the display to work with touch enabled, provided I add in the following pin "restoration" calls: after `getPoint()`:
```c++
  TSPoint_kbv p = m_touch.getPoint();
  pinMode(ADATOUCH_PIN_XM, OUTPUT);
  pinMode(ADATOUCH_PIN_YP, OUTPUT);
```

I found that I needed to increase the minimum touch threshold in order to avoid excess touch events for my particular display. In the config I changed `ADATOUCH_PRESS_MIN` from 10 to 100.

Using the above setup, I am able to get my **STM32F103C8** working with a mcufriend 8-bit parallel TFT (ID=0x9341) using 4-wire resistive touch.

If the above works for you as well, then I will proceed to explore the idea of productizing a patched Adafruit_TouchScreen library and then integrate into GUIslice.

mcbp223, 2019-06-19:
Hi, I have added the restoration by the line 1647, under `#elif defined(DRV_TOUCH_ADA_SIMPLE)`, and the examples started working but not completely, maybe there are still remaining places restore pin mode.
I am observing this behavior as of now:
- when I touch and move there is a yellow pen painting trace, I assume this is by design, maybe triggered by the debug flags
- occasionally the screen goes white, no page repaint
- no page repaint, ex26_ard_calc doesn't open the keyboard popup, ex24_ard_tabs doesn't paint the Extra tab, but when I press on Extra the progress bar stops, and it starts again when I press Main
- in ex04_ard_ctrls checkboxes and slider work, but with the radios I'm having difficulties, not sure if they switch
- the sliders in ex10_ard_textbox seem to work
- I can't get the text in the ex26_ard_calc example to display, when I press Add I expected a 0 in the Result box, even if there is nothing in the Value boxes; I have verified that CbBtnCommon gets called.

I have noticed that you've done previously pin state restoration under FIX_4WIRE_PIN_STATE, that is unfinished for STM32 as of now, thus you've disabled it for `#if defined(ARDUINO_ARCH_STM32) || defined(__STM32F1__)`. So I get better results if I place the restoration you tried in your last reply not after `m_touch.getPoint()`, but a bit later, after `m_touch.pressure()` gets consumed too, right where FIX_4WIRE_PIN_STATE was to be called if active, by the line 1739; with this I get page repaints correctly I think, maybe I'll just need to calibrate for more precision.

ImpulseAdventure, 2019-06-20:
OK -- I believe we're making good progress in getting the STM32 to operate with the pin-sharing display & work around the 4-wire Adafruit touch controller.

Your observations suggest that you have a number of the debug flags enabled.
- Some of the debug modes will interfere with the normal redraw mechanisms.
- My recommendation would be to disable all of the `DBG_*` flags in Section 10 of the config, but leave `DEBUG_ERR` as 2 in Section 5.
- The yellow boxes are as a result of `DBG_TOUCH` being enabled, the intention being to assist the user in ensuring that the touch calibration is correct (small yellow boxes should be drawn under the touch presses).

As you point out, the pin-sharing recovery needs to occur later, otherwise the Adafruit_TouchScreen `pressure()` call may similarly clobber the redraws, potentially resulting in your ex24 & ex26 observations.
- The best placement for the pin restoration will probably be immediately after the existing FIX_4WIRE_PIN_STATE block (unused as you noted) containing the TDrvRestorePinState calls, as follows:
- 
[image]

With the above updates, the examples appear to be displaying/redrawing much better on my test setup, but I am seeing that Adafruit_TouchScreen's getTouch() is often returning a spurious initial Y coordinate, which will cause some touch presses to be ignored. I will continue to debug the getTouch() calls to see if some additional filtering/detection workarounds may be necessary (likely a manifestation of the pin clobber / restoration impacting the touch reading).

thx

mcbp223, 2019-06-20:
Yes, perhaps only fine tuning remains to be done for now, close the issue whenever you feel we're done with testing.
Thank you for your great support and for this great library, and I think it's worth having those pin sharing displays in it, since there are many of them in the wild, of different sizes, and the bluepill gives them more speed than 2560, although less flash memory and less remaining pins.

ImpulseAdventure, 2019-08-11:
I will mark this as open for future enhancement.

As the primary issue is outside of GUIslice, the best solution would be to identify or create an enhanced version of the **Adafruit_Touchscreen** library that supports pin-sharing displays properly.

I have implemented a few workarounds within GUIslice to provide more resilience against the Adafruit_Touchscreen limitations but they won’t be able to address all device combinations.

Although I am tempted to create a revised version of the Adafruit_Touchscreen, I don’t anticipate I will have an opportunity for some time. Ideally we would be able to locate a fork that has already integrated many of these improvements, so if you come across one please let me know!

Thanks again for all the testing

closed 2019-08-11 by ImpulseAdventure