ImpulseAdventure/GUIslice #149: Guislice v12.0 Touch not working

notascooby, 2019-06-21:
Firstly, what a great library and professional docs.
Just installed v12.0 over v11.2 and re-compiled my project in Arduino 1.8.9 for ESP32. Touch screen no longer works.  Calibrated screen using "diag_ard_touch_calib sketch" which worked perfectly. Entered new values in esp-tftespi-default-xpt2046_int.h (ADATOUCH_X_MIN etc.)
and tried your Arduino example ex5 sketch, but same issue, no Touch.
In previous versions I used TFT_ESPI_TOUCH_CALIB  { 321,3498,280,3593,3 }. In v12.0 I have also modified the TFT_ESPI_TOUCH_CALIB  line with { 321,3498,280,3593,3 } in addition to the ADATOUCH_X_MIN etc.
Tried changing ADATOUCH_PRESS_MIN & MAX values but still no Touch. (how do we know what these values should be)
Un-installed Guislice & re-installed but no Touch.
My GuiSlice Configuration:
config.h has /configs/esp-tftespi-default-xpt2046_int.h enabled
config_arg.h has DRV_DISP_TFT_ESPI enabled, & DRV_TOUCH_TFT_ESPI  enabled
Using TFT_eSPI v. 1.4.11. GPIO as follow:
//Used for Display & Touch screen
TFT_MISO  25
TFT_MOSI  23
TFT_SCLK  19
TFT_CS    22               // Chip select control pin. 
TFT_DC    21               // Data Command control pin.
TFT_RST   18               // Reset pin (could connect to RST pin)
TOUCH_CS  14            // Chip select pin (T_CS) of touch screen

Hardware of device
ESP32 development board
3.5" RPI touch_ILI9341 display not modified (very similar to the one in your modification picture)

All has worked well through versions 10.x & 11.x.
Any help appreciated as I have had to revert to v11.2 for now

Thanks

ImpulseAdventure, 2019-06-21:
Thanks very much for reporting this!

You are right... while I had tested the new `DRV_TOUCH_TFT_ESPI` mode (#143) in calibration, I had overlooked the configuration import updates necessary for normal operation.

I have now uploaded a fix for this in the repository (guide: [download & install the latest GitHub release](https://github.com/ImpulseAdventure/GUIslice/wiki/Installing-GUIslice-on-Arduino#installing-from-github-source) ). Assuming this works for your setup, I will push the change into an upcoming release `0.12.1` for the Arduino IDE library manager.

notascooby, 2019-06-23:
Sorry for taking so long.
Downloaded zip file (files dated 21/06/2019 at 10:05) of changed 12.0. Tried today but the touch still does not work.

ImpulseAdventure, 2019-06-23:
Thanks for the followup, notascooby.

The fact that the calibration sketch is working for you is a good indicator that the TFT & Touch are both operating correctly. I just retested with my ESP32 setup, similar config and it appears to be functional, so there must be a loose end here.

If you have an opportunity to try the following, that would be appreciated:
- 1: To ensure we have the same setup, can you confirm which version of TFT_eSPI you are using? (eg. release 1.4.11)
- 2: Please attach the output from the `diag_ard_touch_calib` sketch here
- 3: Please modify your `/configs/esp-tftespi-default-xpt2046_int.h` to change:
  - Section 5: `DEBUG_ERR 2`
  - Section 10: Uncomment `DBG_TOUCH`
- 4: Please attach the output from the `ex02_ard_btn_txt` sketch when you press on the display. Do you see any small yellow boxes appear where you press?

thanks!

notascooby, 2019-06-24:
I can confirm that the amended version of 12.0 does work
The reason it didn't work for me yesterday is that I had been experimenting with the ADATOUCH_PRESS_MIN  option (had value of 5) in the original 12.0 version and had not restored it to the default of 200 when the new version of 12.0 was released. Sorry
In case the information is useful to you.
TFT_eSPI version=1.4.11
The two output files you requested are below.
[OUTPUT from diag_ard_touch_calib.txt](https://github.com/ImpulseAdventure/GUIslice/files/3320975/OUTPUT.from.diag_ard_touch_calib.txt)
[OUTPUT from ex02_ard_btn_txt.txt](https://github.com/ImpulseAdventure/GUIslice/files/3320980/OUTPUT.from.ex02_ard_btn_txt.txt)

Could the diag_ard_touch_calib sketch also display the ADATOUCH_PRESS_MIN/MAX values as I for one didn't know what these values should be.
Thanks Again

closed 2019-06-24 by notascooby

ImpulseAdventure, 2019-06-24:
Excellent... thank you for confirming that the latest version works properly and attaching the output files!

I agree with you that the diagnostic sketch would be a good place to help users identify a suitable range for `ADATOUCH_PRESS_MIN/MAX` and will be adding this feature to an upcoming version.