ImpulseAdventure/GUIslice #150: DRV_TOUCH_TFT_ESPI v0.12.1 Touch doesnt respond quickly

notascooby, 2019-06-27:
After changes to v0.12.1 the touch does work with the RPI 3.5" display & ESP32 however it doesn't always respond correctly.
Not easy to explain, but a button may have to be pressed multiple times to work. Button will in some cases change colour but the programmed requirement for that button won't run.
It is though there is some kind of delay, a de-bounce routine isn't working as it should, or focus is lost.
I reverted to V0.11.2 and all worked correctly. Tried your example ex07_ard_slider. Again it does work but if the lower slider is moved back and forth quickly, the slider above moves and the slider I am trying to move stops moving.
Very vague I know, but if you want me to try anything or answer any questions please let me know.
For now I have reverted back to v0.11.2 again.

My GuiSlice Configuration:
config.h has /configs/esp-tftespi-default-xpt2046_int.h enabled
config_arg.h has DRV_DISP_TFT_ESPI enabled, & DRV_TOUCH_TFT_ESPI enabled
ADATOUCH_PRESS_MIN/MAX settings. Tried defaults (200/4000) & 300/3500 after a recalibration of screen.

Hardware of device
ESP32 development board
3.5" RPI touch_ILI9341 display not modified (very similar to the one in your modification picture)

ImpulseAdventure, 2019-06-27:
### Touch Filtering Added
Thank you for the report. I have updated the repository to add touch filtering specific to `DRV_TOUCH_TFT_ESPI` mode. Please let me know if it improves the operation.

### Details
What you are observing is likely due to noisy touch readings from your display. The changes for DRV_TOUCH_TFT_ESPI (that exist in v0.12.1) utilized the **raw** (unfiltered) touch readings from TFT_eSPI. For some displays, this raw reading can be very noisy. If you ran `diag_ard_touch_calib` and observed the way in which the red dots tracked your touch presses (after completing the calibration phase), you would probably see a wide spread in the dot placements due to noise.

For the update I just made to the repo, I have now added support for touch filtering in this mode, using the algorithm implemented by validTouch() in TFT_eSPI. I believe this should provide the same touch behavior as you would have noticed in GUIslice 0.11.2 (which used the entirety of TFT_eSPI’s integrated touch handler). Upon rerunning diag_ard_touch_calib with this update, you should notice that the red dots track much more closely to your presses.

This added filtering is now enabled by default but can be disabled by commenting out `DRV_TOUCH_TFT_ESPI_FILTER` in `GUIslice_drv_tft_espi.cpp`.

If you have an opportunity to try it out, I would be interested to hear if this update improves the touch response.

Thanks!

notascooby, 2019-06-27:
The added touch filtering has solved the unresponsive touch screen.
Works the same as v0.11.2 now.
Thanks

closed 2019-06-27 by notascooby