ImpulseAdventure/GUIslice #152: Requst: Support ILI9486 with SPI interface

sjbaird, 2019-07-06:
I am trying to use the GUI builder supported by GUIslice to program my LCD but I'm having problems.

I managed to get my hands on a Waveshare LCD that does not seem to be supported by any of the Arduino libraries.

(https://www.waveshare.com/product/modules/oleds-lcds/arduino-lcd/4inch-tft-touch-shield.htm)

They have provided a driver example that does work and I can perform the tests.

(https://www.waveshare.com/wiki/File:4inch_TFT_Touch_Shield_Code.7z)

I have been trying everything I can think of to make this work on the two libraries that actually claim t support this driver.  (MCUFRIEND_kbv, and Ucglib)  Mostly I have configured the pins to match the board and have tried to combine the examples from Waveshare with them but all I get is a blank screen.

ImpulseAdventure, 2019-07-06:
Hi -- GUIslice depends on underlying graphics libraries for particular hardware support.
If you are able to locate a suitable Arduino-compatible library for your display that isn't already supported by GUIslice, then I can consider adding support for it.

In the schematics I have seen of the 4" Waveshare display, the shield is converting a 16-bit parallel interface at the ILI9486 controller to a SPI interface via shift registers. This is likely contributing to the difficulty in finding a compatible graphics library, and may also limit performance. I'm not sure that mcufriend_kbv currently supports this interface variant.

What microcontroller are you using with the display? An UNO or MEGA?

I don't have the same display to test with, but I believe Bodmer's [TFT_eSPI](https://github.com/Bodmer/TFT_eSPI) may support it (after selecting the appropriate User_Setup config). I haven't looked to see whether the **UTFT** library might potentially support this particular interface (eg. `UTFT myGLCD(ILI9486,rs,wr,cs);`

thx

sjbaird, 2019-07-07:
Thank you for your reply, 

I didn't expect such a fast reply so I apologize for the delay.

The second link I provided is a working driver library of sorts created by Waveshare I have it working on both UNO (plug and play) and MEGA (both plug and play and by change LCD_CS to 53 so that I can use an extension cable shield that I built)

This driver library seems to be very close to the ILI9341 display that is supported however I can not get a working configuration.     

Is it possible that you could help me use this driver configuration in the configuration instead of creating something new?

I have looked at the TFT_eSPI library and tried it because the driver board is the closest match. However, being that it is geared for ESP I was unable to get an example to work.

I did try mcufriend_kbv in as many configurations that I could think of however it was not working.

Keep in mind that I am an amateur and can barely make things work when I have working examples. This LCD world blindsided me with its complexity, as I originally tried a used 7-inch black and white printer LCD with absolutely no luck. :)

I am guessing that there are going to be quite a few people who would benefit from getting working support since they don't cost much and they are top of the list at the moment at the "big book store" (azon).

I understand that your time is valuable so I have been working on making the given driver set up work for me however there examples are limited and I'm not sure how to build my own scripts. (At least its forcing me to learn about how to use background functions.)

Thank You again
Steven

sjbaird, 2019-07-07:
Also from what I can tell with my limited understanding, they are driving this with 8-bit and yes it is sluggish but at the end of the day, I just need a graphing display with a few buttons and some images since this will be an interface for a data acquisition device.

ImpulseAdventure, 2019-07-10:
Hi there Steven --

I can appreciate how the "LCD world" can be a bit daunting with complexity! There are so many different display modules, controllers, libraries and connectivity options. Unfortunately, the particular display you are interested in using is a bit unusual in the choice of interface, making it difficult to find many suitable library options as you noted.

For me to add support for this module (which would cover both the [Waveshare 3.5"](https://www.waveshare.com/product/arduino/shields/wavesharer/3.5inch-tft-touch-shield.htm) and [Waveshare 4.0"](https://www.waveshare.com/product/modules/oleds-lcds/arduino-lcd/4inch-tft-touch-shield.htm) TFT shields), it seems like the cleanest approach would be to extract the existing Waveshare driver code, wrap it with an Adafruit-GFX compatible API and release it as an Arduino library. From there, I would need to get my hands on one of these displays to test with. Once we have a compatible Arduino library with a GFX-style interface, it would then be a simple matter for me to add support to GUIslice.

The good news is that I have requested and been granted permission by Waveshare to create such an Arduino library, in case I find an opportunity to proceed.

I have started creating this library and will let you know if I get to the point where I have something for you to test, though note that it may take some time.

Hope that helps!

ImpulseAdventure, 2019-07-11:
Hi @sjbaird -- I made some progress and have now created a preliminary Arduino-compatible ILI9486 SPI library.

If you would like to try it out on your Waveshare display, please send me an email to guislice@gmail.com and I'll send you the details.

thanks!

ImpulseAdventure, 2019-07-20:
**UPDATE**: I have created an Arduino Library for the *Waveshare ILI9486* driver and released a preliminary version at [Waveshare ILI9486](https://github.com/ImpulseAdventure/Waveshare_ILI9486). This version includes some preliminary GFX-compatible APIs and includes a port of the ILI9341 graphicstest.

The next step will be to provide additional APIs that will support integration into GUIslice. Some features (such as touch support) would be deferred to a later update once I have the hardware on-hand to test.

Special thanks to @sjbaird for providing great testing feedback on this library!

ImpulseAdventure, 2019-08-08:
Support for [Waveshare_ILI9486](https://github.com/ImpulseAdventure/Waveshare_ILI9486) has now been integrated into GUIslice, selected by config option `DRV_DISP_WAVESHARE_ILI9486`. The following display modules are supported by this driver mode:
- Waveshare 4" Touch LCD Shield for Arduino (SKU: 13587)
- Waveshare 3.5" Touch LCD Shield for Arduino (SKU: 13506)

Example configurations:
- `ard-shld-waveshare_40_notouch.h` (No touch driver enabled)
- `ard-shld-waveshare_40_xpt2046.h` (Touch driver enabled)

Thanks again for the very helpful feedback, Steven!
I now have these display modules on-hand so testing should be more straightforward going forward.

Please contact me or file a new issue if you encounter any difficulties with using this library with the hardware. Note that there are some minor visual tweaks that I intend to address with the Waveshare library (namely the top-left boundary of certain elements is offset by a pixel).

closed 2019-08-08 by ImpulseAdventure